jlam55555/veikk-linux-driver #2: `hid-generic` stealing instructions

jlam55555, 2019-06-12:
## Bug description

For some older kernel versions (this appeared to be happening in kernel 4.15, but not in 4.18 or 5.1), the `hid-generic` driver seems to want to gobble up the driver the moment it's plugged in, so need to unload it  from `hid-generic`.

The cause of this problem is not confirmed to be the kernel version, but I will do additional testing and see if `hid-generic` has had any changes in newer kernel versions. I've had another user confirm that updating kernel from 4.15 to 4.18 solved the issue.

## Workarounds

### Temporary fix: Force unload device from `hid-generic` driver

If this fix works, it should persist until your next reboot. Do this before installing the driver; if the driver is already installed, remove it and reinstall it after running the following command:

    echo -n "xxxx:2FEB:000x.xxxx" > /sys/bus/hid/drivers/hid-generic/unload

replacing the `x`s with the corresponding values from `ls /sys/bus/hid/devices`. (`2FEB:000x` is the vendor ID, product ID of the Veikk product.) If this is successful, running `ls /sys/bus/hid/drivers/veikk`  should list the Veikk device.

This command may require you to sign in to the root user with `sudo su` to get the necessary permissions, but I've had a user report that even this gives them `Permission denied.` The below fix solved it for that user.

### Permanent fix: Upgrade kernel to 4.18+

The default kernel for Ubuntu 18.04 LTS is 4.15, and older systems have older kernels ([source][1]). If you're using an older kernel, upgrading it to 4.18+ solved the issue in all known cases.

This can be done by [installing Ukuu][2] (commands below for Ubuntu):

    sudo add-apt-repository ppa:teejee2008/ppa
    sudo apt-get update && sudo apt-get install ukuu

Run Ukuu, then download a kernel version 4.18+. Reboot to boot into the new kernel and try installing again.

[1]: https://askubuntu.com/questions/517136/list-of-ubuntu-versions-with-corresponding-linux-kernel-version
[2]: https://linuxhint.com/upgrade-kernel-ubuntu-1804/

artixnous, 2019-06-13:
For feedback purposes, the driver works flawlessly on on my son's machine running Artix, kernel 4.18.4 and Krita. A small configuration utility (or module parameters) to adjust the pressure thresholds and keep the pen points from breaking would be very useful. I had contacted VEIKK asking for a linux driver and they offered instead a refund; I'll point them to you and hope they'll help you.

  Kudos!

jlam55555, 2019-06-14:
@artixnous  I got the email, and thanks for supporting the cause! I guess the timing of writing this driver and your email coincided nicely. That's very nice of you!

Right now, I am working with another user to try to fix the other issue (dealing with Secure Boot), and I was hoping to fix this bug as well. (It's not an issue specific to this driver, but any external driver on a UEFI Secure Boot system, but I'd like to write a brief guide nonetheless.) Unfortunately, I don't have the drawing tablet on me and can't actively work on the driver atm, but rather only work out installation issues (hence the Secure Boot issue). Once I do get back home and can use the device, I'll begin work on the configuration tool and let you know when progress has been made.

On an aside, can you qualify your statement "A quick search in the internet shows many Linux users working with your tablets"? I have seen a few (but not many) posts, such as [this](https://unix.stackexchange.com/questions/507687/graphic-tablet-veikk-pressure-sensitivity-on-linux). I think giving more concrete evidence might be helpful. (On the other hand, I've had four people (including you) reach out because of they have Veikk drivers and use Linux in these past two days, but I'm not sure if this qualifies as "many".)

artixnous, 2019-06-14:
Well, it was a little bit of an overstatement, hopelessly aimed at motivating the company. There is, however, [a FAQ at high-traffic Gearbest](https://www.gearbest.com/graphics-tablets/pp_009933810443.html), which suggests at least some demand.

artixnous, 2019-06-14:
[An AUR package for our convenience](https://aur.archlinux.org/packages/input-veikk-dkms/).

jlam55555, 2019-06-14:
@artixnous Many thanks once again! Besides editing the README to include a link to the AUR package, is there anything else I should do? I've never dealt with packaging and distribution.

artixnous, 2019-06-14:
No, thank you for writing the driver. Packaging is way easier.

Hmm, I didn't mean pen-tilt, let me rephrase: to get the maximum pressure reading from the tablet (this holds true for the windows driver too) one must press the pen really, really hard on the tablet. That's what I meant by 'breaking the tip'. The configuration utility in windows allows for adjusting the readings to get the maximum effect with less force.

jlam55555, 2019-06-14:
@artixnous Oh I was talking about the FAQ:

> **Does the Veikk s640 have pen tilt sensitivity?**
> Yes, the inclination is 45 degrees.

Yes, pressure reading adjustability will probably be the first customization offered (except maybe rotating the tablet/left-handed mode).

By the way, do you know of what form the customization curve is? It looks like a power curve (y=kx^n) with fixed endpoints, but is that the case? I'm not too familiar with drawing programs, so advice would be good.

By the way, I'm going to make a general discussion issue thread so this issue doesn't get cluttered.

Hunanbean, 2019-06-23:
Sincere thanks for this project. 

Upgrading the kernel to 4.18+ fixed the issue of the generic override for me too. Was pulling my hair before i read the fix. the usb.id for the A15 is 0004. I have submitted this for inclusion in the linux usb.id file. The seller I purchased the A15 from claimed it had Tilt detection. It does not have Tilt detection, so no real need to look further on that issue for this drivers sake. I am new to programming, but my goal is to add support for the physical buttons on the A15 tablet itself. I simply added the 0004 id line to your existing code, compiled, and followed your instructions to this point. The pressure detection is now working as expected.

Thank you again, very much, for this driver project and the concise information you relayed on your blog. I now, after reading your blog and looking through the source code, have a general understanding i was struggling to piece together. I do hope someone with the funds comes along and is kind enough to donate a tablet in your direction.

RaSTuS26, 2019-09-19:
Tried the echo to sys trick, got the permission problems mentioned.

I can confirm that though that the driver works on kernels >= 4.18.  I have 2 versions of Linux Mint installed and it works on the later one. Also on 4.15 it goes into a quasi relative mode, in that once you first use it, it stays in the same place regardless of where the pen is, only moves if you apply pressure, but from where it left off.  On 5.0 it remains in absolute mode the whole time and the pressure sensitivity works a treat.

Here's the details:

Not working.
Linux Mint 18.3
Kernel 4.15.0-64-generic                                                                                                                                                           
                                                                                                                                                                                   
Working well.                                                                                                                                                                      
Linux Mint 19.1                                                                                                                                                                    
Kernel 5.0.0-20-generic                                                                                                                                                            

Also put an exclusion zone on the LHS because that's where gkrellm is docked, again, works on the later kernel, not on the 4.15:                                                                                                                                                                                   
/etc/modprobe.d/veikk.conf  Contents:                                                                                                                                              
options veikk bounds_map=6,0,100,100

Here's a screenshot of the positive outcome:
[image]

jlam55555, 2020-08-04:
@RaSTuS26 I only noticed now that I didn't address your issue at all. Are you sure that this VEIKK driver attached to your device in 4.15, or was it still using hid-generic when it was in relative mode? I've seen the latter happen with hid-generic, but not with this driver. If you're still using that setup and want to check, you can use `lspci -v`.

RaSTuS26, 2020-08-04:
> @RaSTuS26 I only noticed now that I didn't address your issue at all. Are you sure that this VEIKK driver attached to your device in 4.15, or was it still using hid-generic when it was in relative mode? I've seen the latter happen with hid-generic, but not with this driver. If you're still using that setup and want to check, you can use `lspci -v`.
> 
> I was also stupid about the permissions thing because I didn't understand how file redirection works. The following should work (I've updated the original post as well):
> 
> ```shell
> echo xxxx:2FEB:xxxx.xxxx | sudo tee /sys/bus/hid/drivers/hid-generic/unbind
> ```

Yes it was using your Veikk driver, not hid-generic.

I've been using the 5 series kernel for quiet some time now, so everything is working well as explained in previous post.

jlam55555, 2020-08-04:
> Yes it was using your Veikk driver, not hid-generic.
> 
> I've been using the 5 series kernel for quiet some time now, so everything is working well as explained in previous post.

Huh, strange. If I ever encounter that I'll look into it. It should be reporting `ABS_*` events and not `REL_*` events, so I'm really not sure what's going on.

RaSTuS26, 2020-08-04:
I'll have another go tomorrow and report back, it's 1:20am here. I've now got three versions compiled, v2 from 2020-02-07, v2 from 2020-08-02 and alpha 3 from 2020-08-04.  Will give each a go and see what differences I notice.

RaSTuS26, 2020-08-05:
OK, ran some quick tests on all versions mentioned in previous post, they all behaved the same.

- works in absolute
- nib acts as mouse left click
- modprobe.d options ignored
- touch sensitivity not functional in GIMP but ok in Krita

/etc/modprobe.d/veikk.conf Contents:
options veikk bounds_map=6,0,100,100

The touch sensitivity problem is only in GIMP, in Krita it works fine.  Don't want to screw with my GIMP settings though so for now I'll stick to Krita.

jlam55555, 2020-08-07:
@RaSTuS26 For GIMP you have to manually set the VEIKK devices in Edit->Input Devices to Mode: Screen (default is disabled for anything but the core pointer). That seems to do it for all versions of GIMP I've tried. No way around that AFAIK.

As for the 1/etc/modprobe.d/veikk.conf` contents, I think your parameter is strange (`bounds_map` got replaced with `screen_size` and `screen_map` in v2.0). See the relevant code from the GUI:

```cpp
format << "options veikk"
           << " screen_size=" << serializeScreenSize()
           << " screen_map=" << serializeScreenMap()
           << " orientation=" << serializeOrientation()
           << " pressure_map=" << serializePressureMap();
```

Of course, in v3, configuration will likely happen through xkb rather than kernel parameters (see the [blog post](http://everything-is-sheep.herokuapp.com/posts/veikk-linux-driver-v3-notes)) so the modparm issue will be moot soon.

RaSTuS26, 2020-08-08:
> @RaSTuS26 Thanks for your input. For GIMP you have to manually set the VEIKK devices in Edit->Input Devices to Mode: Screen (default is disabled for anything but the core pointer). That seems to do it for all versions of GIMP I've tried. No way around that AFAIK, but I wouldn't count this as screwing around with settings too much.
> 
> As for the `/etc/modprobe.d/veikk.conf` contents, I think your parameter is strange (`bounds_map` got replaced with `screen_size` and `screen_map` in v2.0). See the relevant code from the GUI:
> 
> ```c++
> format << "options veikk"
>            << " screen_size=" << serializeScreenSize()
>            << " screen_map=" << serializeScreenMap()
>            << " orientation=" << serializeOrientation()
>            << " pressure_map=" << serializePressureMap();
> ```
> 
> Of course, in v3, configuration will likely happen through xkb rather than kernel parameters (see the [blog post](http://everything-is-sheep.herokuapp.com/posts/veikk-linux-driver-v3-notes)) so the modparm issue will be moot soon.

Hi, I did try that in GIMP and lost all input function for both the mouse and tablet, I had to shut it down with Alt-F4 kbd shortcut.  That's why when I restored the previous settings I wasn't comfortable trying it again. It's ok though, I'm quite happy using Krita as it works out of the box and is a decent editor itself.

My modprobe settings must just be a leftover from an earlier version, I'll clean them out, I don't need to restrict screen size/position any longer anyway, I no longer have Gkrellm permanently docked on all screens after it started to burn into my lcd-tv monitor.

Will read up the blog shortly, thanks for the link as I'll probably stick with alpha3 for the time being.

juliangilbey, 2020-08-11:
Hi, this seems to have resurfaced, and I'd appreciate any suggestions.  I'm compiling on Debian testing, where `uname -a` gives:

```
Linux erdos 5.7.0-2-amd64 #1 SMP Debian 5.7.10-1 (2020-07-26) x86_64 GNU/Linux
```

The Veikk device (an A15 model) appears in the sys directory with three entries:

```
erdos:~ # ls /sys/bus/hid/drivers/hid-generic/
0003:051D:0002.0001/ 0003:2FEB:0004.0008/ module/
0003:05F3:0007.0002/ 0003:2FEB:0004.0009/ new_id
0003:05F3:0007.0003/ 0003:2FEB:0004.000A/ uevent
0003:1A7C:0191.0004/ bind                 unbind
```

Following the advice in some of the earlier posts, I did `modprobe -r veikk`, sent those USB ids to hid-generic/unbind and then did `modprobe veikk`.  But now the Veikk ids don't appear in either the hid-generic directory or the veikk directory.  If I then disconnect and reconnect the tablet, the Veikk ids reappear in the hid-generic directory, not the veikk directory, suggesting that hid-generic is still taking control.

One effect of this is that hovering the pen over the tablet moves the mouse, until I first touch the tablet (= mouse click). After that, it is impossible to move the mouse without touching the tablet (=mouse click and drag).

Any suggestions for how to fix this would be much appreciated!

RaSTuS26, 2020-08-11:
@jlam55555 Have GIMP working now, didn't notice the 'Dynamics' (brush/pencil) options, my tool dock was too small to see them. Will play with that for a bit now.

jlam55555, 2020-08-11:
@juliangilbey That's very strange. I really thought this behavior had gone away a long time ago with 4.18. I'm also surprised that the VEIKK module doesn't bind to the device after reloading the driver. I've never tried Debian but I might test this out and let you know.

Restarting your computer *might* work. It seems that with the veikk module loaded because it's in the `/etc/modules-load.d/` folder, it works rather than hid-generic. I think this worked in kernels <4.18 after recompiling/loading the driver, but I don't have a machine ready atm to verify this.

(Just to make sure, you're using v2 from the master branch, correct? And the driver is up-to-date?)

juliangilbey, 2020-08-11:
@jlam55555 Yes, it's really strange :-(

Ah, I used the v2 release, not the master branch.  I'll try again with that and let you know...

juliangilbey, 2020-08-11:
@jlam55555 Ah, that was it!  Building from master makes it work.  Many thanks!

jlam55555, 2020-08-11:
@juliangilbey The master branch is v2, so I'm guessing you might have been on some really old version. Still not really sure what's going on there, but glad it's solved!

(Anyway, installed Debian 10 for the first time and I think I like this. Will experiment for a little while.)

juliangilbey, 2020-08-11:
@jlam55555 I originally downloaded the .tar.gz from [https://github.com/jlam55555/veikk-linux-driver/releases](https://github.com/jlam55555/veikk-linux-driver/releases) rather than cloning master, but it seems that lots has happened since then!