jlam55555/veikk-linux-driver #4: General improvements / suggestions thread!

jlam55555, 2019-06-14:
Feel free to leave comments, advice, or questions below. Any feedback is welcome!

jlam55555, 2019-06-15:
Currently working on the first configuration utility! The end result would be a command-line ability that would be able to change the pressure mapping function (i.e., applied physical pressure vs. apparent pressure to programs), rotate/flip the tablet, and perhaps change button functions.

(Would be a lot easier if I actually had a tablet at hand...)

artixnous, 2019-06-15:
> By the way, do you know of what form the customization curve is? It looks like a power curve (y=kx^n) with fixed endpoints, but is that the case? I'm not too familiar with drawing programs, so advice would be good.
The windows configuration utility offers 3 points to control the pressure interpretation line (didn't poke much around to see if more can be added): start, end and middle, which can be dragged around to change the start-end thresholds and make either straight or curved (exponential or logarithmic). I found that a straight line starting at 0 and yielding max pressure reading at around 3/4 of actual pressure feels most natural, but this is obviously subjective. So, y=4/3*x might be a good candidate.
Another very useful setting is the mapping of the tablet to the screen. By default, the entire tablet surface is mapped to the entire screen. With custom mapping, an on-screen rectangular box can be defined within which the pointer is allowed to move when controlled by the pen/tablet. Otherwise, my hand tends to lift the pen from the surface when it reaches the end of the drawing area in Krita as if I were using a mouse and reached the end of my desk. I also think one can make only a part of the tablet active in the same way, which can offer a similar tablet-screen matching feel.

jlam55555, 2019-06-19:
v1.1! Check out the command-line utility. Orientation, screen mapping, and pressure mapping are available now. They're available in very basic forms right now, but they'll get refined later.

(i.e., screen mapping is in integer percent values, and pressure mapping is choosing out of one of the available functions; see the README)

artixnous, 2019-06-23:
Made a package for it and updated depends and optdepends:
https://aur.archlinux.org/packages/?O=0&K=veikk

fakedeltatime, 2019-08-16:
The mouse2 and mouse3 buttons are reversed, at least based on how I'm used to with other pens as well as how the same pen functioned in Windows. Since some people using this have probably gotten used to it, and option to switch it in the config would be nice.

jlam55555, 2019-08-16:
@fakedeltatime Noted.

And in general, for the last few weeks, I've been carried away with other projects -- sorry. Will continue working on [the GUI](https://github.com/jlam55555/vdc) (currently very buggy) soon.

rustweaver, 2019-08-18:
I just wanted to confirm that this driver works just fine with Veikk A30, I wasn't testing in-depth, just quick doodling, but the pressure sensitivity is working which is what was missing with default linux driver. I will try to do something about the little touchpad that A30 comes with, if I find the time.

ShapeShifter499, 2019-12-05:
> I just wanted to confirm that this driver works just fine with Veikk A30, I wasn't testing in-depth, just quick doodling, but the pressure sensitivity is working which is what was missing with [...]

I just got a hold of a A30 through a deal with the company. Was hesitant to accept since I am a Linux user but since I found this driver I said yes. It doesn't come for a few days but in the mean time I might as well set up my computer to get ready.   

@rustweaver have you made any progress with getting the trackpad working?

ShapeShifter499, 2019-12-10:
@rustweaver I am running Arch Linux and I just got the A30 tablet. Using default drivers, libwacom, and default settings the tablet appears to work ok including the shortcut buttons and the small side pad.  That side pad seems to act like a gesture pad per what it says in the manual. Is it supposed to act like a touchpad to control the mouse?  Because right now it works to zoom in and out, but it might be able to be programed so you can preform gestures to trigger shortcuts and such.

FMHemerli, 2019-12-10:
@ShapeShifter499 I own an A50 with similar pad. The touch pad on these tablets work like additional buttons with preset functions/gestures (up, down, left, right and 2-fingers click), not an actual touchpad to control the cursor. @jlam55555's driver works like magic for pressure sensitivity but there's gromgull's fork of this driver for A50 buttons that may work for A30.

For reference, I believe Arch and Manjaro (which I use) default drivers are the same, you can manage to get pressure sensitivity if you use kernel 4.14 or lower (for some reason I'm not eager to understand). I do recommend tho that you stick with jlam55555's veikk-s640-driver

ShapeShifter499, 2019-12-10:
@FMHemerli Currently everything BUT that gesture pad works on @jlam55555 driver. I can set shortcuts fine.

However with stock drivers, everything works BUT the pressure sensitivity. I might try to fiddle around later but I'd rather have the sensitivity work then the little gesture pad if I had to have a trade off. But would be nice to get everything working as intended.

jlam55555, 2019-12-30:
Hi everyone! Sorry for being inactive for so long (this was a helluva semester), but I plan to rework the driver from scratch (with new understanding and inspiration from taking an OS class) and make it more customizable and robust. (And hopefully get a GUI to work too!)

Not sure about the gesture pad, since I have no idea how it works. I may work with @rustweaver or whoever is interested (@ShapeShifter499 or @FMHemerli) in helping out with reading the report descriptors/data packets for those devices. (I just tried to digest some of the HID/USBHID spec, hope to put it to some use.)

Hopefully I'll have an update within the next two or three days with something new! Anyways, happy holidays everyone!!!

jlam55555, 2019-12-30:
Can someone with the gesture pad let me know the output of:

```
usbhid-dump -d 2feb
```

and

```
lsusb -vd 2feb:
```

jlam55555, 2020-01-06:
I've pushed the changes onto the [v2](https://github.com/jlam55555/veikk-s640-driver/tree/v2) branch, and will be primarily working there now -- will merge when stable. I've updated the sysfs interface and made the code a little more modular and extensible for future tablets (similar to what Wacom did, but with some differences, and still subject to change). Still have a lot to do (see the TODOs), but will be working on better documentation in the immediate future.

Again, for anyone with the gesture pad, let me know the output of the above commands.

nbogie, 2020-02-01:
> Can someone with the gesture pad let me know the output of...

Here's the output for both of those for the Veikk A50 which has the gesture pad.

[veikk-A50-usbhid-dump-and-lsusb.txt](https://github.com/jlam55555/veikk-linux-driver/files/4142293/veikk-A50-usbhid-dump-and-lsusb.txt)

Very grateful for your work and write-up, thanks!

profejuanca, 2020-06-08:
Hi, the config tool can select a particular display in a multiscreen configuration?

harish-srini, 2020-06-14:
Hi @jlam55555: 
First of all, thank you for providing the basic driver, works like a charm.
I have bought an A15 pro, when can we expect to get button and wheel functionalities included in the driver. 
I will be glad to help to give any inputs regarding the buttons and wheel!

ivancheng111, 2020-10-18:
@jlam55555 
Adding tablet area would be nice

asl97, 2020-11-05:
@jlam55555 did you know there is a S640v2?  I assume the S640v2 just have an updated firmware.

I contacted veikk Tech support and they helped update my S640, and now it support android.

I tested with this driver and it seem to still work the same.

:+1: for creating the driver

rudyon, 2021-04-23:
> Adding tablet area would be nice
This is currently the only reason one of my friends is still using windows. He would switch over if this was supported.

jlam55555, 2021-06-26:
Will be moving active development of the v3 driver (including button mapping) to a new repo.

See details [here](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-869042948).