jlam55555/veikk-linux-driver #5: Error compiling

kmstr113, 2019-06-20:
HI I am having an error compiling the driver. When I run make I get 
make -C /lib/modules/4.18.0-22-generic/build M=/home/jhahn/Documents/Veikk Driver/veikk-s640-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-22-generic'
arch/x86/Makefile:155: CONFIG_X86_X32 enabled but no binutils support
arch/x86/Makefile:244: *** You are building kernel with non-retpoline compiler, please update your compiler..  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-22-generic'
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 2

Any thoughts as to what I am doing wrong ?

jlam55555, 2019-06-21:
I'm not sure. Can you check your gcc version? (`gcc -v`) From posts like [this](https://github.com/NixOS/nixpkgs/issues/34383) it seems that it might be a problem with older gcc versions.

By the way, you might want to remove the space in your path (between "Veikk" and "Driver"). E.g., rename that directory to `Veikk_Driver` or something of the like. I haven't tested it, but it looks like the space might cause errors.

kmstr113, 2019-11-17:
Thank you I will try changing the folder name and see if that resolves it

closed 2020-01-06 by jlam55555

jlam55555, 2020-02-01:
Closed for inactivity.