jlam55555/veikk-linux-driver #8: Mouse3/middle mouse/button 2 gets released if the pen stays still

fakedeltatime, 2019-08-18:
Here's result from `xinput test`. When I do actually release the button after, there's no event.

```
motion a[0]=12850175 a[1]=9699327 a[2]=0 
motion a[0]=12843007 a[1]=9736703 a[2]=0 
button press   2 
motion a[0]=12836351 a[1]=9775103 a[2]=0 
motion a[0]=12830719 a[1]=9815039 a[2]=0 
motion a[0]=12830207 a[1]=9819135 a[2]=0 
motion a[0]=12831743 a[1]=9824255 a[2]=0 
motion a[0]=12835839 a[1]=9828863 a[2]=0 
motion a[0]=12843519 a[1]=9832959 a[2]=0 
motion a[0]=12855295 a[1]=9836543 a[2]=0 
motion a[0]=12870143 a[1]=9839103 a[2]=0 
motion a[0]=12889087 a[1]=9841151 a[2]=0 
motion a[0]=12911103 a[1]=9842175 a[2]=0 
motion a[0]=12937215 a[1]=9842687 a[2]=0 
motion a[0]=12966911 a[1]=9842687 a[2]=0 
motion a[0]=13001215 a[1]=9842687 a[2]=0 
motion a[0]=13038591 a[1]=9842687 a[2]=0 
motion a[0]=13079551 a[1]=9841151 a[2]=0 
motion a[0]=13124095 a[1]=9837567 a[2]=0 
motion a[0]=13171199 a[1]=9831423 a[2]=0 
motion a[0]=13221887 a[1]=9822207 a[2]=0 
motion a[0]=13275647 a[1]=9810431 a[2]=0 
motion a[0]=13330943 a[1]=9795583 a[2]=0 
motion a[0]=13387263 a[1]=9778687 a[2]=0 
motion a[0]=13445631 a[1]=9758719 a[2]=0 
motion a[0]=13505023 a[1]=9735679 a[2]=0 
motion a[0]=13565439 a[1]=9710591 a[2]=0 
motion a[0]=13626879 a[1]=9682943 a[2]=0 
motion a[0]=13686783 a[1]=9653247 a[2]=0 
motion a[0]=13746175 a[1]=9622527 a[2]=0 
motion a[0]=13804031 a[1]=9589247 a[2]=0 
motion a[0]=13860351 a[1]=9554943 a[2]=0 
motion a[0]=13915135 a[1]=9519615 a[2]=0 
motion a[0]=13968895 a[1]=9483263 a[2]=0 
motion a[0]=14020607 a[1]=9445887 a[2]=0 
motion a[0]=14070783 a[1]=9406463 a[2]=0 
motion a[0]=14119935 a[1]=9364991 a[2]=0 
motion a[0]=14167039 a[1]=9319935 a[2]=0 
motion a[0]=14213119 a[1]=9271807 a[2]=0 
motion a[0]=14257663 a[1]=9221119 a[2]=0 
motion a[0]=14299647 a[1]=9167359 a[2]=0 
motion a[0]=14339583 a[1]=9111551 a[2]=0 
motion a[0]=14377471 a[1]=9053695 a[2]=0 
motion a[0]=14412799 a[1]=8994815 a[2]=0 
motion a[0]=14445567 a[1]=8934399 a[2]=0 
motion a[0]=14476287 a[1]=8873471 a[2]=0 
motion a[0]=14503935 a[1]=8812031 a[2]=0 
motion a[0]=14529535 a[1]=8749567 a[2]=0 
motion a[0]=14552575 a[1]=8686079 a[2]=0 
motion a[0]=14573055 a[1]=8623615 a[2]=0 
motion a[0]=14591487 a[1]=8560639 a[2]=0 
motion a[0]=14607359 a[1]=8498687 a[2]=0 
motion a[0]=14620671 a[1]=8436735 a[2]=0 
motion a[0]=14631935 a[1]=8375807 a[2]=0 
motion a[0]=14641151 a[1]=8314879 a[2]=0 
motion a[0]=14648831 a[1]=8255999 a[2]=0 
motion a[0]=14654463 a[1]=8197631 a[2]=0 
motion a[0]=14657535 a[1]=8138751 a[2]=0 
motion a[0]=14659071 a[1]=8081407 a[2]=0 
motion a[0]=14659071 a[1]=8024063 a[2]=0 
motion a[0]=14659071 a[1]=7968255 a[2]=0 
motion a[0]=14657535 a[1]=7914495 a[2]=0 
motion a[0]=14654975 a[1]=7862271 a[2]=0 
motion a[0]=14650367 a[1]=7811071 a[2]=0 
motion a[0]=14643711 a[1]=7760895 a[2]=0 
motion a[0]=14635007 a[1]=7711231 a[2]=0 
motion a[0]=14625279 a[1]=7662591 a[2]=0 
motion a[0]=14614015 a[1]=7614463 a[2]=0 
motion a[0]=14601727 a[1]=7567359 a[2]=0 
motion a[0]=14588927 a[1]=7522815 a[2]=0 
motion a[0]=14575103 a[1]=7479807 a[2]=0 
motion a[0]=14560255 a[1]=7439359 a[2]=0 
motion a[0]=14544895 a[1]=7400447 a[2]=0 
motion a[0]=14529535 a[1]=7363071 a[2]=0 
motion a[0]=14513663 a[1]=7325695 a[2]=0 
motion a[0]=14497791 a[1]=7288319 a[2]=0 
motion a[0]=14481407 a[1]=7250431 a[2]=0 
motion a[0]=14463999 a[1]=7211007 a[2]=0 
motion a[0]=14445567 a[1]=7171071 a[2]=0 
motion a[0]=14425599 a[1]=7129599 a[2]=0 
motion a[0]=14405631 a[1]=7088127 a[2]=0 
motion a[0]=14385151 a[1]=7046655 a[2]=0 
motion a[0]=14364671 a[1]=7007743 a[2]=0 
motion a[0]=14343679 a[1]=6969855 a[2]=0 
motion a[0]=14322687 a[1]=6934015 a[2]=0 
motion a[0]=14301695 a[1]=6900223 a[2]=0 
motion a[0]=14280191 a[1]=6868479 a[2]=0 
motion a[0]=14258687 a[1]=6838271 a[2]=0 
motion a[0]=14236671 a[1]=6809599 a[2]=0 
motion a[0]=14214143 a[1]=6783487 a[2]=0 
motion a[0]=14191103 a[1]=6757887 a[2]=0 
motion a[0]=14167551 a[1]=6734335 a[2]=0 
motion a[0]=14143487 a[1]=6711807 a[2]=0 
motion a[0]=14119423 a[1]=6690815 a[2]=0 
motion a[0]=14095359 a[1]=6670847 a[2]=0 
motion a[0]=14071807 a[1]=6651903 a[2]=0 
motion a[0]=14049279 a[1]=6633471 a[2]=0 
motion a[0]=14027263 a[1]=6616575 a[2]=0 
motion a[0]=14005247 a[1]=6600191 a[2]=0 
motion a[0]=13983231 a[1]=6584831 a[2]=0 
motion a[0]=13962239 a[1]=6570495 a[2]=0 
motion a[0]=13941247 a[1]=6557695 a[2]=0 
motion a[0]=13920767 a[1]=6546431 a[2]=0 
motion a[0]=13900287 a[1]=6536703 a[2]=0 
motion a[0]=13879807 a[1]=6528511 a[2]=0 
motion a[0]=13858815 a[1]=6520831 a[2]=0 
motion a[0]=13837823 a[1]=6515199 a[2]=0 
motion a[0]=13818367 a[1]=6511615 a[2]=0 
motion a[0]=13799423 a[1]=6509567 a[2]=0 
motion a[0]=13782015 a[1]=6508543 a[2]=0 
motion a[0]=13766143 a[1]=6508543 a[2]=0 
motion a[0]=13752319 a[1]=6509567 a[2]=0 
motion a[0]=13740031 a[1]=6512127 a[2]=0 
motion a[0]=13729791 a[1]=6516223 a[2]=0 
motion a[0]=13721599 a[1]=6521855 a[2]=0 
motion a[0]=13715455 a[1]=6527999 a[2]=0 
motion a[0]=13710335 a[1]=6534143 a[2]=0 
motion a[0]=13707263 a[1]=6541311 a[2]=0 
motion a[0]=13705215 a[1]=6548991 a[2]=0 
motion a[0]=13703679 a[1]=6557183 a[2]=0 
motion a[0]=13702655 a[1]=6566399 a[2]=0 
motion a[0]=13702655 a[1]=6576639 a[2]=0 
motion a[0]=13702655 a[1]=6587391 a[2]=0 
motion a[0]=13703167 a[1]=6598655 a[2]=0 
motion a[0]=13704703 a[1]=6609407 a[2]=0 
motion a[0]=13706239 a[1]=6620159 a[2]=0 
motion a[0]=13709311 a[1]=6630911 a[2]=0 
motion a[0]=13712383 a[1]=6640127 a[2]=0 
motion a[0]=13716479 a[1]=6648831 a[2]=0 
motion a[0]=13720575 a[1]=6655487 a[2]=0 
motion a[0]=13724159 a[1]=6661119 a[2]=0 
motion a[0]=13728255 a[1]=6665215 a[2]=0 
motion a[0]=13732351 a[1]=6667775 a[2]=0 
motion a[0]=13735935 a[1]=6669311 a[2]=0 
motion a[0]=13738495 a[1]=6669823 a[2]=0 
motion a[0]=13741055 a[1]=6669823 a[2]=0 
motion a[0]=13743615 a[1]=6669823 a[2]=0 
motion a[0]=13745151 a[1]=6669823 a[2]=0 
motion a[0]=13746687 a[1]=6669823 a[2]=0 
motion a[0]=13747711 a[1]=6668799 a[2]=0 
motion a[0]=13748223 a[1]=6666751 a[2]=0 
motion a[0]=13748735 a[1]=6664191 a[2]=0 
motion a[0]=13748735 a[1]=6661119 a[2]=0 
motion a[0]=13748735 a[1]=6658559 a[2]=0 
motion a[0]=13747199 a[1]=6657023 a[2]=0 
button release 2 
motion a[0]=13746175 a[1]=6654975 a[2]=0 
motion a[0]=13745151 a[1]=6652927 a[2]=0 
motion a[0]=13745663 a[1]=6649343 a[2]=0 
motion a[0]=13747711 a[1]=6644735 a[2]=0 

```

fakedeltatime, 2019-08-19:
It seems like this happens with both of the extra buttons (both mouse2/right and mouse3/middle), I only noticed it with mouse3 at first.

jlam55555, 2020-01-06:
@fakedeltatime Sorry for only getting around to this now.

Is this a consistent issue? I'm a little confused by the title. Isn't the line that says `button release 2` indicating the button release? And this only happens if the pen stays still?

On a side note, v2 is out!

Nicole4740, 2020-04-06:
> @fakedeltatime Sorry for only getting around to this now.
> 
> Is this a consistent issue? I'm a little confused by the title. Isn't the line that says `button release 2` indicating the button release? And this only happens if the pen stays still?
> 
> On a side note, v2 is out!

Hello, since I'm not OP I can't quite confirm that `button release 2` is the release, though I would imagine that is, not sure if you're asking that or stating you believe it to be so, lol. I can confirm that I have this issue as well, whenever I'm trying to keep a steady hand to find the *just right* point of release when using the line tool, it will simply release anyway because my hand was too still. I cannot confirm anything for the left click button on my pen because I don't really use it. For the right mouse button, however, I've noticed that if I don't quickly tap it, it will exit out of the brush selection tool. I hope this helps!