jlam55555/veikk-linux-driver #9: VK1650 support

Sanaki, 2019-08-19:
I noticed you said you intend to support the vk1650 down the road. Assuming you find time to do so, I have one available and would be happy to test builds for you when that's ready.

azraxas, 2019-10-15:
im gonna try to make my 1560 work so ill keep you updated if i do figure it out!

jlam55555, 2020-01-06:
Hi @Sanaki and @azraxas. Any updates and/or success with the driver? It might just be that you need to add in the product id (try `lsusb` and look for `ID 2feb:XXXX`, where `XXXX` is the product id).

I don't know much about the 1560. It looks like it mirrors the screen via HDMI, and has an additional power plug and second USB connection to the computer. Is this correct?

I've just finished rewriting much of the driver from scratch and am willing to help you through issues and debugging if you have some time.

Sanaki, 2020-01-07:
I haven't tried. Figured I'd leave it to the experts first, since I have plenty of projects on my hands already. That said, your description seems accurate. It doesn't have to mirror the display, it can act as secondary. I can add this to my shortlist, but it'll probably be a week or so before I get to it.

jlam55555, 2020-02-07:
@Sanaki I'm not sure if I understood correctly -- did you mean to say 1560 as well? As far as I know, there is no 1650, but there is a 1560.

Either way, I've received a VK1560 from VEIKK to test, and added the ID (1001) to the ID table. Let me know if it works! No button support yet, just the stylus support like for the S640.

Sanaki, 2020-02-07:
Honestly, I think I just made the mistake of replying while tired. I believe you're correct, and I just had a typo in my first post. I'll test it out when I get home.

jlam55555, 2020-06-20:
Closing due to inactivity. VK1560 should have basically functionality (screen works, buttons not yet).

closed 2020-06-20 by jlam55555

Sanaki, 2021-06-27:
Sorry for how long I stepped away from this. Finally got around to testing it and I'm not getting any stylus input at the moment. I'm still testing to see if there could be something awry specifically on my end, but so far it just seems nonfunctional.

jlam55555, 2021-06-27:
I'm not familiar with `ehvz`, but one thing is very concerning: Are you sure you don't have the UC-Logic 1600 Pen? That doesn't appear to be a VEIKK device. Did you buy from a shady seller?

Unless VEIKK used the UC-Logic controller internally at some point in their earlier devices, I don't think this is the fault of this driver. Let me know your thoughts on this. (A picture of the tablet might be helpful to confirm/deny this theory.)

Sanaki, 2021-06-27:
The device looks identical to all images I can find of the VK1560, and is labeled on the rear as being a VK1560. All branding (front under "glass", rear on the label) says Veikk. It was purchased from amazon via Veikk directly (https://www.amazon.com/dp/B07DJ9NGNV/) on 2019-01-19.

As for evhz, it's a handy little tool from here, found via the arch wiki:
https://git.sr.ht/~iank/evhz

In my case I'm on Linux Mint 19.3, but it works fine for me as well.

jlam55555, 2021-06-27:
Hmm, I don't know then. You can try adding the device to the ID table (use the values `idVendor=5543, idProduct=004d`) and see if it has the same interface.

Sanaki, 2021-06-28:
Looks like it does mostly. I reassigned VEIKK_PEN_REPORT to id 9 rather than 1 to get it to work (was getting unhandled input on ID 9 in dmesg before that), and I changed the vendor ID entirely since I'm uncertain how to gracefully add a second vendor ID. I had to redo the mapping as follows to get it approximately correct (currently it's my second monitor to the right of my primary):
[image]
At this point it seems to just work, though obviously I'd prefer not to have a butchered driver for the purpose. Still, my experience in this realm is limited, so butchery is all I can really do at the moment. If you can clean it up to an actual publish-worthy state, that'd be great. I did verify pressure and the buttons register correctly via Krita.

The two added sections of `veikk_vdev.c` were as follows (in case anyone else stumbles across this before it's available):
```c
struct veikk_device_info veikk_device_info_0x004d = {
    .name = "UC-Logic 1600 Pen", .prod_id = 0x004d,
    .x_max = 27536, .y_max = 15489, .pressure_max = 8192,
    .setup_and_register_input_devs = veikk_s640_setup_and_register_input_devs,
    .alloc_input_devs = veikk_s640_alloc_input_devs,
    .handle_raw_data = veikk_s640_handle_raw_data,
    .handle_modparm_change = veikk_s640_handle_modparm_change
};

{ VEIKK_DEVICE(0x004d) },   // UC1600
```
plus the modifications to vendor and pen_report in `veikk.h`.

I assume I could fix the mapping directly in that block, but I'm uncertain of the scaling. I'm happy to test any suggestions though. I did notice the display reports itself to the OS as `DO NOT USE - RTK 24"`, so maybe I got a prototype batch or something.

jlam55555, 2021-06-28:
Sounds good! I'm in the process of rewriting the driver (once more, and hopefully for the last time) for the v3 version, and I'll try to add that in.

Sanaki, 2021-06-28:
Just a quick bit extra, I disabled my primary display and pulled values for both landscape and portrait that appear to align correctly (might be off by a few pixels, the display isn't really -that- precise). Can you provide any insight into any adjustment I might be able to make to that struct to get this stage a bit less arbitrary? Assuming I'm correctly interpreting those values anyway. Perhaps this just is what it is.
[image]

[image]

jlam55555, 2021-07-02:
@Sanaki I don't remember the exact calculation I had for the mapping there. It is just what it is, sorry.

...BUT!

In the new v3 version I'm working on, I just got multi-monitor support to work a few minutes ago, where you can drag-select an area of the screen to map to.

Almost done with the v3 version. It got a lot more elaborate than initially planned. Will pretty much be done once I get dbus integrated. (Hopefully early next week.)

[Relevant file](https://github.com/jlam55555/veikk-config/blob/dev/veikk/cli/coordinate_mapping/screen_area_grabber.py) -- not really integrated into the program yet, but you can see that it grabs all the monitors.