jlam55555/veikk-linux-driver #12: Error in compiling (incompatible gcc/plugin versions)

iamashwin26, 2019-11-10:
I have Manjaro 18.10 and i tried installing AUR package but failed to do so. here's what the package manager prints
`Cloning input-veikk-dkms build files...
Checking input-veikk-dkms dependencies...
Preparing...

Building input-veikk-dkms...
==> Making package: input-veikk-dkms r42.0359cc1-2 (Sunday 10 November 2019 10:27:51 AM IST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating input-veikk git repo...
Fetching origin
  -> Found veikk.conf
  -> Found dkms.conf
==> Validating source files with sha1sums...
    input-veikk ... Skipped
    veikk.conf ... Passed
    dkms.conf ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of input-veikk git repo...
Cloning into 'input-veikk'...
done.
==> Starting pkgver()...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "input-veikk-dkms"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: input-veikk-dkms r42.0359cc1-2 (Sunday 10 November 2019 10:27:55 AM IST)
==> Cleaning up...

Preparing...
Resolving dependencies...
Checking inter-conflicts...
Running post-transaction hooks...
Checking keyring...
Checking integrity...
Loading packages files...
Checking file conflicts...
Checking available disk space...
Installing input-veikk-dkms (r42.0359cc1-2)...
Running post-transaction hooks...
Install DKMS modules
==> dkms install input-veikk/r42.0359cc1 -k 5.2.21-1-MANJARO
Error! Bad return status for module build on kernel: 5.2.21-1-MANJARO (x86_64)
Consult /var/lib/dkms/input-veikk/r42.0359cc1/build/make.log for more information.
Arming ConditionNeedsUpdate...
Transaction successfully finished.

`

The make log is as follows :
`DKMS make.log for input-veikk-r42.0359cc1 for kernel 5.2.21-1-MANJARO (x86_64)
Sun Nov 10 10:27:57 IST 2019
make: Entering directory '/usr/lib/modules/5.2.21-1-MANJARO/build'
  CC [M]  /var/lib/dkms/input-veikk/r42.0359cc1/build/veikk.o
cc1: error: incompatible gcc/plugin versions
cc1: error: fail to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[1]: *** [scripts/Makefile.build:285: /var/lib/dkms/input-veikk/r42.0359cc1/build/veikk.o] Error 1
make: *** [Makefile:1597: _module_/var/lib/dkms/input-veikk/r42.0359cc1/build] Error 2
make: Leaving directory '/usr/lib/modules/5.2.21-1-MANJARO/build'
`

Initially, I had gcc9.2 and I was getting the same error, but on Kubuntu 19.10 which has gcc 8 I was able to compile successfully so I compile gcc7 on majaro and changed the default to gcc7 but still I get the same error.  Am I doing something wrong?

jlam55555, 2020-01-07:
@iamashwin26 Any updates?

If you haven't solved it yet, I see a [similar issue open on the archlinux forums](https://bugs.archlinux.org/task/63644) and [one in the manjaro forums](https://forum.manjaro.org/t/kernel-5-3-and-5-2-rtl8723de-fail-to-build/106139). According to the latter, it's likely something about your gcc version not being the one expected by (compiled with) your kernel. It's probably that you upgraded your gcc to 9.2 sometime in the past, but I'm not sure how to check the default version your kernel was compiled with.

iamashwin99, 2020-01-10:
@jlam55555  you are right, the problem was because the kernel was compiled with a different version of GCC than the one installed. I solved the problem by updating the kernel with one compiled from the latest GCC.

closed 2020-01-11 by jlam55555