jlam55555/veikk-linux-driver #14: LED support/extra devices shown with `lsusb`?

jlam55555, 2020-01-07:
Request for help from the developer: I'm not really sure why Veikk registers three interfaces (?) when plugging in the USB. If you've looked at `dmesg`'s output when using this driver, you can clearly see that "VEIKK S640 probed successfully/removed" shows up three times (this is for the S640 at least -- not sure about other models).

Running `usbhid-dump -d 2feb` as root, I get:

```
002:009:002:DESCRIPTOR         1578425202.646505
 06 0A FF 09 01 A1 01 85 09 09 02 75 08 95 08 15
 00 26 FF 00 81 02 09 03 75 08 95 08 15 00 26 FF
 00 91 02 C0

002:009:001:DESCRIPTOR         1578425202.646749
 05 01 09 06 A1 01 85 03 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 07 19 00 29 FF 26 FF
 00 75 08 95 06 81 00 C0

002:009:000:DESCRIPTOR         1578425202.647238
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 03 15 00 25 01 95 03 75 01 81 02 95 05 81 01
 05 01 09 30 09 31 26 FF 7F 95 02 75 10 81 02 05
 0D 09 30 26 FF 1F 95 01 75 10 81 02 C0 C0 05 0D
 09 02 A1 01 85 02 09 20 A1 00 09 42 09 44 09 45
 15 00 25 01 75 01 95 03 81 02 95 02 81 03 09 32
 95 01 81 02 95 02 81 03 75 10 95 01 35 00 A4 05
 01 09 30 65 13 55 0D 46 34 03 27 10 77 00 00 81
 02 09 31 46 4D 04 27 60 4F 00 00 81 02 B4 09 30
 45 00 26 FF 1F 81 02 C0 C0
```

You can plug each of these into an [HID parser](https://eleccelerator.com/usbdescreqparser/) to see what they mean.

The latter device's first usage (002:009:000, mouse usage) is what's been emitting all of the input reports I've been using -- if you uncomment the `.report = veikk_report` in `veikk_drv.c` you can see the raw report formats and IDs. For me with the S640, all reports have ID 1 and have the same structure (the structure is illustrated with the `struct veikk_pen_report`, which I cast the data to effortlessly). This data packet includes the three buttons available on the S640, X/Y position, and pressure sensitivity -- everything the S640 can output.

This makes me wonder what the other two devices are for, as well as the second usage (digitizer) in the third device. [This line of @gromgull's PR](https://github.com/jlam55555/veikk-linux-driver/pull/10/commits/7591bd238c2462b6fadbf2b4aabcbf17e3b9bc5f#diff-55ac121a0b5bbbfc091163ff9f5a6af8R71) suggests that the digitizer usage for the third device is used for extra buttons for the A50 at least (something I have yet to confirm), but then what about the other devices?

From the HID parser, the first (002:009:002) has a vendor-defined usage page (not helpful to me here), and has a 64 bit (8 bits per report and 8 reports) input and output report descriptors defined with report ID=9. I thought this might be an LED (the other descriptors don't have an output described, and I'm not sure what else could be both an input or output), but I have no idea about the structure of these report descriptors. I tried sending an output report using `hid_hw_output_report` with a report of structure `0x09ffffffffffffffff` (report id=9 and a bunch of 1's) to see what would happen but I got nothing to happen. Again, I never get any input reports from this either.

The second device (002:009:001) has a keyboard usage page. I'm not really sure how to interpret the structure of it or what it could imply. Again, the S640 being the simplest VEIKK tablet, there's no keyboard or key-like structures other than the two buttons and stylus tip switch, so maybe it's only for other VEIKK devices?

Any thoughts? Are these other usages likely to be placeholders, used only for the more complex VEIKK devices? Is this common in industry? Given how quickly VEIKK has been churning out new devices, I wouldn't be too surprised if this were the case and they were using the same HID interface/descriptors with placeholders left for future devices.

(I've peeked at Wacom's driver for LEDs, but it's more complex and less necessary than other things I'd like to tackle at the moment.)

I might also post this on Stack Overflow at some point, but I think it is quite domain-specific and will just leave it here for now.

Link1J, 2020-01-30:
So as some help, here is `usbhid-dump -d 2feb` for the A15.
```
001:006:002:DESCRIPTOR         1580411020.609805
 06 0A FF 09 01 A1 01 85 09 09 02 75 08 95 08 15
 00 26 FF 00 81 02 09 03 75 08 95 08 15 00 26 FF
 00 91 02 C0

001:006:001:DESCRIPTOR         1580411020.610591
 05 0D 09 02 A1 01 85 02 09 20 A1 00 09 42 09 44
 09 45 09 3C 09 43 09 44 15 00 25 01 75 01 95 06
 81 02 09 32 75 01 95 01 81 02 95 01 81 03 05 01
 09 30 09 31 55 0D 65 33 26 FF 7F 35 00 46 40 1F
 75 10 95 02 81 02 05 0D 09 30 26 FF 1F 75 10 95
 01 81 02 C0 C0 05 01 09 06 A1 01 85 03 05 07 19
 E0 29 E7 15 00 25 01 75 01 95 08 81 02 05 07 19
 00 29 FF 26 FF 00 75 08 95 06 81 00 C0

001:006:000:DESCRIPTOR         1580411020.612007
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 03 15 00 25 01 95 03 75 01 81 02 95 05 81 01
 05 01 09 30 09 31 26 FF 7F 95 02 75 10 81 02 05
 0D 09 30 26 FF 1F 95 01 75 10 81 02 C0 C0
```

It looks like you are right about them using the same HID interface/descriptors, as `001:006:000` reports the pen and pen buttons, with `001:006:001` reports the buttons on the unit itself.
When the driver sets up the device, it mentions in the system log the three devices as, in order, a mouse, a keyboard, and a device. The mouse is the pen, the keyboard is the tablet's buttons, and I don't know what the "device" is.

Hopeful this is of some help.

Link1J, 2020-01-30:
Also the buttons report as a keyboard, because they act as such. Report number 3 on the buttons should be handled in the same way a keyboard is.

jlam55555, 2020-02-01:
@Link1J Thanks for your input! A little busy for now but when I have time I'll review this and @nbogie's dump and see if I can pick anything useful out. I guess we'll still be wondering what the "device" is for some time...

If you'd like to, feel free to PR with adding functionality to the report ID of 3 -- I have no idea how the button data is structured there. It looks like @gromgull got some progress [here](https://github.com/jlam55555/veikk-linux-driver/pull/10) in v1 of the driver -- you can use that for inspiration. In v2, it would involving creating new register and data handling fns for the A15.

jlam55555, 2021-06-26:
Closing because I figured out what these are for v3 of the driver. These other HID report are important for button mappings. See the mailing list/improvements issue. v3 of the driver will be moved to a new repo because it is sufficiently different from v1 and v2.

closed 2021-06-26 by jlam55555