jlam55555/veikk-linux-driver #16: Getting modprobe: FATAL: Module veikk not found in directory when attempting to install

PopMyCorn, 2020-01-17:
Hello!

After entering:
> sudo make all install clean

Into the terminal, I get the following:

> make -C /lib/modules/5.3.0-26-generic/build M=/home/[username]/Downloads/veikk-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-26-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-26-generic'
make -C /lib/modules/5.3.0-26-generic/build M=/home/[username]/Downloads/veikk-linux-driver-master modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-26-generic'
  INSTALL /home/[username]/Downloads/veikk-linux-driver-master/veikk.ko
At main.c:160:
> SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
> SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.3.0-26-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-26-generic'
modprobe veikk
modprobe: FATAL: Module veikk not found in directory /lib/modules/5.3.0-26-generic
Makefile:14: recipe for target 'install' failed
make: *** [install] Error 1

The drivers fail to install. I have secure boot disabled and I _believe_ all of the appropriate headers are installed (I've checked and made sure they're up to date). I'm not quite sure what's going on, I don't _think_ it has to do with secure boot, though I might be wrong. I have that disabled in the bios according to ye olde instruction manual. And if it matters, I'm using the latest Zorin OS Core, a relatively fresh install that I did yesterday.

Thanks for any info!

jlam55555, 2020-01-18:
Yeah, doesn't look like the Secure Boot error. The abnormal line is

> Warning: modules_install: missing 'System.map' file. Skipping depmod.

As [this Stack Overflow post suggests](https://stackoverflow.com/questions/57521081/warning-modules-install-missing-system-map-file-skipping-depmod), you can try running `depmod` manually after `make modules_install` (it's usually called during `modules_install`). `depmod` is the tool to update the list of known drivers, so without calling it, `modprobe` doesn't know anything about this driver.

Try the following steps instead and let me know what happens:

```bash
$ sudo make all install && depmod && modprobe veikk
$ modinfo veikk
```

jlam55555, 2020-01-18:
Actually, from reading a little more into the issue, I don't know if that'll solve the problem. Looks like `depmod` will silently fail even if you run it again.

Is the location where you cloned this repo and `/usr/src/` on different partitions? [This](https://forums.gentoo.org/viewtopic-t-423888-start-0.html) seems to imply that that may cause this problem to appear.

PopMyCorn, 2020-01-18:
That might actually be it. I just cloned it into the Downloads folder in
Home. To be sure, should I trying cloning the repo into the src folder
itself and then try installing from there? You'll need to excuse any inane
questions I'm ask, I'm sorry. I haven't used Linux in quite some time and
have only just made the full switch after win7 support was ended.

jlam55555, 2020-01-18:
I'm not a fan of separating the home directory in a different partition (and the rest of the system in a second partition), but if that is the case for you (and if this hypothesis is correct), then cloning anywhere outside of the `/home` directory should work. I'm not too sure about what the convention is with `/usr/src/`, but if it doesn't get the kernel source mixed up it'll probably be fine.

No need to be sorry, these aren't silly q's.

PopMyCorn, 2020-01-18:
Well, I tried putting it outside of the home directory and I received the
same error as in the OP so now I'm quitequite stumped.
As for the home directory being partitioned as it is, to be perfectly
honest, I just followed the default installation set up for Zorin and let
it do its thing. I was prepared to do a fresh reinstall since I hadn't even
finished moving into it but now I'm not sure if that's the problem to begin
with. I wonder what it could be.

jlam55555, 2020-01-19:
Reinstall may or may not fix the issue -- might be better to try to avoid that for now to save you time as well.

[Looks like there's a script dedicated to generate System.map.](https://classicforum.manjaro.org/index.php?topic=22364.0). It's for Manjaro, but hopefully it's also on Ubuntu/Zorin OS.

PopMyCorn, 2020-01-19:
I'll give it a try when I'm at my pc, thanks!

PopMyCorn, 2020-01-19:
So I'm looking at the script and I'm wondering where I should run it from.
In my usr/lib folder, there's no module folder. Would I just run this from
wherever the latest generic headers are installed?

jlam55555, 2020-01-19:
See if something like [this](https://askubuntu.com/a/554734/433872) works? (This is in this repo's README as well)

PopMyCorn, 2020-01-20:
I tired the apt-get and received:

Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-generic is already the newest version (4.15.0.74.76).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.

I figured I would try the answer below to see where essentials are, which
was this command, which I ran as sudo:

sudo find / | grep include/generated/autoconf.h

And that returned

find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
find: ‘/run/user/121/gvfs’: Permission denied
/usr/src/linux-headers-5.0.0-37-generic/include/generated/autoconf.h
/usr/src/linux-headers-5.3.0-26-generic/include/generated/autoconf.h
/usr/src/linux-headers-4.15.0-74-generic/include/generated/autoconf.h

So now I'm not quite sure how to proceed or what's even wrong.

jlam55555, 2020-01-20:
Ok, that does look like what the answerer had (maybe that's how it's structured on Debian-like systems? I haven't used it in a while so not sure). Try changing the build dir in the Makefile like the answer suggested, from:

    BUILD_DIR := /lib/modules/$(shell uname -r)/build

to

    BUILD_DIR := /usr/src/linux-headers-$(shell uname -r)

If that works for you, I'll have to investigate and figure out how to make this work for Debian-like systems.

PopMyCorn, 2020-01-20:
Would that be the Makefile in the repo I'm changing?

PopMyCorn, 2020-01-20:
Ok, so I tried that, in both places I cloned the repo, and tried to install
and I still get the same errors. I also tried the Manjaro mksysmap script
(edited for my filepaths) and that actually didn't even do anything, so I'm
still lacking a System.map file it seems. I'm looking up how generate one
on ubuntu/zorin.

YassienW, 2020-01-24:
Facing the same exact problem on Ubuntu, please let us know if you got it fixed.

PopMyCorn, 2020-01-24:
Ohhh, good to know I'm not alone lol. No progress with Zorin but I just
changed to Solus OS a couple of days ago and I'm going to try later today.
If I get the same problem again, I'll outline what I did to fix it there.
Granted, Solus is mostly its own project so I don't know how well things
would translate from there to Ubuntu or other Debian based distros.

PopMyCorn, 2020-01-26:
So, an update, with good news:

On Solus, installation went flawlessly and I just tested it out on Krita,
spent a solid 2 hours working on a painting without any issue, and now I
can finally really say goodbye windows for good. And in case there's any
Solus users wondering why they're getting "bash: make command not found"
like I was, it's because you need to install a package from the Solus repo
that contains that and other commands. They call it the developer tools
pack, the command to install it is right on the Solus website.

So everything worked out for me just fine but it seems that this is maybe
an issue with Debian based distros? Zorin, Elementary, and Pop! all gave me
the same issue and apparently this has happened to someone else on Ubuntu.
I haven't tried on anything Arch based, mainly because the Manjaro
installer kept freezing on me, so I don't know about those. I also don't
know if I can say this is an issue with this project itself. With Zorin,
the main issue was that there was no System.map and I got the same thing on
Elementary and Pop. Paths for some build essentials and headers were also
different on each of those distros so it may be a case of the user just
needs to make sure they have a system map and edit the build path in the
make file to correspond to theirs on a Debian distro.

jlam55555, 2020-01-26:
@PopMyCorn That's good to hear. Seems like it is a Debian (Debian-like) issue. I don't have access to the tablet while I'm away for school, but I'll spin up a VM and see what I can do with testing out installation.

PopMyCorn, 2020-01-26:
Oh, you can also add me to the list of cases where your driver is confirmed
to work on the Veikk A50 if that's still something you need to keep tabs
on.

>

Nicole4740, 2020-02-02:
Hi all,
I noticed that the bug was titled Debian & derivatives, but I'd just like to share a piece of information.
I'm running Pop!_OS which is based off of Ubuntu (which is based off Debian,) so theoretically, I should have this same issue, but atleast for me, I found a work around. When I first tried installing, I got the same error as OP where it said the System.map file was missing, however I dont have linux-headers-5.3.0-26-generic, instead, the file where it was trying to find veikk.mod was titled 5.3.0-7625-generic. After running `sudo depmod` and `sudo modprobe veikk`, I was able to install it normally and it works just fine. Not sure if this is helpful at all, but I thought if someone can figure out why this works on Pop!_OS but not other ubuntu distros, maybe that would help in solving the problem for everyone.

Diamondop1234, 2020-02-08:
> Hi all,
> I noticed that the bug was titled Debian & derivatives, but I'd just like to share a piece of information.
> I'm running Pop!_OS which is based off of Ubuntu (which is based off Debian,) so [...]

This fixed it for me, now my A30 is working flawlessly with pressure on Pop!_OS.

jlam55555, 2020-02-08:
@PopMyCorn If you ever switch back to a Debian-based system, please update. I'll close this in a few days and make a wiki of common issues, but if what Nicole says doesn't fix it for you, please to reopen this. I've still yet to confirm anything on Debian myself.

PopMyCorn, 2020-02-08:
I don't think I will be, no. I've settled into Solus as my main os on my
desktop and Arch on another machine.

closed 2020-02-18 by jlam55555

YassienW, 2020-02-19:
[What @Nicole4740 recommended](https://github.com/jlam55555/veikk-linux-driver/issues/16#issuecomment-581177802) fixed the issue on Ubuntu. I wanna point out that you might need to disable secure boot to be able to run `sudo modprobe veikk`.

kl-31, 2020-06-02:
Unfortunately Nicole's fix didn't work for me on Ubuntu 18.04, kernel 5.3.0-53-generic. Same issue with the missing System.map. Any ideas would be very much appreciated.

make -C /usr/src/linux-headers-5.3.0-53-generic M=/home/liangkc/veikk-linux-driver modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-53-generic'
cat: /home/liangkc/veikk-linux-driver/modules.order: No such file or directory
  DEPMOD  5.3.0-53-generic
**Warning: modules_install: missing 'System.map' file. Skipping depmod.**
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-53-generic'
depmod
modprobe veikk
mkdir -p /etc/modules-load.d
echo "veikk" > /etc/modules-load.d/veikk.conf
make -C /usr/src/linux-headers-5.3.0-53-generic M=/home/liangkc/veikk-linux-driver clean
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-53-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-53-generic'

prestonsg, 2020-06-18:
First off, jlam55555, thank you for creating the veikk driver and sharing it with the community! 

Unfortunately, I am suffering the same challenge as several other Ubuntu derivative users and Nicole's fix didn't work for me either.  (It looks as if version 2 added the depmod and modprobe steps in the make file anyway.  By the way, I'm on Mint.   Kernel: 4.15.0-88-generic
Attempting to use a Veikk A15pro (it's the A15 with a scroll wheel)

I share the following in the hope that it may be of some value, and help lead to a solution.

When I attempt to install the linux veikk driver, I get the SSL error "No such file or directory: bss_file" that a couple of other folks reported.

My search for a potential solution lead to a brief thread by a user with the same challenge:
[ automatic-module-signing-for-distribution-in-linux](https://unix.stackexchange.com/questions/524683/automatic-module-signing-for-distribution-in-linux)
Which lead to this promising solution:
[no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules](https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules)
Happily, make install didn't result in the SSL error, but it didn't seem to correctly sign and load the driver either.  

Here's the output of "modinfo veikk" 
```
filename:       /lib/modules/4.15.0-88-generic/extra/veikk.ko
license:        GPL
description:    USB VEIKK drawing tablet driver
author:         Jonathan Lam <jlam55555@gmail.com>
version:        2.0
srcversion:     EF50A3B5389DE9F8DC2F0B7
alias:          hid:b0003g*v00002FEBp00001001
alias:          hid:b0003g*v00002FEBp00000006
alias:          hid:b0003g*v00002FEBp00000004
alias:          hid:b0003g*v00002FEBp00000003
alias:          hid:b0003g*v00002FEBp00000002
alias:          hid:b0003g*v00002FEBp00000001
depends:        hid
retpoline:      Y
name:           veikk
vermagic:       4.15.0-88-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4

```
Thanks for any guidance on other steps I should attempt in the quest to find a resolution.

prestonsg, 2020-06-19:
I am happy to report that I was able to get the driver installed.  I upgraded to the latest kernel/headers: 4.15.0-106-generic 

Then generated the signing key using the following: 
$: cd /usr/src/linux-headers-$(uname -r)/certs
$: sudo openssl req -new -x509 -newkey rsa:2048 -keyout signing_key.pem -outform DER -out signing_key.x509 -nodes -subj "/CN=Owner/"

Finally, issued the make install command.  It works.  Yay!
Again, thanks for creating this driver.