jlam55555/veikk-linux-driver #18: im having a bit of trouble with my tablet on krita

Trashy413, 2020-01-25:
My tablet works fine responds and all but when i try and draw on krita it doesnt move the cursor on the art but responds on everything else. It would be great if some one could give me a bit more detail and explain what i need to do to make it work.(PS im on a chromebook so that might also be a factor. Im not that good with code so could someone help me with the code i need to change to fix my issue?

jlam55555, 2020-01-26:
@Trashy413 Honestly, never messed around with a Chromebook, so I have little chance of reproducing your problem. A few questions:

- Your Chromebook is running Linux, correct? If so, which distro/version and kernel version? (Also, which Chromebook? Probably won't be helpful, but who knows.)
- Which tablet? Is its ID in the ID table? (S640, A30, A50, A15)
- Does the tablet work without the driver? (I believe newer kernel versions have some basic support OOTB for tablets, but not pressure sensitivity.) If so, how does the functionality before using the driver and after using the driver compare?
- Does the tablet work on other drawing utilities (e.g., GIMP) with/without the driver?
- Output of `lsmod | grep veikk`?
- Can you give a little more detail on "responds on everything else"?
- Wayland or X11?
- Any errors with build/installation? Any relevant errors in `dmesg`? Does tablet probing show up in `dmesg`?

Also, if on X11, mess around with `xinput; xinput test [device-id]` and `sudo evtest` to look at what events are being generated. Are X, Y, and pressure events being generated?

Let me know if anything is unclear.

jlam55555, 2020-06-20:
Closing due to inactivity.

closed 2020-06-20 by jlam55555