jlam55555/veikk-linux-driver #21: Module veikk not found

ZN0N15, 2020-02-01:
I seem to be having an issue when installing the driver. I keep getting make error: 1 when doing `sudo make all install clean`.

Here is the full log
> make -C /lib/modules/5.3.0-28-generic/build M=/home/zn0n/Documents/veikk-linux-driver modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-28-generic'
  INSTALL /home/zn0n/Documents/veikk-linux-driver/veikk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.3.0-28-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-28-generic'
modprobe veikk
modprobe: FATAL: Module veikk not found in directory /lib/modules/5.3.0-28-generic
Makefile:14: recipe for target 'install' failed
make: *** [install] Error 1 

thanks in advance

jlam55555, 2020-02-01:
@ZN0N15 Closing this as a duplicate of [this issue](https://github.com/jlam55555/veikk-linux-driver/issues/17), which seems to be manifesting itself on Debian-like systems. Let's keep the discussion over there. Unfortunately, I'm a bit busy with school and haven't been able to test this out yet, but hopefully when I have time I'll see what I can do.

That is, I'm assuming you are on a Debian derivative (see the discussion in the other issue). If not, reopen this issue with more details.

closed 2020-02-01 by jlam55555

YaniBeanOfficial, 2021-05-15:
Im having the exact same issue but its giving me 

make -C /lib/modules/5.4.99-12983-g8b7876ab9f5e/build M=/home/rozuokami/veikk-linux-driver modules
make[1]: *** /lib/modules/5.4.99-12983-g8b7876ab9f5e/build: No such file or directory.  Stop.
make: *** [Makefile:8: all] Error 2

I've tried many different commands to fix it and nothing has worked so far