jlam55555/veikk-linux-driver #23: Bad mapping with A50

KibaGasteiz, 2020-02-15:
It doesn't matter what screen resolution is used (1080p,720p or SVGA), the device only lets me to move at the top left part of the screen, i can go as down as to the middle of the screen, and as right as 2/3 of the screen (more or less).

jlam55555, 2020-02-18:
The A50 is working perfectly fine for me, so here are two things that might help get started:

- When you run `sudo evtest` and move the mouse around, what is the maximum value for X and Y? It should be around 32768 for both X and Y for the A50.
- What happens when you try to manually force a different screen mapping with the GUI (make sure to run as root)? E.g., if it goes about 2/3 of the way on a 1920x1080 screen, try setting the screen mapping to go 1.5*1920=2880 width, 1.5*1080=1620 height. What effect does this produce?

KibaGasteiz, 2020-02-23:
When running 'sudo evtest' with the mouse (Logitech G502 Hero) i don't get those values, i just get this:
> Event: time 1582455858.445786, type 2 (EV_REL), code 1 (REL_Y), value -1
> Event: time 1582455858.445786, -------------- SYN_REPORT ------------
> Event: time 1582455858.446783, type 2 (EV_REL), code 0 (REL_X), value -1
> Event: time 1582455858.446783, -------------- SYN_REPORT ------------
> Event: time 1582455858.456783, type 2 (EV_REL), code 1 (REL_Y), value -1
> Event: time 1582455858.456783, -------------- SYN_REPORT ------------
> Event: time 1582455858.477784, type 2 (EV_REL), code 1 (REL_Y), value -1
> Event: time 1582455858.477784, -------------- SYN_REPORT ------------
> Event: time 1582455858.494787, type 2 (EV_REL), code 1 (REL_Y), value -1
> Event: time 1582455858.494787, -------------- SYN_REPORT ------------
> Event: time 1582455858.511782, type 2 (EV_REL), code 1 (REL_Y), value -1
> Event: time 1582455858.511782, -------------- SYN_REPORT ------------
> Event: time 1582455858.514784, type 2 (EV_REL), code 0 (REL_X), value -1
> Event: time 1582455858.514784, -------------- SYN_REPORT ------------
> Event: time 1582455858.532781, type 2 (EV_REL), code 1 (REL_Y), value -1

With the table i got this values at the bottom right part of the device:
> Event: time 1582456059.519189, type 3 (EV_ABS), code 0 (ABS_X), value 25131
> Event: time 1582456059.519189, type 3 (EV_ABS), code 1 (ABS_Y), value 15222

Can't install Veikk Config GUI by the way:

> Construyendo input-veikk-config...
> ==> Creando el paquete: input-veikk-config 1.1-1 (dom 23 feb 2020 12:09:26)
> ==> Comprobando dependencias mientras se ejecuta...
> ==> Comprobando dependencias mientras se compila...
> ==> Recibiendo las fuentes...
>   -> README ha sido encontrado 
>   -> Descargando input-veikk-config.sh...
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> curl: (22) The requested URL returned error: 404 Not Found
> ==> ERROR: Hubo fallos durante la descarga de https://raw.githubusercontent.com/jlam55555/veikk-s640-driver/master/config.sh
>     Cancelando...
> Failed to build input-veikk-config

jlam55555, 2020-02-23:
Sorry, I meant to say tablet cursor, not mouse. As a quick fix, you can try setting `.x_max = 25131, .y_max=15222` in `veikk_device_info_0x0003` in `veikk_vdev.c`.

This roughly seems to match [@lenonk's `usbhid-dump`](https://github.com/jlam55555/veikk-linux-driver/issues/17#issuecomment-582253336), which states the following maximum values:

    0x27, 0x38, 0x63, 0x00, 0x00,  //       Logical Maximum (25399)
    0x27, 0x88, 0x3B, 0x00, 0x00,  //       Logical Maximum (15239)

Interestingly, however, I get a different value than he does: for the A50 what I get is:

```
001:008:002:DESCRIPTOR         1582473461.015513
 06 0A FF 09 01 A1 01 85 09 09 02 75 08 95 08 15
 00 26 FF 00 81 02 09 03 75 08 95 08 15 00 26 FF
 00 91 02 C0

001:008:001:DESCRIPTOR         1582473461.015703
 05 01 09 06 A1 01 85 03 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 07 19 00 29 FF 26 FF
 00 75 08 95 06 81 00 C0

001:008:000:DESCRIPTOR         1582473461.016133
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 03 15 00 25 01 95 03 75 01 81 02 95 05 81 01
 05 01 09 30 09 31 26 FF 7F 95 02 75 10 81 02 05
 0D 09 30 26 FF 1F 95 01 75 10 81 02 C0 C0 05 0D
 09 02 A1 01 85 02 09 20 A1 00 09 42 09 44 09 45
 15 00 25 01 75 01 95 03 81 02 95 02 81 03 09 32
 95 01 81 02 95 02 81 03 75 10 95 01 35 00 A4 05
 01 09 30 65 13 55 0D 46 34 03 27 70 C6 00 00 81
 02 09 31 46 4D 04 27 10 77 00 00 81 02 B4 09 30
 45 00 26 FF 1F 81 02 C0 C0
```

This is giving me headaches because it seems to be using input report ID=2 without the driver plugged in (using the `usbhid` module), and input report ID=1 with the driver installed, and these have different maximum X and Y values.

Do you mind posting the output of `usbhid-dump` on your system? I'm assuming yours will match lenon's. I'll see if I can ask VEIKK about this discrepancy.

---

Also, the config tool AUR is outdated as you've realized. Install it from [the GitHub](https://github.com/jlam55555/veikk-linux-driver-gui) for now; if I have time I'll make a new AUR package.

KibaGasteiz, 2020-04-08:
Sorry, i haven't got much time, this is what i get:

```
[arch@arch-pc ~]$ sudo usbhid-dump -d 2feb
001:005:002:DESCRIPTOR         1586350806.503802
 06 0A FF 09 01 A1 01 85 09 09 02 75 08 95 08 15
 00 26 FF 00 81 02 09 03 75 08 95 08 15 00 26 FF
 00 91 02 C0

001:005:001:DESCRIPTOR         1586350806.504166
 05 0D 09 01 A1 01 85 02 09 20 A1 00 09 42 09 44
 09 45 09 3C 15 00 25 01 75 01 95 04 81 02 75 01
 95 02 81 01 09 32 75 01 95 01 81 02 81 01 75 10
 95 01 35 00 A4 05 01 09 30 65 13 55 0D 46 34 03
 27 38 63 00 00 81 02 09 31 46 4D 04 27 88 3B 00
 00 81 02 B4 09 30 45 00 26 FF 1F 81 02 C0 C0 05
 01 09 06 A1 01 85 03 05 07 19 E0 29 E7 15 00 25
 01 75 01 95 08 81 02 05 07 19 00 29 FF 26 FF 00
 75 08 95 06 81 00 C0

001:005:000:DESCRIPTOR         1586350806.504608
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 03 15 00 25 01 95 03 75 01 81 02 95 05 81 01
 05 01 09 30 09 31 26 FF 7F 95 02 75 10 81 02 05
 0D 09 30 26 FF 1F 95 01 75 10 81 02 C0 C0
```

I tried this:
> setting .x_max = 25131, .y_max=15222 in veikk_device_info_0x0003 in veikk_vdev.c

But still don't work, i edited the /usr/src/input-veikk-r69.7ccdfb1/veikk_vdev.c file, is that ok?

KibaGasteiz, 2020-06-28:
Is there any progress on this? I don't know how can i help.

kprasadvnsi, 2020-07-13:
I am having exactly the same issue with my A50 tablet.

kprasadvnsi, 2020-07-27:
It looks like a kernel issue. My problem is fixed in Kernel 5.7.9

KibaGasteiz, 2020-07-28:
> It looks like a kernel issue. My problem is fixed in Kernel 5.7.9 I am on Fedora 32

Using 5.7.10 on Arch here and still the same issue.

jlam55555, 2020-08-11:
@KibaGasteiz @kprasadvnsi I'm sorry to only return to this issue so late. I obtained an A50 when I posted my previous comment but only got around to testing recently (see all the v3-alpha activity). The VEIKK A50's logical maximum should be 32768 for both the X and Y axes -- does setting x_max and y_max to that value fix this? (This is the default on v3-alpha now.)

After a rough measurement, I believe that 25399 and 15239 refer to the physical dimensions of the VEIKK A50 (i.e., the drawing pad is 25.4 x 15.24cm)

KibaGasteiz, 2020-08-11:
> struct veikk_device_info veikk_device_info_0x0003 = {
>     .name = "VEIKK A50 Pen", .prod_id = 0x0003,
>     .x_max = 32768, .y_max = 32768, .pressure_max = 8192,
>     .setup_and_register_input_devs = veikk_s640_setup_and_register_input_devs,
>     .alloc_input_devs = veikk_s640_alloc_input_devs,
>     .handle_raw_data = veikk_s640_handle_raw_data,
>     .handle_modparm_change = veikk_s640_handle_modparm_change

This is what i got, since i edited the values months ago, but it isn't fixed, tried to use "modprobe veikk" but still not working, it's working with your A50?

jlam55555, 2020-08-11:
@KibaGasteiz Yeah. This part from an earlier post is concerning:

> With the table i got this values at the bottom right part of the device:
>  >Event: time 1582456059.519189, type 3 (EV_ABS), code 0 (ABS_X), value 25131
> > Event: time 1582456059.519189, type 3 (EV_ABS), code 1 (ABS_Y), value 15222

which means that the device is somehow limited to sending events up to those values rather than the 32768 x 32768.

Two things to check come to mind:
- Is this caused by a screen mapping? What's in `/sys/module/veikk/parameters/screen_map` and `/sys/module/veikk/parameters/screen_size`?
- Did you recompile the driver after changing it to 32768? Check the output of `evtest` on the VEIKK device, and paste the content that looks like the following after selecting the VEIKK device (the following is for my keyboard, not a VEIKK device):

```
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)

```

kprasadvnsi, 2020-08-11:
> It looks like a kernel issue. My problem is fixed in Kernel 5.7.9 I am on Fedora 32

I am on Kernel 5.7.12 now and it still works. so the problem is gone since version 5.7.9. So if anybody wanna suggest some test then I would do it.

KibaGasteiz, 2020-08-11:
> Is this caused by a screen mapping? What's in /sys/module/veikk/parameters/screen_map and /sys/module/veikk/parameters/screen_size?

It's only a "0"

> Did you recompile the driver after changing it to 32768? Check the output of evtest on the VEIKK device, and paste the content that looks like the following after selecting the VEIKK device (the following is for my keyboard, not a VEIKK device):

I installed the AUR version, but never tried to compile myself, i will try now, but first i will paste the evtest results:

> Event: time 1597182449.119658, -------------- SYN_REPORT ------------
> Event: time 1597182449.167657, type 3 (EV_ABS), code 0 (ABS_X), value 25389
> Event: time 1597182449.167657, type 3 (EV_ABS), code 1 (ABS_Y), value 15237
> Event: time 1597182449.167657, -------------- SYN_REPORT ------------
> Event: time 1597182449.179655, type 3 (EV_ABS), code 1 (ABS_Y), value 15227
> Event: time 1597182449.179655, -------------- SYN_REPORT ------------
> Event: time 1597182449.185625, type 3 (EV_ABS), code 0 (ABS_X), value 25385
> Event: time 1597182449.185625, -------------- SYN_REPORT ------------
> Event: time 1597182449.191647, type 3 (EV_ABS), code 0 (ABS_X), value 25383
> Event: time 1597182449.191647, type 3 (EV_ABS), code 1 (ABS_Y), value 15220
> Event: time 1597182449.191647, -------------- SYN_REPORT ------------
> Event: time 1597182449.197629, type 3 (EV_ABS), code 1 (ABS_Y), value 15225
> Event: time 1597182449.197629, -------------- SYN_REPORT ------------
> Event: time 1597182449.221665, type 3 (EV_ABS), code 0 (ABS_X), value 25391
> Event: time 1597182449.221665, type 3 (EV_ABS), code 1 (ABS_Y), value 15238
> Event: time 1597182449.221665, -------------- SYN_REPORT ------------
> Event: time 1597182449.233623, type 3 (EV_ABS), code 1 (ABS_Y), value 15231
> Event: time 1597182449.233623, -------------- SYN_REPORT ------------
> Event: time 1597182449.239624, type 3 (EV_ABS), code 1 (ABS_Y), value 15236
> Event: time 1597182449.239624, -------------- SYN_REPORT ------------
> Event: time 1597182449.245623, type 3 (EV_ABS), code 1 (ABS_Y), value 15231
> Event: time 1597182449.245623, -------------- SYN_REPORT ------------
> Event: time 1597182449.251620, type 3 (EV_ABS), code 0 (ABS_X), value 25388
> Event: time 1597182449.251620, type 3 (EV_ABS), code 1 (ABS_Y), value 15235
> Event: time 1597182449.251620, -------------- SYN_REPORT ------------

KibaGasteiz, 2020-08-11:
Installed 2.0 and v3alpha and they won't work at all.
Shoul just "make" and "sudo make all install clean" just work?
I see how the files in "sys/module/veikk" are created, and the parameter files still have "0" in them except for pressure_map that has "6553600".
Can't find the "veikk_vdev.c" file inside "/usr/src/input-veikk...", so i assume that's only for the AUR version, the "veikk_vdev.c" file in the downloaded files where i compile have the corret value in them "32768 x 32768"

jlam55555, 2020-08-12:
@KibaGasteiz  v3-alpha won't work with the AUR package yet (since the AUR PKGBUILD references the v2 files, and the file structure has been simplified from v2 to v3). I wasn't really planning to fix this until v3 became stable.

v2 should work, however. (`make` and `sudo make all install`) *should* work. If it's already installed, you may have to do a `sudo make uninstall` before running the `sudo make install`.

Also, when I meant the evtest results, this time I meant the part that comes before the event reports, e.g., the section that looks something like the following (again, this is for my keyboard, but just to give an idea). I'm wondering if the ABS_X and ABS_Y bounds are misconfigured, and you should be able to tell in this output.

```
Select the device event number [0-22]: 5
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
    Event code 55 (KEY_KPASTERISK)
    Event code 56 (KEY_LEFTALT)
[... 91 lines cut ...]
    Event code 217 (KEY_SEARCH)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 226 (KEY_MEDIA)
    Event code 227 (KEY_SWITCHVIDEOMODE)
    Event code 236 (KEY_BATTERY)
    Event code 237 (KEY_BLUETOOTH)
    Event code 238 (KEY_WLAN)
    Event code 425 (KEY_PRESENTATION)
    Event code 434 (KEY_DOLLAR)
    Event code 435 (KEY_EURO)
    Event code 464 (KEY_FN)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 17 (EV_LED)
    Event code 0 (LED_NUML) state 1
    Event code 1 (LED_CAPSL) state 0
    Event code 2 (LED_SCROLLL) state 0
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
```

KibaGasteiz, 2020-08-12:
> v3-alpha won't work with the AUR package yet (since the AUR PKGBUILD references the v2 files, and the file structure has been simplified from v2 to v3). I wasn't really planning to fix this until v3 became stable.

I already uninstalled the AUR, i was just trying compiling v2 and v3 and both of them didn't work.

> I'm wondering if the ABS_X and ABS_Y bounds are misconfigured, and you should be able to tell in this output.

I have 3 different entries on evtest. Pasting the three here:

> Select the device event number [0-30]: 28
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2feb product 0x3 version 0x100
Input device name: "VEIKK A50 Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    32768
      Resolution       1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    32768
      Resolution       1
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8192
Properties:
  Property type 0 (INPUT_PROP_POINTER)
  Property type 1 (INPUT_PROP_DIRECT)

> Select the device event number [0-30]: 29
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2feb product 0x3 version 0x100
Input device name: "VEIKK A50 Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    32768
      Resolution       1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    32768
      Resolution       1
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8192
Properties:
  Property type 0 (INPUT_PROP_POINTER)
  Property type 1 (INPUT_PROP_DIRECT)

> Select the device event number [0-30]: 30
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2feb product 0x3 version 0x100
Input device name: "VEIKK A50 Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    32768
      Resolution       1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    32768
      Resolution       1
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8192
Properties:
  Property type 0 (INPUT_PROP_POINTER)
  Property type 1 (INPUT_PROP_DIRECT)

jlam55555, 2020-08-13:
Wondering what "both of them didn't work" means. That output looks correct, so this is confusing.

Maybe I'm going in circles now, but now I think it might have to do with the fact that our HID reports are different. Maybe we have different editions of the A50 that have slightly different internals and is reporting differently.

Just to cover all the bases, when you changed the x_max and y_max to 25399 and 15239, did you make sure to recompile the driver afterwards? I'm reading your comment from way back and it's not clear from it whether you did.

KibaGasteiz, 2020-08-13:
Just uninstalled again: https://i.imgur.com/XcCDLFR.png
Then make sure that "changed the x_max and y_max to 25399 and 15239": https://i.imgur.com/kxx1cAg.png
And compiling again: https://i.imgur.com/Atp3tvb.png
Result: Not working at all. (At least using the AUR version let me move in a fraction of the screen).

KibaGasteiz, 2020-12-17:
I don't know what happened, but it's working fine now, may be some update on my OS?
You can close the issue if you want, or ask me for tests so we can understand what was happening. :)

kprasadvnsi, 2020-12-18:
> I don't know what happened, but it's working fine now, may be some update on my OS?
> You can close the issue if you want, or ask me for tests so we can understand what was happening. :)

the same thing happened to me as well. I have tried the A50 tablet on Ubuntu 20.04 without this driver and it maps my tablet correctly and the pressure sensitivity works out of box.

KibaGasteiz, 2020-12-19:
So it has been fixed on the kernel i assume.
Closing this issue :)

closed 2020-12-19 by KibaGasteiz