jlam55555/veikk-linux-driver #24: veikk_set_veikk_screen_size: Deserialization reads uninitialized values

Cebtenzzre, 2020-02-27:
#### Issue Description
The following deserialization code (in `veikk_set_veikk_screen_size`) upsets clang-tidy.
https://github.com/jlam55555/veikk-linux-driver/blob/7ccdfb127f3082034e95da778f22b0ec65d8e7fc/veikk_modparms.c#L51-L57
This is what clang-tidy says about it:
```
veikk_modparms.c:56:57: warning: The left operand of '<=' is a garbage value [clang-analyzer-core.UndefinedBinaryOperatorResult]
    if(ss && (screen_size.width<=0 || screen_size.height<=0))
                                                        ^
```

`height` is a garbage value because `ss` (`u32`, 4 bytes) is reinterpreted as a `struct veikk_screen_size` (16 bytes). Only `x` is actually initialized by line 55.

In revision 81c7207, `veikk_set_veikk_screen_size` used `veikk_ss_to_rect`, which didn't have this problem.

#### Versions
This is a regression introduced by commit 73be60c.

jlam55555, 2020-05-05:
The casting done in https://github.com/jlam55555/veikk-linux-driver/commit/73be60ccc90e0ec9a3a8a14fafd6d8fc6aff2969 was done mostly out of convenience and a (naive?) thought that gcc's behavior should be consistent across most devices. However, I will revert it to the more type-safe casting once I get time to work on this again.

By the way, @Cebtenzzre, did this cause any actual problems for you? Are you building your kernel with clang? Asking out of curiosity. If you have any advice about the safety of this cast, please let me know. AFAIK, `struct veikk_screen_size` should be 4 bytes by any reasonable compiler implementation -- any thoughts on why clang says 16 bytes?

Cebtenzzre, 2020-05-05:
Sorry, I've only _just_ realized that `veikk_screen_size` refers to both a variable of type `struct veikk_rect` AND an unrelated type with some overlap in the names of its members (`struct veikk_screen_size`). That threw me for a loop when I looked at this originally. I'll have to revisit my analysis, maybe the code works fine...

I'm more familiar with C++, where type punning like this is undefined behavior and variable names can't be the same as struct names.

And I actually haven't used this driver yet, I just threw a static analyzer at it to see if it was of sufficient quality to _consider_ using.

jlam55555, 2020-05-05:
"This code isn't as bad as it looked to me at first" -- haha, I guess that's a good thing to hear. I agree about fixing the UB though, and the poor choice of variable name. Will fix later.