jlam55555/veikk-linux-driver #25: "make" does not work despite linux headers installed

paoloperrone, 2020-03-18:
Hello. I have a problem during the "make" process. 

Here is the output:
```
paolo@ThinkPad:~/Software/veikk-linux-driver-master$ make
make -C /lib/modules/4.15.0-91-lowlatency/build M=/home/paolo/Software/veikk-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-91-lowlatency'
  CC [M]  /home/paolo/Software/veikk-linux-driver-master/veikk_drv.o
In file included from ./include/linux/input.h:22:0,
                 from ./include/linux/hid.h:35,
                 from /home/paolo/Software/veikk-linux-driver-master/veikk.h:10,
                 from /home/paolo/Software/veikk-linux-driver-master/veikk_drv.c:5:
/home/paolo/Software/veikk-linux-driver-master/veikk_drv.c: In function 'veikk_report':
/home/paolo/Software/veikk-linux-driver-master/veikk_drv.c:93:46: error: 'struct hid_report' has no member named 'application'
              report->id, report->type, report->application, report->maxfield,
                                              ^
./include/linux/device.h:1420:58: note: in definition of macro 'dev_info'
 #define dev_info(dev, fmt, arg...) _dev_info(dev, fmt, ##arg)
                                                          ^~~
/home/paolo/Software/veikk-linux-driver-master/veikk_drv.c:92:5: note: in expansion of macro 'hid_info'
     hid_info(hdev, "parsing report: %d %d %d %d %d\n",
     ^~~~~~~~
scripts/Makefile.build:330: recipe for target '/home/paolo/Software/veikk-linux-driver-master/veikk_drv.o' failed
make[2]: *** [/home/paolo/Software/veikk-linux-driver-master/veikk_drv.o] Error 1
Makefile:1577: recipe for target '_module_/home/paolo/Software/veikk-linux-driver-master' failed
make[1]: *** [_module_/home/paolo/Software/veikk-linux-driver-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-91-lowlatency'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
```

What is exactly the problem?

My kernel is lowlatency, the very same happens if I use the generic one. The linux headers of the latest version are installed.

ykzhou, 2020-03-21:
encountered exactly the same errors on ubuntu 16.04 with headers-4.15.0-91-generic: 

veikk_drv.c: In function ‘veikk_report’:
veikk_drv.c:93:46: error: ‘struct hid_report’ has no member named ‘application’
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-91-generic'

sshwishh, 2020-03-22:
i did everything in the guide: "How to resolve the “/lib/modules/3.13.0-27-generic/build: No such file or directory” error message?"

but still getting output

`len-Aspire-E5-471G:~/Downloads/veikk-linux-driver-master$ make
make -C /lib/modules/5.3.0-42-generic/build M=/home/len/Downloads/veikk-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-42-generic'
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "__divdi3" [/home/len/Downloads/veikk-linux-driver-master/veikk.ko] undefined!
scripts/Makefile.modpost:92: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1658: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-42-generic'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
`

shrimpwagon, 2020-03-26:
Same: Linux Mint 19.1 (4.15.0-20-generic)

arrow2nd, 2020-04-14:
I had the same error, but I solved it with the following method.(ZorinOS 15.2)
`sudo apt install gcc`
`make`
`sudo depmod`
`sudo modprobe veikk`
`sudo make all install clean`

jlam55555, 2020-05-05:
@paoloperrone @ykzhou @sshwishh @shrimpwagon I'm sorry, those lines were part of my messy debuggin and can/should be commented out. You should be able to remove them without problem. It might be a problem with an older kernel version. Unfortunately, I have only been testing on Fedora and Arch (which run the latest kernels by default). Will likely spin up a VM to try it out sometime.

(But no guarantees until school year is over, so in roughly two weeks. In the meantime, you're welcome to try this out on a newer kernel version.)

jlam55555, 2020-05-05:
See if this commit helps: https://github.com/jlam55555/veikk-linux-driver/commit/2bb2a1e9615f9c27a12107d666ffe43f689be6df

I was testing how much information the tablet provides in the `veikk_report` function, and in my haste didn't get to test much with it. I believe `report->application` may have been added on a later kernel version. For now, it should be safe to comment this out (as I've done in commit above) -- it is not essential to the driver.

paoloperrone, 2020-05-05:
Hi, thank you for the update.
It still doesn't work unfortunately. This time "make" works, but the install process does not. Here's the output:

```
paolo@ThinkPad:~/Software/veikk-linux-driver-master$ sudo make all install clean
make -C /lib/modules/4.15.0-99-lowlatency/build M=/home/paolo/Software/veikk-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-99-lowlatency'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-99-lowlatency'
make -C /lib/modules/4.15.0-99-lowlatency/build M=/home/paolo/Software/veikk-linux-driver-master modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-99-lowlatency'
  INSTALL /home/paolo/Software/veikk-linux-driver-master/veikk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-99-lowlatency
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-99-lowlatency'
mkdir -p /etc/modules-load.d
modprobe veikk
modprobe: FATAL: Module veikk not found in directory /lib/modules/4.15.0-99-lowlatency
Makefile:14: recipe for target 'install' failed
make: *** [install] Error 1
```

jlam55555, 2020-05-05:
Look at Nicole's solution here https://github.com/jlam55555/veikk-linux-driver/issues/27. It looks like `depmod` is not automatically called in the `make install` recipe. If that works, I'll close this issue (but I still have to resolve that issue)

paoloperrone, 2020-05-05:
I see. Now it works, thanks!

jlam55555, 2020-05-05:
Closing now, as it appears the issue is fixed with the new commit. @ykzhou @shrimpwagon @sshwishh feel free to reopen if you still face the same issue.

closed 2020-05-05 by jlam55555