jlam55555/veikk-linux-driver #27: "FATAL: Module veikk not found in directory" and a potential fix

Nicole4740, 2020-03-23:
Hello. I'm using Linux Mint (Ubuntu based) and whenever I try to follow the instructions provided, it says: FATAL: Module veikk not found in directory. 
What I've found to work on this is following the instructions as given (You MUST follow the initial instructions, it will not work if you don't.) Once you get the error, type sudo depmod, and then sudo modprobe veikk, and then re type sudo make all install clean, and it should work just fine. As far as I know, this works on Pop!_OS 19.10, and Linux Mint 19.3. This should work with other ubuntu based distributions.

iamashwin99, 2020-03-26:
Hi I faced the same error and this solution worked for me on Manjaro 19.0.2 KDE. Hers some more info on the error.
```bash
ashwin@ashwin-pc veikk-s640-driver]$ sudo make all install clean
make -C /lib/modules/5.4.24-1-MANJARO/build M=/home/ashwin/veikk-s640-driver modules
make[1]: Entering directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
make -C /lib/modules/5.4.24-1-MANJARO/build M=/home/ashwin/veikk-s640-driver modules_install
make[1]: Entering directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
  INSTALL /home/ashwin/veikk-s640-driver/veikk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.24-1-MANJARO
make[1]: Leaving directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
modprobe veikk
libkmod: kmod_config_parse: /etc/modprobe.d/bbswitch.conf line 1: ignoring bad line starting with 'bbswitch'
echo "veikk" > /etc/modules-load.d/veikk.conf
make -C /lib/modules/5.4.24-1-MANJARO/build M=/home/ashwin/veikk-s640-driver clean
make[1]: Entering directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
  CLEAN   /home/ashwin/veikk-s640-driver/Module.symvers
make[1]: Leaving directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
```

Nicole4740, 2020-03-26:
> Hi I faced the same error and this solution worked for me on Manjaro 19.0.2 KDE. Hers some more info on the error.
> 
> ```shell
> ashwin@ashwin-pc veikk-s640-driver]$ sudo make all install clean
> make -C /lib/modules/5.4.24-1-MANJARO/build M=/home/ashwin/veikk-s640-driver modules
> make[1]: Entering directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
>   Building modules, stage 2.
>   MODPOST 1 modules
> make[1]: Leaving directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
> make -C /lib/modules/5.4.24-1-MANJARO/build M=/home/ashwin/veikk-s640-driver modules_install
> make[1]: Entering directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
>   INSTALL /home/ashwin/veikk-s640-driver/veikk.ko
> At main.c:160:
> - SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
> - SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
> sign-file: certs/signing_key.pem: No such file or directory
>   DEPMOD  5.4.24-1-MANJARO
> make[1]: Leaving directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
> modprobe veikk
> libkmod: kmod_config_parse: /etc/modprobe.d/bbswitch.conf line 1: ignoring bad line starting with 'bbswitch'
> echo "veikk" > /etc/modules-load.d/veikk.conf
> make -C /lib/modules/5.4.24-1-MANJARO/build M=/home/ashwin/veikk-s640-driver clean
> make[1]: Entering directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
>   CLEAN   /home/ashwin/veikk-s640-driver/Module.symvers
> make[1]: Leaving directory '/usr/lib/modules/5.4.24-1-MANJARO/build'
> ```

Nice! I'm glad to see that this even works on Manjaro as well. Thanks for posting that!

jlam55555, 2020-05-05:
@Nicole4740 @iamashwin99 Does https://github.com/jlam55555/veikk-linux-driver/commit/815cff1836e031d6d9a6fa988276d3c449cefe61 fix the error without the "workaround"? (Explicitly added `depmod` in the install recipe.)

iamashwin99, 2020-05-09:
Ya, that commit works for me. Thanks for the patch!

Nicole4740, 2020-05-11:
> @Nicole4740 @iamashwin99 Does [815cff1](https://github.com/jlam55555/veikk-linux-driver/commit/815cff1836e031d6d9a6fa988276d3c449cefe61) fix the error without the "workaround"? (Explicitly added `depmod` in the install recipe.)

Im a tad busy right now with quarantine school work and the like, but I'll let you know soon enough! Also another thing that might be of interest to you is that my patch didn't work on Elementary OS for some reason, I don't remember the exact error code so later today I'll make a live USB and mess with it.

Nicole4740, 2020-05-11:
Okay, after running `make` and `sudo make all install clean` on an Elementary OS live USB I recieved the following error:
`elementary@elementary:~/veikk-linux-driver$ make
make -C /lib/modules/5.3.0-51-generic/build M=/home/elementary/veikk-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-51-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
make[1]: gcc: Command not found
  CC [M]  /home/elementary/veikk-linux-driver/veikk_drv.o
/bin/sh: 1: gcc: not found
scripts/Makefile.build:288: recipe for target '/home/elementary/veikk-linux-driver/veikk_drv.o' failed
make[2]: *** [/home/elementary/veikk-linux-driver/veikk_drv.o] Error 127
Makefile:1656: recipe for target '_module_/home/elementary/veikk-linux-driver' failed
make[1]: *** [_module_/home/elementary/veikk-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-51-generic'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
elementary@elementary:~/veikk-linux-driver$ sudo make all install clean
make -C /lib/modules/5.3.0-51-generic/build M=/home/elementary/veikk-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-51-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
make[1]: gcc: Command not found
  CC [M]  /home/elementary/veikk-linux-driver/veikk_drv.o
/bin/sh: 1: gcc: not found
scripts/Makefile.build:288: recipe for target '/home/elementary/veikk-linux-driver/veikk_drv.o' failed
make[2]: *** [/home/elementary/veikk-linux-driver/veikk_drv.o] Error 127
Makefile:1656: recipe for target '_module_/home/elementary/veikk-linux-driver' failed
make[1]: *** [_module_/home/elementary/veikk-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-51-generic'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2`

jlam55555, 2020-05-11:
@Nicole4740 Looks like gcc is not installed. That sounds very possible on a live usb environment.

(Edited your post for formatting btw. Never knew I could do that)

urdad-m, 2020-05-17:
I type in the sudo make all install clean command and after typing my pass  it gives me an error message saying make: *** No rule to make target install. stop.

urdad-m, 2020-05-17:
I type in the `sudo make all install clean` command and after typing my pass it gives me an error message saying 

> make: *** No rule to make target install. stop.

jlam55555, 2020-05-17:
@urdad-m This might be a silly question, but just wanted to cover all the bases. You cloned this repo and are running the `make` command inside the repo directory, right?

urdad-m, 2020-05-18:
@jlam55555 I try running the `make` command however it also gives me the same error message

jlam55555, 2020-05-18:
@urdad-m That doesn't answer my question. Are you calling the `make` command when you are in the correct directory? You must be in the same directory as where you cloned this repository.

I am closing this issue because the original issue in this thread is resolved. @Nicole4740  @urdad-m if you have a separate issue, please open in your own thread.

closed 2020-05-18 by jlam55555

OceanNuclear, 2020-11-18:
I'm having the same bug again, I'll open another issue if that's okay.