jlam55555/veikk-linux-driver #29: function veikk_report struct hid_report has no member named applcation error

jeromewiley, 2020-04-25:
Hi James,

Relative Linux-newb here. Was going to start looking at the .drv file with Sublime in a bit, but just in case you see this message:

I bought a brand new, beautiful Veikk A15 Pro tablet, and am super excited to start using it. I knew there was a 50/50 chance I wouldn't be able to use with Linux, but was excited to see your driver on Github.

Got through part of the install, then got this error:

```
jerome@nuc:~/veikk-linux-driver-master⟫ sudo make all install clean
make -C /lib/modules/4.4.0-121-generic/build M=/home/jerome/veikk-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-121-generic'
  CC [M]  /home/jerome/veikk-linux-driver-master/veikk_drv.o
In file included from include/linux/input.h:22:0,
                 from include/linux/hid.h:35,
                 from /home/jerome/veikk-linux-driver-master/veikk.h:10,
                 from /home/jerome/veikk-linux-driver-master/veikk_drv.c:5:
/home/jerome/veikk-linux-driver-master/veikk_drv.c: In function ‘veikk_report’:
/home/jerome/veikk-linux-driver-master/veikk_drv.c:93:46: **error**: ‘struct hid_report’ has no member named ‘application’
              report->id, report->type, report->application, report->maxfield,
                                              ^
include/linux/device.h:1176:58: note: in definition of macro ‘dev_info’
 #define dev_info(dev, fmt, arg...) _dev_info(dev, fmt, ##arg)
                                                          ^
/home/jerome/veikk-linux-driver-master/veikk_drv.c:92:5: note: in expansion of macro ‘hid_info’
     hid_info(hdev, "parsing report: %d %d %d %d %d\n",
     ^
scripts/Makefile.build:277: recipe for target '/home/jerome/veikk-linux-driver-master/veikk_drv.o' failed
make[2]: *** [/home/jerome/veikk-linux-driver-master/veikk_drv.o] Error 1
Makefile:1426: recipe for target '_module_/home/jerome/veikk-linux-driver-master' failed
make[1]: *** [_module_/home/jerome/veikk-linux-driver-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-121-generic'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
```
I checked some of the other issues, and also Googled various text strings based on this error, but could find nothing. Let me know if you can help, thanks!

jlam55555, 2020-05-05:
My guess is that it's because you're on an old kernel (4.4). I've only tested on 4.18 and newer (currently on 5.6.7 with Arch). Can you try installing a newer kernel version (e.g., with Ukuu or manually)?

jlam55555, 2020-05-05:
Sorry for the delayed reply. Likely a dupe of https://github.com/jlam55555/veikk-linux-driver/issues/25. You should be able to comment out that line. `veikk_report` was for debugging purposes only.

jlam55555, 2020-05-05:
From the other thread, this issue seems fixed with the new commit. Reopen if you still encounter the same issue.

closed 2020-05-05 by jlam55555