jlam55555/veikk-linux-driver #30: Make error.

Shigetorum, 2020-05-06:
Hi, i am facing this error when i run the command "make"
```
make -C /lib/modules/5.3.0-51-generic/build M=/home/shigetorum/Escritorio/aaa/veikk-linux-driver-master modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.3.0-51-generic'
  CC [M]  /home/shigetorum/Escritorio/aaa/veikk-linux-driver-master/veikk_drv.o
  CC [M]  /home/shigetorum/Escritorio/aaa/veikk-linux-driver-master/veikk_vdev.o
  CC [M]  /home/shigetorum/Escritorio/aaa/veikk-linux-driver-master/veikk_modparms.o
  LD [M]  /home/shigetorum/Escritorio/aaa/veikk-linux-driver-master/veikk.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/shigetorum/Escritorio/aaa/veikk-linux-driver-master/veikk.mod.o
  LD [M]  /home/shigetorum/Escritorio/aaa/veikk-linux-driver-master/veikk.ko
make[1]: se sale del directorio '/usr/src/linux-headers-5.3.0-51-generic'
```

jlam55555, 2020-05-06:
@Shigetorum This is the unexpected output of running `make`. Will close this issue if you don't clarify what the issue is.

jlam55555, 2020-05-11:
Closing because unclear.

closed 2020-05-11 by jlam55555