jlam55555/veikk-linux-driver #33: A15 Pro X & Y Axis Are Opposite Then Screen X & Y Axis

kantst0p, 2020-06-17:
Instead of reading it the same as the screen x & y axis, it seems to have flipped the x & y axis around on the tablet to work in a vertical orientation on the screen instead of horizontally(the correct orientation). Running  linux-headers-4.4.0-178-generic  on Trisquel 8.0.

jlam55555, 2020-06-20:
For now, I would recommend using the [configuration tool](https://github.com/jlam55555/veikk-linux-driver-gui) to change the orientation, but I don't know why this is the problem, since it is using the default orientation for all tablets (and shouldn't be different only for the A15 Pro). I want to see if other users report this same issue before attempting any changes.

kantst0p, 2020-06-23:
I haven't been able to run the gui, I am a beginner in using the terminal to run commands. I have followed the instructions you have provided in the 'Compiling and executing the tool' section and am able to execute the 'qmake && make all clean' command, however when I execute the 'sudo ./veikk-linux-driver-gui' command(in root), it says **"sudo: ./veikk-linux-driver-gui: command not found**". I would also like to mention, the first time trying to install/configure the settings/package, my computer died suddenly(because it was left unplugged), it had two errors before that, and I was in the midst of reporting the issue when my work was lost, unable to be retrieved/remembered. I resolved at least one issue by now, which was the tablet not responding, but am lost to what the second problem was.

kantst0p, 2020-06-24:
I've also just noticed the input is not working, no button or touch, only the cursor works.

jlam55555, 2020-07-11:
@kantst0p Sorry for the late reply. Button support is coming in V3 of the driver, now that I think I've figured how to deal with mapping in user space. So the driver is working as intended for now.

closed 2020-07-12 by kantst0p