jlam55555/veikk-linux-driver #34: So, does driver abandoned?

Casul51, 2020-07-11:
As A30 owner I'm really looking for updates and this seems completely abandoned by author.

jlam55555, 2020-07-11:
(Not abandoned, just not at the forefront of my mind right now...)

This should work with the A30, do you have any specific problems or are you waiting for button support? Please detail your question more.

If you're wondering about button support and mapping, I'm planning a V3 of this driver and associated (user-space) mapper/GUI. Because I'm still somewhat of a beginner with the APIs Linux has to offer, it's been going slowly, but I do have an idea of how to do it. Might take a few weeks to implement however.

Casul51, 2020-07-16:
Actually waiting for button support but also I'm waiting for proper dual screen support (because there is really hard to proper screen mapping when there is few screens with different resolution, like it was more easier with old driver version)

lenonk, 2020-12-30:
I'm currently trying to use it with 3 screens, and it's *very* confused.  I had to set the variables to nonsensical values to get it to work properly, but it *does* work.  I currently have 2 1080p monitors, one to the left and one to the right of a 1440p monitor.

Maybe I'll look into it and submit a patch.

Below is what the GUI currently shows, with the values that constrain the table to only the 1440p monitor:

[image]

Disclaimer:  I have almost no experience with drawing tablets and setting them up.  These values *seem* nonsensical to me, but I could be wrong.