jlam55555/veikk-linux-driver #40: v3-alpha testers

jlam55555, 2020-08-11:
This thread will be for open discussion (e.g., questions, suggestions, answers, what-have-you's) for those testing v3 of the driver. Please post all questions related to v3-alpha here before it becomes a release.

The relevant blog posts related to this driver:
- [v3 updates](http://everything-is-sheep.herokuapp.com/posts/veikk-linux-driver-v3-notes)
- [button mapping](http://everything-is-sheep.herokuapp.com/posts/button-mapping-journeys)

When you first post, please list your Linux distribution, kernel version, and devices, and please try to list relevant output (e.g., with `xinput test`).

@meelash @sparshjain265

jlam55555, 2020-08-11:
I just received the default keycodes for all of the devices from VEIKK, and I'll try to implement them over the next day or so. When this is done, the first button should report <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>Shift</kbd>+<kbd>Keypad 0</kbd>, the second should report <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>Shift</kbd>+<kbd>Keypad 1</kbd>, etc.

I detailed why I ended up using these long keycodes in the blog post about button mapping, but I'd like any feedback on that as a design decision.

Current devices with button mapping working: A50, VK1560

Current devices with button mapping not working: A30, A15, A15Pro, VK1560Pro, VK2200 (will try to get these working over the next day or two and let you know)

In the meantime you can let me know if the pointer events are working properly with v3-alpha, and (when I've added the mappings for all of the devices) if the buttons are correctly reporting the aforementioned key combinations. (E.g., post the output of `xinput list` after pressing some buttons)

jlam55555, 2020-08-11:
Now that I've looked at the button codes, the only one that I notice might cause problems is the A15 Pro. The 5th key (third row, first button) is mapped to <kbd>Ctrl</kbd>, and some other buttons are mapped to <kbd>Ctrl</kbd>+ other buttons, which might hide the <kbd>Ctrl</kbd> key.

    A15 Pro
    +--------------------
    |   _ 
    |  ( ) <- dial
    |   -
    |
    |  BTN_1  | BTN_2
    |  BTN_3  | BTN_4
    |  BTN_5  | BTN_6
    |  BTN_7  | BTN_8
    |  BTN_9  | BTN_10
    |  BTN_11 | BTN_12

@sparshjain265 (Or anyone else with the A15 Pro) Can you try this: 

1. Uncomment the `#VEIKK_DEBUG_MODE` macro in the code
2. Compile the driver: `make all uninstall install`, only call uninstall if already installed
3. Run `dmesg -w`
4. Press some keys (both single keys and key combinations), especially with BTN_5, BTN_4, BTN_11, BTN_12. Also try turning the dial left and right. Record which keys correspond with what `dmesg` output.
5. Paste the output of `dmesg` in a comment (make sure to indicate which keys correspond to which parts).

Thanks.

RaSTuS26, 2020-08-11:
@sparshjain265
Linux Mint 19.3, Kernel 5.4.0-42-generic, device A30, using alpha3 driver updated 3 hours ago

`> xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech Wireless Mouse PID:0084          id=9    [slave  pointer  (2)]
⎜   ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0 Mouse        id=11   [slave  pointer  (2)]
⎜   ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0 Consumer Control     id=12   [slave  pointer  (2)]
⎜   ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0 Consumer Control     id=13   [slave  pointer  (2)]
⎜   ↳ Realtek RTL2832U reference design         id=15   [slave  pointer  (2)]
⎜   ↳ lircd-uinput                              id=16   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Sleep Button                              id=8    [slave  keyboard (3)]
    ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0      id=10   [slave  keyboard (3)]
    ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0 System Control       id=14   [slave  keyboard (3)]
    ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0 Consumer Control     id=17   [slave  keyboard (3)]
    ↳ Microsoft Microsoft® 2.4GHz Transceiver v8.0 Consumer Control     id=18   [slave  keyboard (3)]
    ↳ Realtek RTL2832U reference design         id=19   [slave  keyboard (3)]
    ↳ lircd-uinput                              id=20   [slave  keyboard (3)]
    ↳ VEIKK A30 Pen                             id=21   [slave  keyboard (3)]`

Still GIMP pressure not working with 'Edit/Input Devices/VEIKK A30 Pen' enabled in screen mode, but fine in Krita.

xev doesn't report any keycodes for the buttons.

jlam55555, 2020-08-11:
Thanks @RaSTuS26. *Almost* everything you said is expected, and I might not have been entirely clear. (Also, I wrote `xinput list` but meant `xinput test [device=id]` (in your case `xinput test 21` for the pen). xev doesn't report any keycodes for the buttons because I didn't implement the mapping for the A30 yet (again, hopefully will have a draft up by this afternoon).

For the buttons, there should be (a lot of) debugging output right now in `dmesg -w` whenever you move the mouse or press a button. You can experiment with this. 

The only thing that's unexpected is still the GIMP pressure..... Maybe some debugging with `evtest` and `xinput test` will help with that.

RaSTuS26, 2020-08-11:
Cheers jlam55555, I didn't really expect the buttons to work from reading your notes, but thought I'd try, then report back anyway.

Here is the output from xinput:
```
 xinput test 21
no event registered...
```

The output from evtest:
```
/dev/input/event23:     VEIKK A30 Pen
Select the device event number [0-23]: 23
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2feb product 0x2 version 0x100
Input device name: "VEIKK A30 Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   5680
      Min        0
      Max    32768
      Resolution     100
    Event code 1 (ABS_Y)
      Value  25091
      Min        0
      Max    32768
      Resolution     100
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8192
Properties:
  Property type 0 (INPUT_PROP_POINTER)
```

'dmesg -w' only showed output for pen events, not buttons, as expected.

A shortened dmesg output:
```
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_report: report id 1
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90001 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90002 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90003 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 10030 value 12641
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 10031 value 20018
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage d0030 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_report: report id 1
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90001 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90002 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90003 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 10030 value 12732
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 10031 value 19747
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage d0030 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_report: report id 1
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90001 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90002 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 90003 value 0
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 10030 value 12732
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage 10031 value 19747
[Tue Aug 11 21:21:42 2020] veikk 0003:2FEB:0002.000D: in veikk_event: usage d0030 value 0
```

jlam55555, 2020-08-11:
@RaSTuS26 Try again with the [newest commit](https://github.com/jlam55555/veikk-linux-driver/commit/ef90b1bc0cadbc9ef0ada63e8b51b4102eebc46d). A new "VEIKK A30 Keyboard" input *should* show up in evtest and xinput, and hopefully things will just work.

(Same thing, `evtest`, `xinput test`, and `dmesg` outputs will all be helpful.)

RaSTuS26, 2020-08-11:
Here we go jlam55555

'xinput list' output:```
    ↳ VEIKK A30 Pen                             id=21   [slave  keyboard (3)]
    ↳ VEIKK A30 Pen                             id=22   [slave  keyboard (3)]
    ↳ VEIKK A30 Pen                             id=23   [slave  keyboard (3)]
```
Tried all 3 id's, same output as previous (no event registered...)

'evtest' devices:```
/dev/input/event23:     VEIKK A30 Pen
/dev/input/event24:     VEIKK A30 Pen
/dev/input/event25:     VEIKK A30 Pen
```
Again, tried all 3 inputs, no live output given.

'dmesg -w' output for button presses:```
[Tue Aug 11 22:12:07 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:08 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:08 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:08 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:09 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:10 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:10 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:10 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:10 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:10 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:10 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
[Tue Aug 11 22:12:11 2020] veikk 0003:2FEB:0002.0011: Unknown input report with id 3
```

jlam55555, 2020-08-11:
I believe you've compiled and loaded the version from the master branch -- this is the behavior of the V2 driver (and what the V3 driver is intended to fix).

RaSTuS26, 2020-08-11:
jlam55555 No, definitely from the alpha3 branch, the latest commit was only 10mins old when I downloaded it.  It did make a difference, I now had 3 hid-inputs instead of 1, so something did register.  As mentioned I also made sure to unload previous driver with 'rmmod' before activating (replugging A30).

This is the branch link I'm using:
https://github.com/jlam55555/veikk-linux-driver/tree/v3-alpha

jlam55555, 2020-08-11:
@RaSTuS26 I understand that it made a difference, but it's not the latest update. See [this line](https://github.com/jlam55555/veikk-linux-driver/blob/v3-alpha/hid-veikk.c#L284): the format is different from what you're showing.

Sorry if this is a stupid question, but how did you download it? Did you make sure to run `git checkout v3-alpha` after downloading it? (`git status` should show "On branch v3-alpha")

RaSTuS26, 2020-08-11:
@jlam55555 I'm an idiot, I compiled from the old v2 directory.  Now installed latest v3alpha, will report shortly.

I don't use git, just download the zip.

RaSTuS26, 2020-08-11:
@jlam55555 ok, testing now.

'xinput list'
```
    ↳ VEIKK A30 Keyboard                        id=21   [slave  keyboard (3)]
    ↳ VEIKK A30 Pen                             id=22   [slave  keyboard (3)]
```

'xinput test 21' output:
```
> xinput test 21
key press   37
key press   64
key press   50
key press   90
key release 37
key release 64
key release 50
key release 90

key press   37
key press   64
key press   50
key press   87
key release 37
key release 64
key release 50
key release 87

key press   37
key press   64
key press   50
key press   88
key release 37
key release 64
key release 50
key release 88

key press   37
key press   64
key press   50
key press   83
key release 37
key release 64
key release 50
key release 83
```

'evtest' output:
```
Event: time 1597150932.477877, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597150932.477877, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597150932.477877, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597150932.477877, type 1 (EV_KEY), code 82 (KEY_KP0), value 1
Event: time 1597150932.477877, -------------- SYN_REPORT ------------
^[[2;8~Event: time 1597150932.578007, type 1 (EV_KEY), code 82 (KEY_KP0), value 2
Event: time 1597150932.578007, -------------- SYN_REPORT ------------
Event: time 1597150932.578007, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597150932.578007, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597150932.578007, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597150932.578007, type 1 (EV_KEY), code 82 (KEY_KP0), value 0
Event: time 1597150932.578007, -------------- SYN_REPORT ------------
Event: time 1597150934.385827, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597150934.385827, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597150934.385827, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597150934.385827, type 1 (EV_KEY), code 79 (KEY_KP1), value 1
Event: time 1597150934.385827, -------------- SYN_REPORT ------------
Event: time 1597150934.486023, type 1 (EV_KEY), code 79 (KEY_KP1), value 2
Event: time 1597150934.486023, -------------- SYN_REPORT ------------
Event: time 1597150934.526021, type 1 (EV_KEY), code 79 (KEY_KP1), value 2
[... 186 lines cut ...]
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 79 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 82 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 0 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 0 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 0 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 75 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 0 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 0 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 74 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 78 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 179 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: KEY 180 VALUE 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e0 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e1 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e2 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e3 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e4 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e5 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e6 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 700e7 value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_event: usage 7001d value 0
[Tue Aug 11 23:04:23 2020] veikk 0003:2FEB:0002.0014: in veikk_report: report id 3
```

Yay, it's working Jonathan, great work mate.

jlam55555, 2020-08-11:
@RaSTuS26 Cheers! Now go have some fun with [xbindkeys](https://wiki.archlinux.org/index.php/Xbindkeys). Keyboard shortcuts like `Control+Alt+Shift+KP_0` should work.

(I also realized I made a mistake with the mapping: the fourth button should map to Keypad 3 but I accidentally wrote Keypad 4. Will fix this in the next commit.)

In the meantime I'll be waiting on input from @sparshjain for the A15 Pro.

RaSTuS26, 2020-08-11:
@jlam55555 Have GIMP working now, didn't notice the 'Dynamics' (brush/pencil) options, my tool dock was too small to see them. Will play with that for a bit now.

jlam55555, 2020-08-11:
On a side note, will experiment a little today with a USB sniffer on Windows with the Windows driver to see if I can reverse engineer the USB packets. I really wonder if VEIKK performs the key mappings in hardware after sending some command to it since mapping like I'm currently attempting in this driver feels a little too abstruse, but I have no idea about the internals of the driver in Mac or Windows. I've asked VEIKK about it but no clear answer yet.

If there is some command to send to the tablet to remap keys that would be a godsend.

Never tried USB sniffing so most likely nothing will come out of this and this will be a spectacular fail, but who knows?

RaSTuS26, 2020-08-11:
Have fun, good luck.

sparshjain265, 2020-08-11:
Ubuntu 20.04.1 LTS, kernel 5.4.0-42-generic, device A15 Pro, using the latest commits on alpha3. 

To compile, I had to add `depmod` in the makefile

Outputs:

xinput list
```
    ↳ VEIKK A15 Pro Pen                       	id=20	[slave  keyboard (3)]
```

sudo xinput test 20
```
no event registered...
```

dmesg -w # on key presses, nothing gets printed
```
[ 1199.936500] input: VEIKK A15 Pro Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:2FEB:0006.0006/input/input25
[ 1200.000791] veikk 0003:2FEB:0006.0006: hidraw2: USB HID v1.00 Mouse [VEIKK.INC A15PRO] on usb-0000:00:14.0-2/input0
[ 1200.000798] veikk 0003:2FEB:0006.0006: VEIKK A15 Pro probed successfully.
[ 1200.001991] veikk 0003:2FEB:0006.0007: DEV RDESC (len 125)
[ 1200.001994] 5 d 9 2 a1 1 85 2 9 20 a1 0 9 42 9 44 9 45 9 3c 9 43 9 44 15 0 25 1 75 1 95 6 81 2 9 32 75 1 95 1 81 2 95 1 81 3 5 1 9 30 9 31 55 d 65 33 26 ff 7f 35 0 46 40 1f 75 10 95 2 81 2 5 d 9 30 26 ff 1f 75 10 95 1 81 2 c0 c0 5 1 9 6 a1 1 85 3 5 7 19 e0 29 e7 15 0 25 1 75 1 95 8 81 2 5 7 19 0 29 ff 26 ff 0 75 8 95 6 81 0 c0 
[ 1200.002471] veikk 0003:2FEB:0006.0007: hidraw3: USB HID v1.00 Keyboard [VEIKK.INC A15PRO] on usb-0000:00:14.0-2/input1
[ 1200.002475] veikk 0003:2FEB:0006.0007: VEIKK A15 Pro probed successfully.
[ 1200.003275] veikk 0003:2FEB:0006.0008: DEV RDESC (len 36)
[ 1200.003277] 6 a ff 9 1 a1 1 85 9 9 2 75 8 95 8 15 0 26 ff 0 81 2 9 3 75 8 95 8 15 0 26 ff 0 91 2 c0 
[ 1303.913372] usb 1-2: USB disconnect, device number 6
[ 1303.914020] veikk 0003:2FEB:0006.0006: device removed successfully.
[ 1303.951163] veikk 0003:2FEB:0006.0007: device removed successfully.
[ 1303.951314] veikk 0003:2FEB:0006.0008: device removed successfully.
[ 1305.667323] usb 1-2: new full-speed USB device number 7 using xhci_hcd
[ 1305.817132] usb 1-2: New USB device found, idVendor=2feb, idProduct=0006, bcdDevice= 0.00
[ 1305.817141] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1305.817146] usb 1-2: Product: A15PRO
[ 1305.817151] usb 1-2: Manufacturer: VEIKK.INC
[ 1305.817155] usb 1-2: SerialNumber: 0000001
[ 1305.821210] veikk 0003:2FEB:0006.0009: DEV RDESC (len 62)
[ 1305.821215] 5 1 9 2 a1 1 85 1 9 1 a1 0 5 9 19 1 29 3 15 0 25 1 95 3 75 1 81 2 95 5 81 1 5 1 9 30 9 31 26 ff 7f 95 2 75 10 81 2 5 d 9 30 26 ff 1f 95 1 75 10 81 2 c0 c0 
[ 1305.821792] input: VEIKK A15 Pro Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:2FEB:0006.0009/input/input27
[ 1305.879718] veikk 0003:2FEB:0006.0009: hidraw2: USB HID v1.00 Mouse [VEIKK.INC A15PRO] on usb-0000:00:14.0-2/input0
[ 1305.879728] veikk 0003:2FEB:0006.0009: VEIKK A15 Pro probed successfully.
[ 1305.881567] veikk 0003:2FEB:0006.000A: DEV RDESC (len 125)
[ 1305.881572] 5 d 9 2 a1 1 85 2 9 20 a1 0 9 42 9 44 9 45 9 3c 9 43 9 44 15 0 25 1 75 1 95 6 81 2 9 32 75 1 95 1 81 2 95 1 81 3 5 1 9 30 9 31 55 d 65 33 26 ff 7f 35 0 46 40 1f 75 10 95 2 81 2 5 d 9 30 26 ff 1f 75 10 95 1 81 2 c0 c0 5 1 9 6 a1 1 85 3 5 7 19 e0 29 e7 15 0 25 1 75 1 95 8 81 2 5 7 19 0 29 ff 26 ff 0 75 8 95 6 81 0 c0 
[ 1305.882467] veikk 0003:2FEB:0006.000A: hidraw3: USB HID v1.00 Keyboard [VEIKK.INC A15PRO] on usb-0000:00:14.0-2/input1
[ 1305.882476] veikk 0003:2FEB:0006.000A: VEIKK A15 Pro probed successfully.
[ 1305.889139] veikk 0003:2FEB:0006.000B: DEV RDESC (len 36)
[ 1305.889144] 6 a ff 9 1 a1 1 85 9 9 2 75 8 95 8 15 0 26 ff 0 81 2 9 3 75 8 95 8 15 0 26 ff 0 91 2 c0 
```

dmesg -w # on pen input
```
[ 2455.959877] veikk 0003:2FEB:0006.0009: in veikk_report: report id 1
[ 2455.963901] veikk 0003:2FEB:0006.0009: in veikk_event: usage 90001 value 0
[ 2455.963903] veikk 0003:2FEB:0006.0009: in veikk_event: usage 90002 value 0
[ 2455.963905] veikk 0003:2FEB:0006.0009: in veikk_event: usage 90003 value 0
[ 2455.963907] veikk 0003:2FEB:0006.0009: in veikk_event: usage 10030 value 17698
[ 2455.963908] veikk 0003:2FEB:0006.0009: in veikk_event: usage 10031 value 18318
[ 2455.963910] veikk 0003:2FEB:0006.0009: in veikk_event: usage d0030 value 0
[ 2455.963911] veikk 0003:2FEB:0006.0009: in veikk_report: report id 1
[ 2455.967862] veikk 0003:2FEB:0006.0009: in veikk_event: usage 90001 value 0
[ 2455.967864] veikk 0003:2FEB:0006.0009: in veikk_event: usage 90002 value 0
[ 2455.967865] veikk 0003:2FEB:0006.0009: in veikk_event: usage 90003 value 0
[ 2455.967867] veikk 0003:2FEB:0006.0009: in veikk_event: usage 10030 value 17698
[ 2455.967868] veikk 0003:2FEB:0006.0009: in veikk_event: usage 10031 value 18318
[ 2455.967870] veikk 0003:2FEB:0006.0009: in veikk_event: usage d0030 value 0
[ 2455.967871] veikk 0003:2FEB:0006.0009: in veikk_report: report id 1
```

evtest devices
[... 26 lines cut ...]
      Resolution     100
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8192
Properties:
  Property type 0 (INPUT_PROP_POINTER)
Testing ... (interrupt to exit)
```
evtest # on key presses, nothing gets printed

evtest # on pen input
```
Event: time 1597166127.510962, type 3 (EV_ABS), code 0 (ABS_X), value 16886
Event: time 1597166127.510962, type 3 (EV_ABS), code 1 (ABS_Y), value 18283
Event: time 1597166127.510962, -------------- SYN_REPORT ------------
Event: time 1597166127.516485, type 3 (EV_ABS), code 0 (ABS_X), value 16902
Event: time 1597166127.516485, type 3 (EV_ABS), code 1 (ABS_Y), value 18301
Event: time 1597166127.516485, -------------- SYN_REPORT ------------
Event: time 1597166127.522484, type 3 (EV_ABS), code 0 (ABS_X), value 16923
Event: time 1597166127.522484, type 3 (EV_ABS), code 1 (ABS_Y), value 18334
Event: time 1597166127.522484, -------------- SYN_REPORT ------------
```

@jlam55555 this looks like the undesired output (I double-checked if I have the latest commit and working in branch v3-alpha)

jlam55555, 2020-08-12:
@sparshjain265 No, this is the desired output, thank you for showing me this. I may not have been clear about this, but since I hadn't mapped the keyboard buttons yet for the A15 Pro (or the A15, VK1560 Pro, and VK2200), no events should show up in `xinput` or `evtest` yet. But there should be output for the button presses in `dmesg`, which is what I'm asking you for now.

I see that dmesg shows events for pen events, but does it also show button events? If it does, please annotate the dmesg output like I mentioned [here](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-671701615).

sparshjain265, 2020-08-12:
@jlam55555 no, dmesg doesn't show button events, only pen events

meelash, 2020-08-15:
Hi all!
I'm using Fedora 32, kernel version is 5.7.14-200.fc32.x86_62 and I have an A30.
Installed kernel libraries and ran `make`, ran with no errors.
On `sudo make install clean` I get the following:
```
make -C /lib/modules/5.7.14-200.fc32.x86_64/build M=/home/admin/Downloads/veikk-linux-driver modules
make[1]: Entering directory '/usr/src/kernels/5.7.14-200.fc32.x86_64'
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/5.7.14-200.fc32.x86_64'
make -C /lib/modules/5.7.14-200.fc32.x86_64/build M=/home/admin/Downloads/veikk-linux-driver modules_install
make[1]: Entering directory '/usr/src/kernels/5.7.14-200.fc32.x86_64'
  INSTALL /home/admin/Downloads/veikk-linux-driver/veikk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.7.14-200.fc32.x86_64
make[1]: Leaving directory '/usr/src/kernels/5.7.14-200.fc32.x86_64'
modprobe veikk
make -C /lib/modules/5.7.14-200.fc32.x86_64/build M=/home/admin/Downloads/veikk-linux-driver clean
make[1]: Entering directory '/usr/src/kernels/5.7.14-200.fc32.x86_64'
  CLEAN   /home/admin/Downloads/veikk-linux-driver/Module.symvers
make[1]: Leaving directory '/usr/src/kernels/5.7.14-200.fc32.x86_64'
[admin@localhost veikk-linux-driver]$ uname -r
5.7.14-200.fc32.x86_64
```
Regarding the SSL errors, I do not have Secure Boot enabled, if that is relevant. In spite of the errors, it appears to be doing something as evidenced by- (1) a leftover mouse pointer stuck on the screen in some position, with a new pointer existing and being in use. (2) `/lib/modules/5.7.14-200.fc32.x86_64/extra/` has a file veikk.ko added to it, (3) modprobe veikk does not return any errors (also doesn't return any other feedback at all).

`xinput list` returns the following:
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-stylus:17                      	id=7	[slave  pointer  (2)]
⎜   ↳ xwayland-eraser:17                      	id=8	[slave  pointer  (2)]
⎜   ↳ xwayland-cursor:17                      	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:17                    	id=6	[slave  keyboard (3)]
```
I gather that's not what we're looking for, as I see the output of other users has an actuall veikk entry. For reference, the only other device connected to my machine is an apple bluetooth keyboard.

Any suggestions of what to test next?

My ultimate goal is to be able to (1) configure one of the stylus buttons to switch to a scroll mode where swiping up and down is the equivalent of a scroll wheel, and (2) Same thing but a zoom in/out mode, with the 1 being the most necessary of the two. What is the prognosis to being able to do something like this using this driver; is this a pipedream?

Thanks!

jlam55555, 2020-08-16:
@meelash 

- [Errors are normal, looks like it's installing properly.](https://github.com/jlam55555/veikk-linux-driver/issues/3)
- The differences you're seeing appear to be because you're using Wayland rather than X (I just looked it up, looks like it's the default for GNOME on Fedora.) I haven't gotten a chance to test anything with Wayland, so I'm not sure how well things should work -- you can try to switch to X11 mode when you log in to test this out (should be one of the icons on the top).
- Your goal is not a pipedream! Again, with `xbindkeys` and something like `xvkbd` the keys should be easily remappable. Just make sure you have the latest version pulled from GitHub and installed, and try some of the debugging steps like mentioned [here](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-671701615). Feel free to paste the output. Hopefully you'll be getting something similar to what RaSTuS26 is getting (also an A30, buttons are reporting the correct events).

meelash, 2020-08-16:
Yep, so, after switching to x11, the output of xinput is:
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ VEIKK.INC A30                           	id=8	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ VEIKK.INC A30 Keyboard                  	id=9	[slave  keyboard (3)]
    ↳ Magic Keyboard                          	id=10	[slave  keyboard (3)]
```
After installing using `sudo make install clean`, I have: 
```⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ VEIKK A30 Pen Pen (0)                   	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Magic Keyboard                          	id=10	[slave  keyboard (3)]
    ↳ VEIKK A30 Pen                           	id=8	[slave  keyboard (3)]
    ↳ VEIKK A30 Keyboard                      	id=9	[slave  keyboard (3)]
```
I notice the key mapping on the pen changed from the default already.

After unplugging/replugging I'm getting the same results as @RaSTuS26 xinput test of the Keyboard. xinput test 8 gives no events registered. xinput test 11 give all three buttons, motion, and pressure.

jlam55555, 2020-08-16:
Holy crap, starting to get something to work with the third, VEIKK proprietary interface after [playing around with USB sniffers](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-671986578) (wasn't expecting to get anything out of it). Didn't get very far but this looks very promising. This should eliminate a lot of the weird mapping issues with overlapping keycodes (e.g., on the A50, A15, and A15 Pro).

For reference, was sniffing using USBlyzer on Windows (trialware) and Wireshark + usbmon on Linux. The basic premise is that I noticed that when the Windows VEIKK driver was installed, the device reported different input reports from the proprietary device, which should be easier to work with. I also noticed that there was a certain set of commands (output reports) sent from the computer to the VEIKK device which seems to toggle this. I'm playing around with the exact nature of the output reports now.

At least, all I've gotten to work with the new system is the A50 Pro keyboard (not the pen yet). Will test on the other devices soon and report back.

jlam55555, 2020-08-16:
@meelash Your output from `xinput list` is a little strange -- I'm not sure why the name is `VEIKK A30 Pen Pen`. There should only be a `VEIKK A30 Pen` and a `VEIKK A30 Keyboard`, which report their respective events (e.g., like what @RaSTuS26 showed. Not sure what's going on there; will investigate after my current USB adventures.

jlam55555, 2020-08-19:
Update on the proprietary driver: a lot of success! The changes are now on the [v3-alpha-ff0a branch](https://github.com/jlam55555/veikk-linux-driver/tree/v3-alpha-ff0a) (ff0a is the usage code of the proprietary HID interface). I'm lucky that between the S640, A50, and VK1560 there's a lot of variability, so I get to test out most of the features (e.g., gesture pad, rotating buttons, regular buttons, etc.).

**It'd be great if we could have people test this, and show the output of `evtest`.** Make sure you're on the new branch! What you should be seeing is the pen working normally, and the buttons and gesture pad outputting <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>Alt</kbd>+<kbd>Meta (Windows key)</kbd>+(<kbd>F1</kbd> through <kbd>F24</kbd>). With the way it's set up now, I *believe* it should work for all devices that are in the driver's device id table, but of course I've only been able to test on my devices (A50, S640, VK1560).

### What was achieved/decided:
- Using the proprietary third HID interface (rather than the two generic interfaces from before)
- Third interface uses a bitmap of keys, much easier to deal with than the pre-programmed key combinations on the generic HID interfaces, so remapping is **much** cleaner/easier
- `xbindkeys` is weird (seems to lose focus of current window when pressing a key combination, so you can't enter text into the current window) so probably will make the v3 configuration tool work with [Autokey](https://github.com/autokey/autokey) (super nice Python tool with broader implications than remapping keys, has [good support](https://github.com/autokey/autokey/wiki/Current-Linux-distributions-shipping-AutoKey) and documentation, and is familiar to people who've used AutoHotkey on Windows (this is the Linux port of it AFAIK)).
- A lot of code restructuring and cleanup. Abolishment of the short-lived, terribly-confusing pseudo-usages!

### Brief overview of changes in v3-alpha-ff0a
You don't have to read this, but here's a brief overview of what was accomplished so far (I'll probably write another blog post on this when things get a little more stable):
- Sniffed USB packets while testing the pen/pressing buttons on Windows with the driver installed, saw that the events being sent were different than the ones received on Linux. (They were being sent with report ID 9, which is (you can see this with the output of `usbhid-dump`)). In these, the keys weren't being sent as HID usages, but rather in a bitmap (which is much more compact and logical; the first button corresponded to the first bit, the second the second, etc.).
- Sniffed USB packets when plugging in the device on Windows (with driver) vs. without driver (also Linux without driver), found that there were some 9-byte packets being sent to the tablet only when the Windows driver was being used:
    - `0x09 0x01 0x04 0x00 0x00 0x00 0x00 0x00 0x00`
    - `0x09 0x02 0x02 0x00 0x00 0x00 0x00 0x00 0x00`
    - `0x09 0x03 0x04 0x00 0x00 0x00 0x00 0x00 0x00`

    After some messing around, I discovered that if I sent these to the device (using `hid_dev->ll_driver->output_report`, this would then switch the internal interfaces over to sending events over the proprietary device. In particular, the first switches the pen over, the second the keyboard, and the latter the gesture pad (if applicable).
    - It also turns out that if you send them all at roughly the same time, weird issues may pop up. (My guess is that the VEIKK tablet cannot process the incoming reports very quickly.) So I had to put the `output_report` calls in a workqueue at different delays, which solved the issue. (The length of the delays is still something to experiment with.)
- It turns out that the proprietary interface emits devices with three different "sub-report types": the pen, keyboard (if applicable), and gesture pad (if applicable). To keep things simple, I gave each one of these its own `struct input_dev`, so now a "Gesture Pad" may show up in `xinput list` or `evtest`.
- Also a whole lot of internal restructuring/refactoring of functions. With the three report types, I decided to make three input types rather than two (pen, keyboard [if applicable] &rightarrow; pen, buttons [if applicable], and gesture-pad [if applicable]) and creating more functions to set these up separately. It turns out that this makes the code a lot more similar to Wacom's driver now, which is kind of cool; feels like it's bringing everything full circle, and now I can more fully understand/appreciate a lot of the pieces happening there.
- As part of the refactoring, I rewrote some of the functions to have more centralized exit points and made shorter functions (`veikk_probe` got broken down into shorter functions), both as per the kernel coding style guide.
- This update greatly simplified a lot of the remapping code and the whole idea of linking the HID devices, so this gets rid of a lot of the old code (e.g., storing the `struct veikk_device`s in a linked list, all the pseudo-usage stuff, etc.).

I know I was messing around a while with the idea of "pseudo-usages" to make the interface consistent across VEIKK devices, and I know some (e.g., @sparshjain265) were playing around with the pseudousages table. This update abolishes that darned complicated mess. Sorry if anyone spent any time messing around with those, but this should make things much easier going forward.

RaSTuS26, 2020-08-20:
@jlam55555 Testing alpha3-ff0a

evtest devices:
```
/dev/input/event23:     VEIKK A30 Pen
/dev/input/event24:     VEIKK A30 Keyboard
/dev/input/event25:     VEIKK A30 Gesture Pad
```

buttons test 1 to 4 in order (event24):
```
Event: time 1597899507.975070, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597899507.975070, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597899507.975070, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597899507.975070, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597899507.975070, type 1 (EV_KEY), code 59 (KEY_F1), value 1
Event: time 1597899507.975070, -------------- SYN_REPORT ------------
^[O6PEvent: time 1597899508.081955, type 1 (EV_KEY), code 59 (KEY_F1), value 2
Event: time 1597899508.081955, -------------- SYN_REPORT ------------
Event: time 1597899508.126020, type 1 (EV_KEY), code 59 (KEY_F1), value 2
Event: time 1597899508.126020, -------------- SYN_REPORT ------------
Event: time 1597899508.126020, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597899508.126020, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597899508.126020, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597899508.126020, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597899508.126020, type 1 (EV_KEY), code 59 (KEY_F1), value 0
Event: time 1597899508.126020, -------------- SYN_REPORT ------------
Event: time 1597899511.351056, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597899511.351056, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597899511.351056, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597899511.351056, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597899511.351056, type 1 (EV_KEY), code 60 (KEY_F2), value 1
Event: time 1597899511.351056, -------------- SYN_REPORT ------------
^[O6QEvent: time 1597899511.439055, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597899511.439055, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597899511.439055, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597899511.439055, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597899511.439055, type 1 (EV_KEY), code 60 (KEY_F2), value 0
Event: time 1597899511.439055, -------------- SYN_REPORT ------------
Event: time 1597899513.625037, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597899513.625037, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597899513.625037, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597899513.625037, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597899513.625037, type 1 (EV_KEY), code 61 (KEY_F3), value 1
Event: time 1597899513.625037, -------------- SYN_REPORT ------------
^[O6REvent: time 1597899513.726003, type 1 (EV_KEY), code 61 (KEY_F3), value 2
Event: time 1597899513.726003, -------------- SYN_REPORT ------------
Event: time 1597899513.770017, type 1 (EV_KEY), code 61 (KEY_F3), value 2
Event: time 1597899513.770017, -------------- SYN_REPORT ------------
Event: time 1597899513.810015, type 1 (EV_KEY), code 61 (KEY_F3), value 2
Event: time 1597899513.810015, -------------- SYN_REPORT ------------
Event: time 1597899513.810015, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597899513.810015, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597899513.810015, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597899513.810015, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597899513.810015, type 1 (EV_KEY), code 61 (KEY_F3), value 0
Event: time 1597899513.810015, -------------- SYN_REPORT ------------
Event: time 1597899515.133030, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597899515.133030, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597899515.133030, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597899515.133030, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597899515.133030, type 1 (EV_KEY), code 62 (KEY_F4), value 1
Event: time 1597899515.133030, -------------- SYN_REPORT ------------
^[O6SEvent: time 1597899515.211044, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597899515.211044, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597899515.211044, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597899515.211044, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597899515.211044, type 1 (EV_KEY), code 62 (KEY_F4), value 0
Event: time 1597899515.211044, -------------- SYN_REPORT ------------
```
[... 17 lines cut ...]
Event: time 1597900167.497215, type 1 (EV_KEY), code 191 (KEY_F21), value 0
Event: time 1597900167.497215, -------------- SYN_REPORT ------------
Event: time 1597900168.075248, type 1 (EV_KEY), code 194 (KEY_F24), value 1
Event: time 1597900168.075248, -------------- SYN_REPORT ------------
Event: time 1597900168.109247, type 1 (EV_KEY), code 194 (KEY_F24), value 0
Event: time 1597900168.109247, -------------- SYN_REPORT ------------
Event: time 1597900168.675212, type 1 (EV_KEY), code 191 (KEY_F21), value 1
Event: time 1597900168.675212, -------------- SYN_REPORT ------------
Event: time 1597900168.699243, type 1 (EV_KEY), code 191 (KEY_F21), value 0
Event: time 1597900168.699243, -------------- SYN_REPORT ------------
[... cut ...]
Event: time 1597900169.359245, -------------- SYN_REPORT ------------
Event: time 1597900169.383208, type 1 (EV_KEY), code 194 (KEY_F24), value 0
Event: time 1597900169.383208, -------------- SYN_REPORT ------------
Event: time 1597900170.241240, type 1 (EV_KEY), code 191 (KEY_F21), value 1
Event: time 1597900170.241240, -------------- SYN_REPORT ------------
Event: time 1597900170.265198, type 1 (EV_KEY), code 191 (KEY_F21), value 0
Event: time 1597900170.265198, -------------- SYN_REPORT ------------
```

If you require any other output let me know Jonathon.

RaSTuS26, 2020-08-20:
@jlam55555 Playing with the Touchpad, all directions work except "right"

xbindkeys --key for "left":
```
m:0x10 + c:201
Mod2 + XF86TouchpadOff
```

... for "right":
```
m:0x10 + c:202
Mod2 + NoSymbol
```

... for "up":
```
m:0x10 + c:199
Mod2 + XF86TouchpadToggle
```

... for "down":
```
m:0x10 + c:200
Mod2 + XF86TouchpadOn
```

Mapped them in GIMP to scroll-<direction>, but I have no scroll-right.

Also was able to map buttons 1-4 to certain tool selections in GIMP no problems.

jlam55555, 2020-08-20:
@RaSTuS26 Ok, now I realize that I forgot to send the modifier keys on the gesture pad events :/ They're supposed to have the Ctrl+Alt+Shift+Meta like the other keys.

Hmm, so I tried to map the gesture pad keys to the higher-numbered function keys, but it looks like their keysyms have been overwritten with other things.

How did you map them in GIMP? Was it through the keyboard shortcuts menu? Or was it through xbindkeys?

RaSTuS26, 2020-08-20:
@jlam55555, I used GIMP's keyboard shortcuts menu.

sparshjain265, 2020-08-20:
Alright @jlam55555 this time things are much better :clinking_glasses: 
The 12 main buttons and the dial are producing the following output in the evtest. However, the button placed in the center of the dial does not produce any output in the evtest.

`sudo evtest`
```
/dev/input/event23:	VEIKK A15 Pro Pen
/dev/input/event24:	VEIKK A15 Pro Keyboard
/dev/input/event25:	VEIKK A15 Pro Gesture Pad
Select the device event number [0-25]: 24
```
Comment: A15 Pro doesn't have a Gesture Pad but it still shows! Not an issue, just noticed.

buttons test (1 - 12) in the order you listed [here](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-671701615):

BTN_1:
```
Event: time 1597924614.449884, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597924614.449884, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597924614.449884, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597924614.449884, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597924614.449884, type 1 (EV_KEY), code 59 (KEY_F1), value 1
Event: time 1597924614.449884, -------------- SYN_REPORT ------------
Event: time 1597924614.535354, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597924614.535354, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597924614.535354, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597924614.535354, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597924614.535354, type 1 (EV_KEY), code 59 (KEY_F1), value 0
Event: time 1597924614.535354, -------------- SYN_REPORT ------------
^[[1;8P
```

BTN_2:
```
Event: time 1597924662.488297, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597924662.488297, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597924662.488297, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597924662.488297, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597924662.488297, type 1 (EV_KEY), code 60 (KEY_F2), value 1
Event: time 1597924662.488297, -------------- SYN_REPORT ------------
Event: time 1597924662.574343, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597924662.574343, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597924662.574343, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597924662.574343, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597924662.574343, type 1 (EV_KEY), code 60 (KEY_F2), value 0
Event: time 1597924662.574343, -------------- SYN_REPORT ------------
^[[1;8Q
```

BTN_3:
```
Event: time 1597924700.833308, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597924700.833308, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597924700.833308, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597924700.833308, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597924700.833308, type 1 (EV_KEY), code 61 (KEY_F3), value 1
Event: time 1597924700.833308, -------------- SYN_REPORT ------------
Event: time 1597924700.936984, type 1 (EV_KEY), code 61 (KEY_F3), value 2
Event: time 1597924700.936984, -------------- SYN_REPORT ------------
Event: time 1597924700.936984, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597924700.936984, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597924700.936984, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597924700.936984, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597924700.936984, type 1 (EV_KEY), code 61 (KEY_F3), value 0
Event: time 1597924700.936984, -------------- SYN_REPORT ------------
^[[1;8R
```

BTN_4:
```
Event: time 1597924750.632255, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
[... 166 lines cut ...]
Event: time 1597925167.689039, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597925167.689039, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597925167.689039, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597925167.689039, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597925167.689039, type 1 (EV_KEY), code 185 (KEY_F15), value 0
Event: time 1597925167.689039, -------------- SYN_REPORT ------------
```

Dial Test (Rotate Counter-Clockwise):
```
Event: time 1597925201.637875, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1597925201.637875, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1597925201.637875, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1597925201.637875, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1597925201.637875, type 1 (EV_KEY), code 184 (KEY_F14), value 1
Event: time 1597925201.637875, -------------- SYN_REPORT ------------
Event: time 1597925201.639787, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1597925201.639787, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1597925201.639787, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1597925201.639787, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1597925201.639787, type 1 (EV_KEY), code 184 (KEY_F14), value 0
Event: time 1597925201.639787, -------------- SYN_REPORT ------------
```

Dial Test (Center Button): No Output!!!

RaSTuS26, 2020-08-20:
@jlam55555 Just noticed the screen mapping of the pen on the tablet is wrong.  In Y-axis things seem ok, but in X-axis left side is correct position, but reaches right side of the screen when the pen is positioned only 2/3 (approx) across, then goes off screen when moved past that.

sparshjain265, 2020-08-20:
+1 [this](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-677636730) is happening with me too

jlam55555, 2020-08-20:
@sparshjain265 Thanks for the detailed output. I'm not sure about the center wheel. The center wheel on my VK1560 reports correctly (there are seven regular buttons, and it registers as the eighth button). Extrapolating from that, I think that your center button should report as the 13th button, and I thought I accounted for it. So I'm not exactly sure what's going on in there; you may get some more information by seeing the `dmesg` output when you press that center wheel button.

@sparshjain265 @RaSTuS26 I didn't realize the wrong screen mapping until you pointed it out. It looks like the proprietary interface output different maximums than the generic interfaces. It doesn't seem to match up with the report descriptor (i.e., `usbhid-dump`, which seems to show two sets of logical maximums: 32768x32768 for the generic interfaces, and one set indicating physical dimensions), so we'll probably have to do guess-and-check by seeing what `evtest` reports when you go to the bottom-right of your screen. Unfortunately, unless we can find a better way, this will just have to be done manually for each device.

This is what I've found with my devices (will put in the next commit):
- VK1560: x_max: 34420, y_max: 19360 (aspect ratio: 16/9)
- S640: x_max: 30480, y_max: 20320 (aspect ratio: 3/2)
- A50: x_max: 50800, y_max: 30480 (aspect ratio: 5/3)

These numbers do seem to have sensible aspect ratios that match their dimensions, e.g., the S640 drawing area is 6"x4" (aspect ratio 3/2), the A50 is 10"x6", the VK1560 is the same aspect ratio as a 1080p screen. This is as opposed to the generic devices, which were usually 32768x32768 despite the aspect ratios.

If you could provide the maximum values (or create PR's with them) I'll gladly welcome those.

(On a side note, I'm starting to wonder if a little bit over overscan/underscan (not sure if these are the right words) near the edges of the screen might be desirable. If the device is exactly lined up with the edges of the screen, sometimes it's hard to press buttons near the edges of the screen. I'm thinking about making this an option in the configuration GUI.)

jlam55555, 2020-08-20:
@RaSTuS26 I see what you mean about GIMP now. I might have to choose a range of somewhat-generic keys that are more common than F13-F24. Perhaps the keypad (KP) keys will do? (This is what I had in mind originally). E.g., something like (with the Control+Shift+Meta+Alt modifiers on top):

- Buttons 1-12: F1-F12
- Wheel center/left/right: KP0-2
- Gesture pad two-finger tap, up, down, left, right: KP5-9

(This leaves KP3, KP4 open for additional keys in the future if necessary)

How does this scheme sound?

RaSTuS26, 2020-08-20:
@jlam55555 In the alpha3 from 2020-08-11 the screen mapping works correctly and the mouse cursor tracks pen position. In ff0a the mouse cursor doesn't move. 

Correction, if I unplug the tablet then replug with ff0a, then the mouse cursor does track the pen movement, so seems it needs to be reinitalized to work properly, mapping is still wrong though.

evtest on pen at top left (truncated due to size)
```
Event: time 1597931175.935684, type 3 (EV_ABS), code 0 (ABS_X), value 275
Event: time 1597931175.935684, type 3 (EV_ABS), code 1 (ABS_Y), value 928
Event: time 1597931175.935684, -------------- SYN_REPORT ------------
Event: time 1597931175.957649, type 3 (EV_ABS), code 1 (ABS_Y), value 950
Event: time 1597931175.957649, -------------- SYN_REPORT ------------
Event: time 1597931175.965682, type 3 (EV_ABS), code 1 (ABS_Y), value 970
Event: time 1597931175.965682, -------------- SYN_REPORT ------------
Event: time 1597931175.973655, type 3 (EV_ABS), code 1 (ABS_Y), value 983
Event: time 1597931175.973655, -------------- SYN_REPORT ------------
Event: time 1597931175.985666, type 3 (EV_ABS), code 0 (ABS_X), value 280
Event: time 1597931175.985666, -------------- SYN_REPORT ------------
Event: time 1597931175.989665, type 3 (EV_ABS), code 0 (ABS_X), value 285
Event: time 1597931175.989665, -------------- SYN_REPORT ------------
Event: time 1597931175.993668, type 3 (EV_ABS), code 0 (ABS_X), value 290
Event: time 1597931175.993668, -------------- SYN_REPORT ------------
Event: time 1597931175.997679, type 3 (EV_ABS), code 0 (ABS_X), value 296
Event: time 1597931175.997679, -------------- SYN_REPORT ------------
Event: time 1597931176.001659, type 3 (EV_ABS), code 0 (ABS_X), value 301
Event: time 1597931176.001659, -------------- SYN_REPORT ------------
Event: time 1597931176.005705, type 3 (EV_ABS), code 0 (ABS_X), value 304
Event: time 1597931176.005705, -------------- SYN_REPORT ------------
Event: time 1597931176.009659, type 3 (EV_ABS), code 0 (ABS_X), value 306
Event: time 1597931176.009659, type 3 (EV_ABS), code 1 (ABS_Y), value 971
Event: time 1597931176.009659, -------------- SYN_REPORT ------------
Event: time 1597931176.013657, type 3 (EV_ABS), code 0 (ABS_X), value 307
Event: time 1597931176.013657, type 3 (EV_ABS), code 1 (ABS_Y), value 963
Event: time 1597931176.013657, -------------- SYN_REPORT ------------
Event: time 1597931176.017654, type 3 (EV_ABS), code 0 (ABS_X), value 309
Event: time 1597931176.017654, -------------- SYN_REPORT ------------
Event: time 1597931176.021668, type 3 (EV_ABS), code 0 (ABS_X), value 314
Event: time 1597931176.021668, type 3 (EV_ABS), code 1 (ABS_Y), value 953
Event: time 1597931176.021668, -------------- SYN_REPORT ------------
Event: time 1597931176.025703, type 3 (EV_ABS), code 0 (ABS_X), value 317
Event: time 1597931176.025703, type 3 (EV_ABS), code 1 (ABS_Y), value 945
Event: time 1597931176.025703, -------------- SYN_REPORT ------------
Event: time 1597931176.029709, type 3 (EV_ABS), code 0 (ABS_X), value 322
Event: time 1597931176.029709, type 3 (EV_ABS), code 1 (ABS_Y), value 936
Event: time 1597931176.029709, -------------- SYN_REPORT ------------
Event: time 1597931176.033650, type 3 (EV_ABS), code 0 (ABS_X), value 328
Event: time 1597931176.033650, type 3 (EV_ABS), code 1 (ABS_Y), value 928
Event: time 1597931176.033650, -------------- SYN_REPORT ------------
Event: time 1597931176.037655, type 3 (EV_ABS), code 0 (ABS_X), value 336
Event: time 1597931176.037655, type 3 (EV_ABS), code 1 (ABS_Y), value 916
Event: time 1597931176.037655, -------------- SYN_REPORT ------------
Event: time 1597931176.041708, type 3 (EV_ABS), code 0 (ABS_X), value 346
Event: time 1597931176.041708, type 3 (EV_ABS), code 1 (ABS_Y), value 903
Event: time 1597931176.041708, -------------- SYN_REPORT ------------
Event: time 1597931176.045685, type 3 (EV_ABS), code 0 (ABS_X), value 357
Event: time 1597931176.045685, type 3 (EV_ABS), code 1 (ABS_Y), value 888
Event: time 1597931176.045685, -------------- SYN_REPORT ------------
Event: time 1597931176.049654, type 3 (EV_ABS), code 0 (ABS_X), value 370
Event: time 1597931176.049654, type 3 (EV_ABS), code 1 (ABS_Y), value 873
Event: time 1597931176.049654, -------------- SYN_REPORT ------------
Event: time 1597931176.053659, type 3 (EV_ABS), code 0 (ABS_X), value 382
Event: time 1597931176.053659, type 3 (EV_ABS), code 1 (ABS_Y), value 856
Event: time 1597931176.053659, -------------- SYN_REPORT ------------
Event: time 1597931176.057659, type 3 (EV_ABS), code 0 (ABS_X), value 395
Event: time 1597931176.057659, type 3 (EV_ABS), code 1 (ABS_Y), value 838
Event: time 1597931176.057659, -------------- SYN_REPORT ------------
Event: time 1597931176.061707, type 3 (EV_ABS), code 0 (ABS_X), value 409
Event: time 1597931176.061707, type 3 (EV_ABS), code 1 (ABS_Y), value 819
Event: time 1597931176.061707, -------------- SYN_REPORT ------------
Event: time 1597931176.065659, type 3 (EV_ABS), code 0 (ABS_X), value 425
Event: time 1597931176.065659, type 3 (EV_ABS), code 1 (ABS_Y), value 801
[... 7 lines cut ...]
```
Event: time 1597931406.513871, type 3 (EV_ABS), code 0 (ABS_X), value 50740
Event: time 1597931406.513871, type 3 (EV_ABS), code 1 (ABS_Y), value 31306
Event: time 1597931406.513871, -------------- SYN_REPORT ------------
Event: time 1597931406.533892, type 3 (EV_ABS), code 0 (ABS_X), value 50662
Event: time 1597931406.533892, type 3 (EV_ABS), code 1 (ABS_Y), value 31142
[... cut ...]
Event: time 1597931412.699845, type 3 (EV_ABS), code 0 (ABS_X), value 50471
Event: time 1597931412.699845, -------------- SYN_REPORT ------------
Event: time 1597931412.703815, type 3 (EV_ABS), code 0 (ABS_X), value 50447
Event: time 1597931412.703815, type 3 (EV_ABS), code 1 (ABS_Y), value 30333
Event: time 1597931412.703815, -------------- SYN_REPORT ------------
Event: time 1597931412.707811, type 3 (EV_ABS), code 0 (ABS_X), value 50423
Event: time 1597931412.707811, -------------- SYN_REPORT ------------
```

RaSTuS26, 2020-08-20:
> @RaSTuS26 I see what you mean about GIMP now. I might have to choose a range of somewhat-generic keys that are more common than F13-F24. Perhaps the keypad (KP) keys will do? (This is what I had in mind originally). E.g., something like (with the Control+Shift+Meta+Alt modifiers on top):
> 
>     * Buttons 1-12: F1-F12
> 
>     * Wheel center/left/right: KP0-2
> 
>     * Gesture pad two-finger tap, up, down, left, right: KP5-9
> 
> 
> (KP3, KP4 will be available for additional keys in the future if necessary)
> 
> How does this scheme sound?

That sounds fine, the [modifiers]+F1-F4 are mapping fine.  It's only the gesture pad playing up, so if they could be mapped to [modifiers]+Keypad Numbers that hopefully would work correctly.

RaSTuS26, 2020-08-20:
@jlam55555

> @RaSTuS26
> 
>     * Sometimes unplugging and re-plugging is needed after compilation, which has been a known issue for a while.
> 
>     * It looks like your A30 might have the same logical dimensions as the A50 (50800x30480). What happens if you change the `x_max` and `y_max` in `veikk_model_0x0002` to these values? Is the mapping corrected?

Works much better, left and bottom are fine, but top is short by about 5-6 pixels and right is short by about 12-14 pixels.

sparshjain265, 2020-08-20:
@jlam55555 following is the dmesg output when I press the wheel center button - 
```
[14245.799597] veikk 0003:2FEB:0006.000E: raw report size: 9
[14245.799602] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 66
[14245.799604] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 1
[14245.799604] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 1
[14245.799605] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.799606] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.799607] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 1
[14245.799608] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.799608] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.799609] veikk 0003:2FEB:0006.000E: in veikk_report: report id 9
[14245.903784] veikk 0003:2FEB:0006.000E: raw report size: 9
[14245.903806] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 66
[14245.903811] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 1
[14245.903815] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.903819] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.903823] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.903827] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 1
[14245.903832] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.903836] veikk 0003:2FEB:0006.000E: in veikk_event: usage ff0a0002 value 0
[14245.903840] veikk 0003:2FEB:0006.000E: in veikk_report: report id 9
```

evtest output with max observed value of y:
```
Event: time 1597932010.804731, type 3 (EV_ABS), code 1 (ABS_Y), value 31750
Event: time 1597932010.804731, -------------- SYN_REPORT ------------
```

evtest output with max observed value of x:
```
Event: time 1597932141.693986, type 3 (EV_ABS), code 0 (ABS_X), value 50800
Event: time 1597932141.693986, -------------- SYN_REPORT ------------
```

drawing area for a15 pro is also 10"x6" and 30480 should be the max value for y but I notice that my tab is registering pen input beyond the designated drawing area, which explains the observed max value of y as 31750. 
Also, my tab is registering pen inputs slightly beyond the designated drawing area in the top/left/bottom boundary but is precise (at least by naked eye observation) in the right boundary.

sparshjain265, 2020-08-20:
> (On a side note, I'm starting to wonder if a little bit over overscan/underscan (not sure if these are the right words) near the edges of the screen might be desirable for the non-monitor tablets. If the device is exactly lined up with the edges of the screen, sometimes it's hard to press buttons near the edges of the screen. I'm thinking about making this an option in the configuration GUI.)

@jlam55555 This sounds like a good idea, I believe instead of using 0 as min, some small positive value can be used (can be added as a parameter in the model structure?) and instead of using actual max as the max, something slightly less than max can be used in `input_set_abs_params()` function calls. Alternatively, the inverse (in some sense) can be done in `input_abs_set_res()` function calls.

jlam55555, 2020-08-20:
@sparshjain265 About the center wheel button: I just realized that I tried to do this:

- `buttons_state` is 16 bits
- `buttons_state | 0x10000`

I'm kind of stupid sometimes. (Either `buttons_state` has to be expanded to 32 bits or I have to treat that specially, since the A15 Pro is the only tablet that uses that 17th bit. Don't worry though, this is an easy fix).

jlam55555, 2020-08-20:
@RaSTuS26 By "short," it sounds like your situation is very similar to @sparshjain265's and the maximum is slightly above 30480. Does setting the `x_max` to 31750 work correctly for you?

(If that's the case, it's an 8/5 aspect ratio, which is strange because it doesn't match the drawing area physical aspect ratio.)

RaSTuS26, 2020-08-20:
@jlam55555, Still didn't reach right side properly.  I'm now using "x_max = 50600, .y_max = 31750" and it covers the whole perfectly.

jlam55555, 2020-08-20:
Another thing I'm starting to notice: when remapping keys on the buttons, the modifiers are great, but on the gesture pad, it doesn't work so well with Autokey. My guess is that it's because the gesture pad events are hardrepeated, which causes all of the modifiers to be rapidly switched on and off, which probably creates problems for autokey. Either I have to find out how to debounce the four gestures or switch them to key that wouldn't require modifiers. More on that later.

sparshjain265, 2020-08-20:
@jlam55555 I think the 50800x31750 (8/5 aspect ratio) is when we include the inputs received from outside the designated drawing area, and if we tried to note the min/max values of both x and y, we should get a proper 5/3 aspect ratio.

About that buttons_state, I think making it 32 bit is a better option considering new devices in the future may have more buttons.

RaSTuS26, 2020-08-20:
@sparshjain265, 50600 works fine, when it's 50800 it doesn't reach.  My current working figures are 50600x31750.

sparshjain265, 2020-08-20:
@RaSTuS26 oh alright, I'm sorry. Yes, I meant to say less than 50800 (which 50600 is, so yay, my hunch was right).

RaSTuS26, 2020-08-20:
@sparshjain265, How does  50600x31750 work for you?

sparshjain265, 2020-08-20:
@RaSTuS26 well, I reach the right end of my screen slightly before I reach the right end of the drawing area in my tab. With 50800, it's seemingly exact.

RaSTuS26, 2020-08-20:
@sparshjain265, Very strange, even when using unaltered ff0a version my max x was 50576 on same 10x6 surface. Probably different manufacturing tolerances, wouldn't suprise me if even two of the same model gave slightly different results.

sparshjain265, 2020-08-20:
@RaSTuS26 100% agreed

jlam55555, 2020-08-21:
I did kind of get the gesture pad repeating to work with autokey by manually doing some throttling, which is okay but not great...

I'm really tempted to write my own little input mapping tool daemon to integrate with the configuration tool. `xbindkeys` doesn't have the ability to spawn keyboard shortcuts, autokey is kind of slow, and both methods end up creating long keyboard combinations.

Also, `libevdev` looks like a pretty easy-to-use library. I know I said in the blog post I would try to avoid this but it looks like a good option after considering the others.

jlam55555, 2020-08-24:
@RaSTuS26 in v1-v2 use the generic devices, which mostly seem to output a maximum of 32768, but switching to use the proprietary devices seem to report a different maximum.

As for why, I don't know. And since it's proprietary, I don't think there's likely any documentation on it.

RaSTuS26, 2020-08-27:
> @jlam55555, Using 3-alpha-ff0a_2020-08-20 and changing x/y to 50600x31750, works much the same as 2020-08-19 version. Gesture right still doesn't seem to work in GIMP, but does work on evtest.
> 
> evtest output for up, down, left, right gestures:
> 
> ```
> Event: time 1598074793.836334, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
> Event: time 1598074793.836334, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
> Event: time 1598074793.836334, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
> Event: time 1598074793.836334, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
> Event: time 1598074793.836334, type 1 (EV_KEY), code 191 (KEY_F21), value 1
> 
> Event: time 1598074799.362341, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
> Event: time 1598074799.362341, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
> Event: time 1598074799.362341, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
> Event: time 1598074799.362341, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
> Event: time 1598074799.362341, type 1 (EV_KEY), code 192 (KEY_F22), value 1
> 
> Event: time 1598074807.898263, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
> Event: time 1598074807.898263, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
> Event: time 1598074807.898263, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
> Event: time 1598074807.898263, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
> Event: time 1598074807.898263, type 1 (EV_KEY), code 193 (KEY_F23), value 1
> 
> Event: time 1598074810.076318, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
> Event: time 1598074810.076318, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
> Event: time 1598074810.076318, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
> Event: time 1598074810.076318, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
> Event: time 1598074810.076318, type 1 (EV_KEY), code 194 (KEY_F24), value 1
> ```
> 
> EDIT:
> This is how they're mapped in GIMP
> 
> ```
> (gtk_accel_path "<Actions>/view/view-scroll-up" "<Primary><Shift><Super>XF86TouchpadToggle")
> (gtk_accel_path "<Actions>/view/view-scroll-down" "<Primary><Shift><Super>XF86TouchpadOn")
> (gtk_accel_path "<Actions>/view/view-scroll-left" "<Primary><Shift><Super>XF86TouchpadOff")
> Right gesture not being interpreted
> ```

@jlam55555, I'm still a little concerned about this.  Why is GIMP not getting the correct kbd mapping as shown in evtest? It's like it interprets those shortcuts totally differently to the system.

jlam55555, 2020-08-27:
@RaSTuS26 F13-F24 are not common keys (at least nowadays, not sure in the past), so it looks like people didn't care to be consistent between the naming of the input codes (defined in `/usr/include/linux/input-event-codes.h`) and the X11 keysyms (defined in `/usr/include/X11/keysymdef.h`) for these keys. Again, this is part of the reason I wanted to move to using the configuration tool, because then we don't have to mess around with all these modifier keys (which is moving along nicely, got a basic mapping script to work).

RaSTuS26, 2020-08-27:
@jlam55555, much thanks for the explanation Jonathan.

Can't wait to see what the configuration tool offers, it'll certainly be a welcome addition.

RaSTuS26, 2020-09-17:
@jlam55555, I hope you're ok, haven't heard from you here or on twitter for a while.

jlam55555, 2020-09-19:
@RaSTuS26 Thanks for the concern, got hit hard with school. I really do want to get the configuration tool done so I can finally mark this project complete with all the originally intended features, but I don't know when I have any reasonable amount of time to finish this.

RaSTuS26, 2020-09-19:
@jlam55555, good to hear you're OK,  School is most important of course, best of luck with that, hope you go well there.

jlam55555, 2021-06-26:
With a free weekend and some extra knowledge over the last year, I think I am going to split off the v3 project into a new repo, and continue support for the v2 driver here. I've been very busy with studies/school projects/Master's thesis, but I've dedicated this weekend to a 48-hour hackathon-style to work on this driver and this driver only.

---

I've already written about what the v3 driver brings and how we can get which buttons are pressed, but there were still a number of difficulties:
- How do we get these values to the userspace? As I [wrote in this post](http://eis.lambdalambda.ninja/posts/button-mapping-journeys), there are issues with each solution.
- How do we communicate with the configuration tool? In v2 we are using sysfs config, which is very unwieldy. It also doesn't support the ["separation of mechanism and policy" principle](https://en.wikipedia.org/wiki/Separation_of_mechanism_and_policy).

In the v3 driver, I took out most of the "policy" out of the driver (in preparation for the move to userspace configuration), and in the v3-ff0a branch, I figured out how to get usable values for the button mappings. The code for the v3-ff0a branch is basically finished, but I never figured out a clean userspace/config solution. But I now have a plan in mind for a system:

- driver would basically be as it is in v3-ff0a currently, with some minor changes
- driver would send a predetermined set of event codes, like BTN_0 through BTN_9, BTN_WHEEL, etc. to userspace
- mapping with xbindkeys is a little wacky (e.g., uses keysyms rather than keycodes, only supports X keysyms (which are different and fewer than evdev keycodes)), so I will write my own buttom mapper with evdev
- we need to have a daemon that listens on all VEIKK device events, which means a separate thread for each evdev device (button/gesture pad -- can use X mapping for pen input)
- need to use udev to listen to new VEIKK devices being added, so that I can spin up a new mapping thread for it
- config tool will need to communicate with the current daemon, so I will use a dbus connection for IPC
- python daemon will run as a systemd service (I don't imagine the number of non-systemd users to be large, but then they can just put the python script in some boot script)

I thought the above would all be too complicated and involve complicated systemd, evdev, udev, dbus, and other configs/bindings. Luckily, it seems that Python makes it easy to deal with the latter three, and then I just need to figure out how to package a Python project for distribution. I usually don't like Python because it always seems to have versioning issues, but the C/C++ APIs involve too much boilerplate code and these Python packages are very mature.

There will be one python script (veikk) for the daemon, and another (veikkctl) that will be a CLI to access the daemon. I will only be writing a CLI for this (I suck at writing GUI applications, as you can tell with the config application for the v2 driver), but anyone is free to adapt a GUI for it.

---

You might also ask, why a new repo? Is that just abandoning all the existing issues with the current repo and config tool?

Kind of? I don't want to confuse users with different versions of the same software in the same repo, especially when the current one has been in use for a little while. It's best to distinguish the versions explicitly here. I will call the new repos veikk-linux-driver-v3 and veikk-linux-config-v3, and will create them shortly.

I will also update the README with the current status of the project -- it is NOT being actively maintained right now because of lack of time, but v3 is still in the works. If and when the v3 version becomes stable in the near future, I will update the README again to reflect that again.

I will update this post shortly with the repo information.

cheeseballz, 2021-08-07:
Hello. I am sorry ,i am just a normie and new to github and stuff,does this driver support vk640?