jlam55555/veikk-linux-driver #41: Installing under Debian and derivatives: suggested patch

juliangilbey, 2020-08-11:
Thanks for writing this driver!

Since Debian and derivatives don't have a System.map file, the install step fails.  As has been pointed out in other bug reports, running depmod manually solves it.

So could I suggest that the `modules_install` line in Makefile is modified to read:

```
make -C $(BUILD_DIR) M=$(CURDIR) modules_install || depmod
```

That should make it build on more systems without manual tweaking.

Best wishes!

jlam55555, 2020-08-11:
That's smart, will test it out on both Debian and non-Debian systems to make sure it works smoothly.

Thanks

juliangilbey, 2020-08-11:
Or alternatively, place a "-" in front of the modules_install line so it doesn't interrupt the installation if it fails.

juliangilbey, 2020-08-11:
Ah, I was looking at the current version of the Makefile, which already has depmod on its own line.  I assumed, perhaps incorrectly, that this is intended for those systems where `modules_install` fails.  But if it's needed for some separate reason, then using `... modules_install || depmod` won't work, as the `depmod` command will not be called if `modules_install` succeeds.  If that is the case, then a `-` in from the the `modules_install` line will be better, as it will allow the rest of the install target to run on a Debian-based system.

Best wishes!

jlam55555, 2020-08-12:
Sorry, I'm not familiar with the `-` syntax, could you be a little more explicit with how this might be used? (Is it a bash syntax or a Makefile syntax?)

(I'm only using Arch Linux these past few days so haven't been able to test the exact behavior of the depmod stuff for Debian-like OSes. Thanks a lot for your input.)

juliangilbey, 2020-08-12:
@jlam55555 It's Makefile syntax: writing

```
        -command
```

instead of

```
        command
```

in a Makefile means "execute command, but don't exit with an error if the command fails".  So it is sort of equivalent to writing

```
        command || true
```

Best wishes!