jlam55555/veikk-linux-driver #42: Error: ‘struct hid_report’ has no member named ‘application’

colinplease, 2020-08-13:
I'm trying to install version 2 on a machine running Ubuntu 16.04.7 LTS and am getting an error message (details below of what the make command gives me): It would be great to use my A15 on an Ubuntu machine rather than the Windows machine I currently have to use!!!  -  Thanks Colin

user@abcd23:~/bin/veikk/veikk-linux-driver-2.0$ make
make -C /lib/modules/4.15.0-112-generic/build M=/home/user/bin/veikk/veikk-linux-driver-2.0 modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-112-generic'
  CC [M]  /home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.o
In file included from ./include/linux/input.h:22:0,
                 from ./include/linux/hid.h:35,
                 from /home/user/bin/veikk/veikk-linux-driver-2.0/veikk.h:10,
                 from /home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.c:5:
/home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.c: In function ‘veikk_report’:
/home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.c:105:46: error: ‘struct hid_report’ has no member named ‘application’
              report->id, report->type, report->application, report->maxfield,
                                              ^
./include/linux/device.h:1420:58: note: in definition of macro ‘dev_info’
 #define dev_info(dev, fmt, arg...) _dev_info(dev, fmt, ##arg)
                                                          ^
/home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.c:104:5: note: in expansion of macro ‘hid_info’
     hid_info(hdev, "parsing report: %d %d %d %d %d\n",
     ^
scripts/Makefile.build:330: recipe for target '/home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.o' failed
make[2]: *** [/home/user/bin/veikk/veikk-linux-driver-2.0/veikk_drv.o] Error 1
Makefile:1582: recipe for target '_module_/home/user/bin/veikk/veikk-linux-driver-2.0' failed
make[1]: *** [_module_/home/user/bin/veikk/veikk-linux-driver-2.0] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-112-generic'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

DavidFirth, 2020-08-14:
I had the same problem a while ago, and found (looking at previous issue-reports here) that others also had had the same problem, with Ubuntu 16.04.  Maybe just revert to version 1.1 of the driver?  (that works fine for me, with my A30 tablet)

But just yesterday I decided to bite the bullet and upgrade to Ubuntu 18.04.  The automated update went surprisingly smoothly.  And the Veikk driver version 1.1 still works fine (I haven't tried version 2 with Ubuntu 18.04 yet, but I expect it will probably work fine with 18.04).  Ubuntu 16.04 expires in a few months, in any case.

colinplease, 2020-08-14:
Thanks David.

closed 2020-08-14 by colinplease