jlam55555/veikk-linux-driver #43: unable to "make" in ubuntu

test2a, 2020-08-17:


JamesTheAwesomeDude, 2020-10-02:
I'm on Fedora, and I had a lot better luck using DKMS instead of _manually_ accessing Make myself. ([Ubuntu has this, too](https://help.ubuntu.com/community/DKMS).)

Try something like the following:

1. `sudo apt install dkms`
2. `sudo git clone https://github.com/jlam55555/veikk-linux-driver.git /usr/src/input-veikk-git`
3. `sudoedit /usr/src/input-veikk-git/dkms.conf`
   * paste this in:
     ```makefile
     PACKAGE_NAME="input-veikk"
     PACKAGE_VERSION="git"
     BUILT_MODULE_NAME[0]="veikk"
     DEST_MODULE_LOCATION[0]="/extra/"
     AUTOINSTALL="yes"
     ```
4. `sudo dkms install input-veikk/git`
5. reboot

This worked for me, and _unlike_ using `make` yourself, you **don't have to re-do it every single kernel update**.

(Also, don't run `sudo make`; that'll create a bunch of root-owned files in _your_ home folder. If you're going to run `make`, run it as your user; save `sudo` for `make install`. But don't do that, either, if you're using dkms. In fact, you should run `sudo make uninstall` from within your current copy of the code, before setting up dkms.)

SnoopTheDog, 2020-11-21:
@JamesTheAwesomeDude following your instructions on fedora 33 I get the following console output:
```[snoopie@localhost ~]$ sudo dkms install input-veikk/git

Creating symlink /var/lib/dkms/input-veikk/git/source ->
                 /usr/src/input-veikk-git

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j6 KERNELRELEASE=5.9.8-200.fc33.x86_64 -C /lib/modules/5.9.8-200.fc33.x86_64/build M=/var/lib/dkms/input-veikk/git/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.9.8-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/input-veikk/git/build/make.log for more information.
```
and the following in make.log:

```DKMS make.log for input-veikk-git for kernel 5.9.8-200.fc33.x86_64 (x86_64)
Sat 21 Nov 22:46:28 CET 2020
make: Entering directory '/usr/src/kernels/5.9.8-200.fc33.x86_64'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make: *** [Makefile:719: include/config/auto.conf] Error 1
make: Leaving directory '/usr/src/kernels/5.9.8-200.fc33.x86_64'```

help would be greatly appreciated (using tablet VEIKK s640)

JamesTheAwesomeDude, 2020-11-27:
> `include/generated/autoconf.h or include/config/auto.conf are missing.`

@SnoopTheDog Try

* Updating, and also rebooting (so that the *running* kernel is the same version as the *available* kernel files)
* [installing `kernel-devel kernel-header`](https://unix.stackexchange.com/a/324262)

SnoopTheDog, 2020-11-30:
done, no difference. Updated output:
```
[snoopie@localhost veikk-linux-driver]$ make
make -C /lib/modules/5.9.10-200.fc33.x86_64/build M=/home/snoopie/veikk-linux-driver modules
make[1]: Entering directory '/usr/src/kernels/5.9.10-200.fc33.x86_64'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:719: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/kernels/5.9.10-200.fc33.x86_64'
make: *** [Makefile:8: all] Error 2

```

JamesTheAwesomeDude, 2020-11-30:
@SnoopTheDog Are you _certain_ you've installed `kernel-devel` *and since rebooted*?

What is the output of the following?

```sh
find /usr/src -name autoconf.h -o -name auto.conf -exec rpm -qf {} \;
```

SnoopTheDog, 2020-11-30:
Output:
```
[snoopie@localhost Downloads]$ find /usr/src -name autoconf.h -o -name auto.conf -exec rpm -qf {} \;
kernel-devel-5.9.8-200.fc33.x86_64
kernel-devel-5.9.10-200.fc33.x86_64
```
Also output of `dnf install kernel-devel`
```
[snoopie@localhost Downloads]$ dnf install kernel-devel
Error: This command has to be run with superuser privileges (under the root user on most systems).
[snoopie@localhost Downloads]$ sudo dnf install kernel-devel
[sudo] password for snoopie: 
Last metadata expiration check: 0:48:54 ago on Mon 30 Nov 2020 22:15:51 CET.
Package kernel-devel-5.9.8-200.fc33.x86_64 is already installed.
Package kernel-devel-5.9.10-200.fc33.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
```

asl97, 2020-12-01:
@SnoopTheDog You shouldn't hijack someone else issue, you aren't even on ubuntu.
Their issue is that they have a whitespace in their path.

`make -C /lib/modules/4.15.0-112-generic/build M=/home/user/Downloads/lo/`**Macro Solution**`/veikk-linux-driver-2.0 modules`
`make[1]: Entering directory '/usr/src/linux-headers-4.15.0-112-generic'`
`/bin/sh: 1: cannot create /home/user/Downloads/lo/`**Macro**`/.cache.mk: Directory nonexistent`

Not sure on your issue