jlam55555/veikk-linux-driver #46: Problems with stylus proximity.

jonalakes, 2020-09-13:
My veikk s640 just arrived, I couldn't figure it out how to set it up until I saw a tutorial. But unfortunately, I still have a very annoyed problem. I'm using Xournal++ to take some college notes in Xubuntu and Pop!OS, and for some reason, even without touching the surface of the tablet, I'm able  draw a thiny and soft line on the paper, sometimes invisible, but when I save the document, the disaster appears. I have to add that I can write on the canvas without pressing any button from the stylus. The same thing happen to me while I'm working with others apps. I even installed the GUI from the Github repository, but it didn't work. I just don't know what to do next. I´d be very grateful if any of you give me a piece of advice to solve this. Thanks.

asl97, 2020-11-21:
There probably some pressure the tip has, you can try removing the tip and setting it back.

If that doesn't work, there might be hardware issue, you can try setting it to ignore low pressure,
in controller terminology, basically setting a deadzone.

The button is for right click and middle click, left click is control by the tip pressure, so that's why you are able to draw without touching the surface of the tablet.

jlam55555, 2021-06-26:
Closing now. Doesn't seem to be an issue with the driver.

closed 2021-06-26 by jlam55555