jlam55555/veikk-linux-driver #51: Driver Spamming Kernel Logs

ewhac, 2020-10-17:
The current version of the driver (master branch) spams the kernel logs with coordinate readings from the tablet.  This does not make the kernel logs or its downstream consumers happy.

It looks like it's coming from the `printk()` call in `veikk_s640_handle_raw_data()`.  If so, this should probably be changed to `dev_dbg()` instead, so that the logging severity filters can screen it out.

## To Reproduce: ##

  * Compile the Veikk driver; load it using `insmod`.
  * Plug in a Veikk tablet.
  * Wave the pen briefly over the tablet surface.
  * Display the kernel logs using `dmesg`.

The logs will contain an enormous quantity of raw pen coordinate/pressure values.

## Logging Sample ##

```
[1035713.602701] usb 2-9.2: new full-speed USB device number 20 using xhci_hcd
[1035713.703786] usb 2-9.2: New USB device found, idVendor=2feb, idProduct=0006, bcdDevice= 0.00
[1035713.703789] usb 2-9.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1035713.703791] usb 2-9.2: Product: A15PRO
[1035713.703792] usb 2-9.2: Manufacturer: VEIKK.INC
[1035713.703794] usb 2-9.2: SerialNumber: 0000001
[1035713.713100] input: VEIKK A15 Pro Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.0/0003:2FEB:0006.0012/input/input41
[1035713.713243] veikk 0003:2FEB:0006.0012: hidraw2: USB HID v1.00 Mouse [VEIKK.INC A15PRO] on usb-0000:00:14.0-9.2/input0
[1035713.713245] veikk 0003:2FEB:0006.0012: VEIKK A15 Pro Pen probed successfully.
[1035713.714343] input: VEIKK A15 Pro Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.1/0003:2FEB:0006.0013/input/input42
[1035713.714487] veikk 0003:2FEB:0006.0013: hidraw3: USB HID v1.00 Keyboard [VEIKK.INC A15PRO] on usb-0000:00:14.0-9.2/input1
[1035713.714489] veikk 0003:2FEB:0006.0013: VEIKK A15 Pro Pen probed successfully.
[1035713.715213] input: VEIKK A15 Pro Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.2/0003:2FEB:0006.0014/input/input43
[1035713.715321] veikk 0003:2FEB:0006.0014: hidraw4: USB HID v1.00 Device [VEIKK.INC A15PRO] on usb-0000:00:14.0-9.2/input2
[1035713.715323] veikk 0003:2FEB:0006.0014: VEIKK A15 Pro Pen probed successfully.
[1035727.757854] 0 0 0
[1035727.779849] 14126 7588 0
[1035727.785852] 14126 7588 0
[1035727.791851] 14126 7588 0
[1035727.797848] 14126 7588 0
[1035727.803848] 14126 7588 0
[1035727.809852] 14126 7588 0
[1035727.815849] 14126 7588 0
[1035727.821847] 14126 7550 0
[1035727.825849] 14126 7522 0
[1035727.829847] 14126 7503 0
[1035727.833847] 14107 7474 0
[1035727.837846] 14093 7455 0
[1035727.841852] 14093 7426 0
[1035727.845852] 14077 7400 0
[1035727.849848] 14063 7378 0
[1035727.853848] 14048 7362 0
[1035727.857847] 14035 7338 0
[1035727.861849] 14022 7315 0
[1035727.865847] 14010 7293 0
[1035727.869846] 13997 7270 0
[1035727.873846] 13985 7251 0
[1035727.877848] 13974 7231 0
[1035727.881851] 13960 7210 0
[1035727.885864] 13948 7187 0
[1035727.889848] 13937 7166 0
[1035727.893851] 13927 7143 0
[1035727.897871] 13915 7125 0
[1035727.901848] 13903 7105 0
[1035727.905852] 13893 7084 0
[1035727.909852] 13893 7063 0
[ ...blah, blah, blah...]
```

ewhac, 2020-10-17:
This patch fixes the issue for me:

```diff
diff --git a/veikk_vdev.c b/veikk_vdev.c
index 9ec5af5..f88493f 100644
--- a/veikk_vdev.c
+++ b/veikk_vdev.c
@@ -94,6 +94,7 @@ static int veikk_s640_setup_and_register_input_devs(struct veikk *veikk) {
 static int veikk_s640_handle_raw_data(struct veikk *veikk, u8 *data, int size,
                                       unsigned int report_id) {
     struct input_dev *pen_input = veikk->pen_input;
+    struct device *dev = &veikk->hdev->dev;
     struct veikk_pen_report *pen_report;
 
     switch(report_id) {
@@ -107,7 +108,7 @@ static int veikk_s640_handle_raw_data(struct veikk *veikk, u8 *data, int size,
         pen_report = (struct veikk_pen_report *) data;
 
         // TODO: remove
-        printk("%u %u %u\n",pen_report->x, pen_report->y, pen_report->pressure);
+        dev_dbg (dev, "%u %u %u\n", pen_report->x, pen_report->y, pen_report->pressure);
 
         input_report_abs(pen_input, veikk->x_map_axis,
                          veikk->x_map_dir*pen_report->x);
```

jlam55555, 2021-09-26:
See commit

closed 2021-09-26 by jlam55555