jlam55555/veikk-linux-driver #53: Missing ../crypto/bio/bss_file.c at the make install step

OceanNuclear, 2020-11-18:
Hi, I was trying to follow the instructions in the README, and got through the
```make``` step with no problem. I got the ```sudo make all install clean``` step, but encountered issue #3 (```Required key not available```) so I went ahead and disabled SecureBoot, and now I'm here.

While SecureBoot is disabled I re-ran ```make``` and ```make all install clean```, but the ```install``` step threw some errors:

Here are my terminal outputs:
```bash
~/Downloads/veikk-linux-driver/ make
make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
  CC [M]  /home/ocean/Downloads/veikk-linux-driver/veikk_drv.o
  CC [M]  /home/ocean/Downloads/veikk-linux-driver/veikk_vdev.o
  CC [M]  /home/ocean/Downloads/veikk-linux-driver/veikk_modparms.o
  LD [M]  /home/ocean/Downloads/veikk-linux-driver/veikk.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/ocean/Downloads/veikk-linux-driver/veikk.mod.o
  LD [M]  /home/ocean/Downloads/veikk-linux-driver/veikk.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
~/Downloads/veikk-linux-driver/ sudo make all install clean
make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
  INSTALL /home/ocean/Downloads/veikk-linux-driver/veikk.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-123-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
depmod
modprobe veikk
mkdir -p /etc/modules-load.d
echo "veikk" > /etc/modules-load.d/veikk.conf
make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver clean
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
  CLEAN   /home/ocean/Downloads/veikk-linux-driver/.tmp_versions
  CLEAN   /home/ocean/Downloads/veikk-linux-driver/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
```

I noticed that this is very similar to the bug raised in #27 , which was said to be solved and closed. But obviously that's not the case.
I tried to follow the advice in #27 and run the ```sudo depmod``` and ```sudo modprobe veikk`` (both of which returned no output in the terminal) and tried ```sudo make all install clean``` again. Still, same error.

OceanNuclear, 2020-11-18:
Update:
Correct me if I'm wrong, but if I understand make files correctly,
```sudo make all install clean``` runs the "all", "install" and "clean" blocks sequentially.

If I run ```sudo make install``` and ```sudo make clean``` after that line above (```sudo make all install clean```, which gave those 4 lines of errors), they get executed without errors:

```
(canary) ~/Downloads/veikk-linux-driver/ sudo make install
make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
  DEPMOD  4.15.0-123-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
depmod
modprobe veikk
mkdir -p /etc/modules-load.d
echo "veikk" > /etc/modules-load.d/veikk.conf
(canary) ~/Downloads/veikk-linux-driver/ sudo make clean
make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver clean
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
```

So clearly the error is caused by whatever the ```make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver modules``` line did, which gets undone by the ```make -C /lib/modules/4.15.0-123-generic/build M=/home/ocean/Downloads/veikk-linux-driver clean``` in the Makefile. (I don't know if this helps/is already obvious to you or not.)

Please let me know if you have any insights on this.

OceanNuclear, 2020-11-18:
Very similar issue to/Duplicate of #45

GabeIggy, 2020-11-19:
I also have this problem.

OceanNuclear, 2020-11-20:
@GabeIggy I simply refunded my A50 tablet and swapped it for a Huion of the same price. I'd suggest you to do this asap. 

I'll close this issue and list "refund-and-replace" as the workaround once I receive it and can confirm that the Huion is a usable tablet on linux, since no one seems to be active on these threads about missing bss_file.c.

The Huion is supposed to have more functionality as well, and works with Windows/android/linux; whereas the veikk windows driver couldn't be installed on my windows machine for some reason and I can't be asked to contact veikk about it after already having messed with this linux driver for 2 hours.

asl97, 2020-11-21:
@OceanNuclear just to be clear, I ain't affiliated with this driver, I do not represent the view of the author of this driver or veikk.
> sudo depmod and sudo modprobe veikk (both of which returned no output in the terminal)

they aren't suppose to output anything
> While SecureBoot is disabled I re-ran make and make all install clean, but the install step threw some errors:

They aren't really error, it a message that says you are missing the signing key

https://askubuntu.com/questions/820883/how-to-resolve-ssl-error-during-make-modules-install-command#1178467

It shouldn't matter if you really successfully disabled SecureBoot, in other words, it should be properly installed already at this step.
> If I run sudo make install and sudo make clean after that line above, (sudo make all install clean) they get executed without errors

I am calling bluff here or your system is mess up, `sudo make install` should error after a `make all install clean` because the `clean` would remove the require file for install:

```
make[1]: Entering directory '/usr/src/kernels/5.8.15-201.fc32.x86_64'
cat: /home/asl97/projects/veikk/veikk-s640-driver/modules.order: No such file or directory
```
> whereas the veikk windows driver couldn't be installed on my windows machine for some reason and I can't be asked to contact veikk about it after already having messed with this linux driver for 2 hours.

Fun fact, this is a 3rd party **unofficial** driver for linux,

Not contacting veikk (who would have commercial reason to provide support) for your windows problem because the third party driver didn't respond to you (who don't have any obligation to help you at all) is fairly stupid if you ask me.

asl97, 2020-11-21:
@OceanNuclear TLDL: you installed it properly already, otherwise it would have thrown an error like the issue #45 you linked:
```
modprobe: ERROR: could not insert 'veikk': Operation not permitted
Makefile:12: recipe for target 'install' failed
make: *** [install] Error 1
```

That "error" can be ignored, it just means it wasn't signed, it didn't throw the above error, meaning it installed properly.

OceanNuclear, 2020-11-21:
>Not contacting veikk (who would have commercial reason to provide support) for your windows problem because the third party driver didn't respond to you

I'm perfectly aware of this, thank you very much. However I use Linux as my main work station so even if I were to successfully install the driver on windows after wasting another day of correspondence with the official veikk tech support side, I will still have to turn off Linux -> turn on Windows -> save the hand-drawn file -> turn off windows -> turn back on Linux to continue editing every time I want to draw something. That would really defeat the point of getting the tablet, which was to make hand-drawing easier, not harder.

>I am calling bluff here or your system is mess up

My system might have been quite messed up at one point, I'll admit.

If it was successfully installed, then I'm confused as to why ```lsusb``` detected no change when I plugged in the tablet, and there was no responses from the tablet whatsoever. I tried 3 different cables, so it's definitely not a cable issue. All of them lead to the led light on the tablet lighting up, so it was very unlikely to be a hardware issue.

(My assumption is that once the driver is installed I should be able to plug and play; and since it doesn't plug and play, I thought the driver wasn't installed correctly.)

asl97, 2020-11-21:
> there was no responses from the tablet whatsoever.
There should be a fallback driver even when you didn't install this, I assume it was the same on windows?

That is a sign of a faulty device.

> My assumption is that once the driver is installed I should be able to plug and play;
I assume you did restart the system after installing?
it's rare but stuff can happen.

> lsusb detected no change when I plugged in the tablet
If you mean it didn't show up in lsusb, there could be a number of reason,

1. The driver didn't load because it wasn't signed (check with lsmod and maybe other tool) and there no fallback driver.
2. The data pin could be damaged,  if the power pin are working, it would light up but would not detect
3. I do not know where you bought it from, if it's a third party, the device itself could be dead, eg: from a bad firmware update or physical damage.

OceanNuclear, 2020-11-21:
Yep, it was the same on Windows. So perhaps it was a faulty device but I didn't know of it.

>I assume you did restart

Restarted multiple times, with secure boot and without. None of those times worked.

I brought it off amazon, so it's very likely from the vendor veikk themselves.

Regardless, I have already returned the tablet, and ordered a different one. Should arrive next week.

OceanNuclear, 2020-11-25:
Closing issue as it might have been a hardware problem, not a software problem (See comment above).

Workaround: order new tablet (Huion, or try your luck with veikk again). Huion H610ProV2 has the same specs and price, and can be used on android phones as well. The installation of that driver went smoothly.

Also if you're using ubuntu, Wacom tablets are reported to work straight out-of-the-box without any of this trouble of driver installation, so you may consider that instead.

Other notes: when a functional tablet (i.e. my Huion) is plugged into linux, ```lsusb``` should detect a change even if it doesn't plug and play immediately.

closed 2020-11-25 by OceanNuclear

asl97, 2020-11-26:
Just to clarify, all tablet should work out of the box, running in fallback mouse mode using the `hid-generic` driver,

From https://everything-is-sheep.herokuapp.com/posts/veikk-linux-driver-v3-notes , too lazy to convert to GFM table.
<table>
<thead>
<tr>
<th>Model</th>
<th>USB Interface 1</th>
<th>USB Interface 2</th>
<th>USB Interface 3</th>
</tr>
</thead>
<tbody>
<tr>
<td>S640</td>
<td>Digitizer, Mouse*</td>
<td>Keyboard (unused)</td>
<td>VEIKK proprietary</td>
</tr>
<tr>
<td>A50 (mine matches <a href="https://github.com/jlam55555/veikk-linux-driver/issues/4#issuecomment-581004043" target="_blank">@nbogie's comment</a>)</td>
<td>Digitizer, Mouse*</td>
<td>Keyboard*</td>
<td>VEIKK proprietary</td>
</tr>
<tr>
<td>A15 (from <a href="https://github.com/jlam55555/veikk-linux-driver/issues/14#issuecomment-580415066" target="_blank">@LinkJ's comment</a>)</td>
<td>Mouse*</td>
<td>Digitizer, Keyboard*</td>
<td>VEIKK proprietary</td>
</tr>
<tr>
<td>VK1560</td>
<td>Mouse</td>
<td>Digitizer*, Keyboard*</td>
<td>VEIKK proprietary</td>
</tr>
</tbody>
</table>

Even Wacom should require driver installation for pressure detection.

Just a note, Ubuntu 14.04 has known issues.
https://everything-is-sheep.herokuapp.com/posts/veikk-linux-driver-v3-notes#bugsandworkarounds

> The installation of that driver went smoothly.

@OceanNuclear I take it that you were require to install some driver for the Huion too.

jlam55555, 2021-06-26:
@asl97 Thanks for helping respond when I was busy.

@OceanNuclear I agree, the most telltale sign of something fishy going on was the fact that `lsusb` doesn't show any change with the tablet getting plugged in. This should happen regardless of whether the driver was installed or not. Hopefully the Huion works well for you.