jlam55555/veikk-linux-driver #54: "make" doesn't work on raspberry pi

amelgikha, 2020-12-27:
when i type make, the feedback error is like bellow:

pi@raspberrypi:~/Downloads/veikk-linux-driver-master $ make
make -C /lib/modules/5.4.79-v7l+/build M=/home/pi/Downloads/veikk-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.79-v7l+'
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "__aeabi_ldivmod" [/home/pi/Downloads/veikk-linux-driver-master/veikk.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1645: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.79-v7l+'
make: *** [Makefile:8: all] Error 2

please help me to fix this issued

mirrorsonthewall, 2020-12-28:
The two issues I tested on Ubuntu were to first check if Secureboot was enabled (it needs to be disabled) and to also try manually loading with some different commands.

Here is an [issue #56](https://github.com/jlam55555/veikk-linux-driver/issues/56) on how installing on Ubuntu worked when tested

which is based off of [issue #3](https://github.com/jlam55555/veikk-linux-driver/issues/3) which deals with disabling Secureboot 

and [issue #27](https://github.com/jlam55555/veikk-linux-driver/issues/27) which required some extra commands to install with.

The Ubuntu install also had Errors 1 and 2 but I'm not sure if they're the same.

mcyph, 2021-02-03:
This looks to be caused by 64 bit division on a non-32 bit kernel. 

I've submitted a PR to https://github.com/jlam55555/veikk-linux-driver/issues/54 to use `div64_s64` from `<linux/math64.h>`, which should fix this.

kip-the-elder, 2021-04-16:
I have got this to work on the 64 bit beta OS. Using just the 64bit kernel did not work for me.