jlam55555/veikk-linux-driver #56: Working Step-By-Step For Ubuntu (Add To README?)

mirrorsonthewall, 2020-12-28:
This issue post is based off of other closed issues.

Secureboot was enabled, so from [issue #3](https://github.com/jlam55555/veikk-linux-driver/issues/3) it needed to be disabled and [this guide says how to disable it](https://itadminguide.com/disable-secure-boot-in-ubuntu/). (Could this guide be added to Issue #3 or linked in README?)

(Note: When it asks to enter password character 7 for example, if you set the password as 12345678 then you can enter "7", or if it asks for password character 4, can enter "4")

(Note: I believe Secureboot must remain "disabled" for this driver to work, see ["Is It Safe To Disable Secureboot?"](https://askubuntu.com/questions/843656/is-it-safe-to-disable-secure-boot) - the answer appears to be generally "yes".)

Then after rebooting, ([solution from issue #27](https://github.com/jlam55555/veikk-linux-driver/issues/27)) navigate to the veikk-linux-driver-master directory and:

sudo depmod

sudo modprobe veikk

sudo make all install clean

(maybe we could add to this post if others have problems and we work through them? I don't think other issues presented themselves)

jlam55555, 2021-06-26:
Thanks, I will add this to the README. I thought I had mentioned Secure Boot on the README at some point but it looks like I didn't (explicitly, at least).

ssteinerx, 2023-09-12:
The 
``` bash
depmod
modprobe veikk 
```
saved me from:
``` bash
arning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-32-generic'
modprobe veikk
modprobe: FATAL: Module veikk not found in directory /lib/modules/6.2.0-32-generic
make: *** [Makefile:15: install] Error 1
```