jlam55555/veikk-linux-driver #62: What license does this project use? (Question for package submission)

nicbk, 2021-06-05:
Hey,

I packaged this driver (and the corresponding GUI utility) for NixOS ([driver pkg submission](https://github.com/NixOS/nixpkgs/pull/125856), [gui pkg submission](https://github.com/NixOS/nixpkgs/pull/125859)). However, I was not able to find any information regarding what license these programs use. I declared the packages as `non-free` to not assume and/or infringe upon the rights of @jlam55555.

I am asking whether these programs actually have any licenses (GPL/MIT/Unlicense etc.), so that I can go ahead and update the packages.

Thank you,
Nicolás Kennedy

asl97, 2021-06-06:
@nicbk did you... Idk, try searching license? (lol)

https://github.com/jlam55555/veikk-linux-driver/blob/bb938f011669993473ea8938f1585a3c15605fbb/veikk.h#L20

nicbk, 2021-06-06:
Thank you. I apologize for my laziness.

I'm changing the package to "GPL v2 only". However, according to [the kernel licensing guidelines](https://www.kernel.org/doc/html/latest/process/license-rules.html), it is still ambiguous as to whether the these programs are "GPL v2 only" or "GPL v2 or later". As far as I can tell, I think that it is legal for me to declare these packages as "GPL v2 only", so I will stick to that for now. However, if this package is indeed "GPL v2 or later", then it would be appropriate for the package to be changed to reflect that specific license.

For that reason, I will keep this issue open until there is further clarification.

Hunanbean, 2021-06-06:
asl97: the license is ambiguous at best and not posted on the github page in the normal way. Professionalism is always preferred. Be well.

nicbk: No worries. The license is not posted in the expected location on the git pages. I would not consider it as laziness on your part. I also have been unable to determine the exact license. Thank you for working towards extending the availability of this very helpful driver project.

jlam55555, 2021-06-07:
@nicbk @Hunanbean @asl97 The reason the license is missing is because I honestly have no idea what the license should be, or how to properly implement it. Call me ignorant, but I've not had a project go into the open domain before and I'm really bad at understanding legalese.

I put GPL in there at the beginning because I thought it was a requirement for it to access other GPL symbols. Can you advise on what the appropriate license (whether GPL or not) should be? (Similarly, for the config GUI, I don't know what the licensing for Qt applications are. If they're too strict, I might end rewriting the v3 configuration tool in GTK.) I don't have any intention of limiting the rights of this software to anyone, especially since it's not especially good or complete software.

And like Hunanbean said, thank you for your work with packaging. I'm sorry that I've been having trouble managing my time outside of school, so this work is greatly appreciated.

asl97, 2021-06-07:
@Hunanbean I was address the fact that the packages was declared as "non-free", ambiguous as it may be, all GPL version (to my knowledge) is "free".

The license may not be in the expected location for github but it is for an driver.
> Here's the output of "modinfo veikk"
> 
> ```
> filename:       /lib/modules/5.11.20-300.fc34.x86_64/extra/veikk.ko
> license:        GPL
> description:    USB VEIKK drawing tablet driver
> author:         Jonathan Lam <jlam55555@gmail.com>
> ```

There are a number of location where it is proper to place license info, eg: the License, README in root of project folder, in each file header, the about/help page, in the metadata, etc.  Software are created and published in more places than just github.

@jlam55555 
> I put GPL in there at the beginning because I thought it was a requirement for it to access other GPL symbols.
is there limitation in using an more open license?

asl97, 2021-06-07:
I notice the modules on my system only uses the following:
```
$ modinfo -l /lib/modules/*/*/*/*/*.ko.* | sort -u
Dual BSD/GPL
Dual MPL/GPL
GPL
GPL v2
```

jlam55555, 2021-06-26:
@nicbk Sorry for the delayed response, was busy with work again.

I think I will add a "GPLv2 or later" license to this project and a GPLv3 license to

jlam55555, 2021-06-26:
To everyone in this thread:

I think I will add a GPLv2 (or later) clause to this repository and a GPLv3 clause to the config repository. As far as I can tell, the Linux kernel is licensed under GPLv2 (only), and GPLv3 is not backwards-compatible with GPLv2, Qt is mostly licensed under (L)GPLv3, and GPL is automatically forward compatible (i.e., there is a clause "or (at your option) any later version" in the GPL licenses).

Again, I'm very new to open source licenses, so please correct me if I'm wrong. I will add the GitHub LICENSE template here and on the other repo if someone confirms this.

On an aside, this weekend I will be working on the v3 driver with button support in a new repo (new project) this weekend, because I have a better idea of what needs to be done for button mapping to work and it's sufficiently different from the v2 mess. More info in the mailing list/improvements thread to come later today.