jlam55555/veikk-linux-driver #65: S640 not getting recognized in Ubuntu 20.04 even as a mouse

rwikdutta, 2021-07-06:
I purchased Veikk S640 drawing tablet and it is working on Windows and also Android. But I am not being able to use it on Ubuntu 20.04 which is my primary work machine, not even as a mouse. I also tried installing your driver ( both the alpha as well as the stable version, but nothing is happening ). Can you help?

Here is the output of tail -f /var/log/syslog when I inserted the device.
```
ul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.046156] usb 1-3: new full-speed USB device number 16 using xhci_hcd
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.200204] usb 1-3: New USB device found, idVendor=2feb, idProduct=0001, bcdDevice= 0.00
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.200206] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.200207] usb 1-3: Product: S640
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.200208] usb 1-3: Manufacturer: VEIKK.INC
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.200208] usb 1-3: SerialNumber: 0000001
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.203817] input: VEIKK S640 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:2FEB:0001.0022/input/input63
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.262171] veikk 0003:2FEB:0001.0022: hidraw1: USB HID v1.00 Mouse [VEIKK.INC S640] on usb-0000:00:14.0-3/input0
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.262173] veikk 0003:2FEB:0001.0022: VEIKK S640 Pen probed successfully.
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.263253] input: VEIKK S640 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:2FEB:0001.0023/input/input64
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.263326] veikk 0003:2FEB:0001.0023: hidraw2: USB HID v1.00 Device [VEIKK.INC S640] on usb-0000:00:14.0-3/input1
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.263327] veikk 0003:2FEB:0001.0023: VEIKK S640 Pen probed successfully.
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.264208] input: VEIKK S640 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:2FEB:0001.0024/input/input65
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.264272] veikk 0003:2FEB:0001.0024: hidraw3: USB HID v1.00 Device [VEIKK.INC S640] on usb-0000:00:14.0-3/input2
Jul  6 20:51:07 rwikdutta-Latitude-7490 kernel: [ 2780.264276] veikk 0003:2FEB:0001.0024: VEIKK S640 Pen probed successfully.
Jul  6 20:51:07 rwikdutta-Latitude-7490 mtp-probe: checking bus 1, device 16: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Jul  6 20:51:07 rwikdutta-Latitude-7490 mtp-probe: bus: 1, device: 16 was not an MTP device
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) config/udev: Adding input device VEIKK S640 Pen (/dev/input/mouse2)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) No input driver specified, ignoring this device.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) This device may have been added with another device file.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) config/udev: Adding input device VEIKK S640 Pen (/dev/input/mouse3)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) No input driver specified, ignoring this device.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) This device may have been added with another device file.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) config/udev: Adding input device VEIKK S640 Pen (/dev/input/mouse4)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) No input driver specified, ignoring this device.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) This device may have been added with another device file.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) config/udev: Adding input device VEIKK S640 Pen (/dev/input/event8)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) VEIKK S640 Pen: Applying InputClass "libinput tablet catchall"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) Using input driver 'libinput' for 'VEIKK S640 Pen'
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) systemd-logind: got fd for /dev/input/event8 13:72 fd 98 paused 0
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) VEIKK S640 Pen: always reports core events
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) Option "Device" "/dev/input/event8"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) Option "_source" "server/udev"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: is tagged by udev as: Tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: tablet 'VEIKK S640 Pen' unknown to libwacom
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: device is a tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: device removed
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:2FEB:0001.0022/input/input63/event8"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) XINPUT: Adding extended input device "VEIKK S640 Pen" (type: TABLET, id 10)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: is tagged by udev as: Tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: tablet 'VEIKK S640 Pen' unknown to libwacom
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event8  - VEIKK S640 Pen: device is a tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 mtp-probe: checking bus 1, device 16: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Jul  6 20:51:07 rwikdutta-Latitude-7490 mtp-probe: bus: 1, device: 16 was not an MTP device
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) config/udev: Adding input device VEIKK S640 Pen (/dev/input/event11)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) VEIKK S640 Pen: Applying InputClass "libinput tablet catchall"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) Using input driver 'libinput' for 'VEIKK S640 Pen'
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) systemd-logind: got fd for /dev/input/event11 13:75 fd 99 paused 0
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) VEIKK S640 Pen: always reports core events
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) Option "Device" "/dev/input/event11"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) Option "_source" "server/udev"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event11 - VEIKK S640 Pen: is tagged by udev as: Tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event11 - VEIKK S640 Pen: tablet 'VEIKK S640 Pen' unknown to libwacom
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event11 - VEIKK S640 Pen: device is a tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event11 - VEIKK S640 Pen: device removed
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:2FEB:0001.0024/input/input65/event11"
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) XINPUT: Adding extended input device "VEIKK S640 Pen" (type: TABLET, id 11)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event11 - VEIKK S640 Pen: is tagged by udev as: Tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event11 - VEIKK S640 Pen: tablet 'VEIKK S640 Pen' unknown to libwacom
[... cut ...]
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) XINPUT: Adding extended input device "VEIKK S640 Pen" (type: TABLET, id 12)
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event10 - VEIKK S640 Pen: is tagged by udev as: Tablet
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event10 - VEIKK S640 Pen: tablet 'VEIKK S640 Pen' unknown to libwacom
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) event10 - VEIKK S640 Pen: device is a tablet
```

asl97, 2021-07-07:
```
Jul  6 20:51:07 rwikdutta-Latitude-7490 mtp-probe: checking bus 1, device 16: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Jul  6 20:51:07 rwikdutta-Latitude-7490 mtp-probe: bus: 1, device: 16 was not an MTP device
```
This tells me it might be an outdated kernel issue.

Other than that, It is recognized as a tablet, try it in a program with support for tablet.

V3 is still in very early development as far as I am aware of.

You said you tried the V2 driver, maybe the logs there would give more info.

rwikdutta, 2021-07-07:
@asl97 I tried it in Krita. It is not working in Krita too.

> You said you tried the V2 driver, maybe the logs there would give more info.

Can you tell me what logs exactly do you want? I can uninstall v3 driver and then re-install v2 driver. But it will be helpful if you can tell me you want the logs from which files/which command?

asl97, 2021-07-07:
Let me try that again.

```
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) No input driver specified, ignoring this device.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) No input driver specified, ignoring this device.
Jul  6 20:51:07 rwikdutta-Latitude-7490 /usr/lib/gdm3/gdm-x-session[1768]: (II) No input driver specified, ignoring this device.
```
This tells me it might be an outdated kernel/missing conf issue.

It is missing the underlying driver/config that handles the device, this veikk driver just handle the input.

On my system, it says hidraw is used.
```
Jul  7 14:57:06 localhost kernel: veikk 0003:2FEB:0001.0001: hidraw0: USB HID v1.00 Mouse [VEIKK.INC S640] on usb-0000:00:14.0-2/input0
```

Sorry that I don't know the specific instruction to help you with your issue,  hopefully this hint helps you find the solution.

jlam55555, 2021-07-10:
@rwikdutta The kernel lines that begin with "veikk" are coming from the driver, so the driver does seem to work and be recognized. It seems that the issue is something involving how the X server is handling those events. 

Just to be sure, can you try `evtest` and see if the VEIKK device is emitting events there? You can also try `xinput` and then `xinput test [device-id]` to see if X is not receiving those events. If this is the issue, I may need to set some of the properties on the input device. I'm not really sure what the rules are so it might take some fiddling.

@asl97 I was able to take some time this week to work on the v3 driver, but I've moved it to a different place. See [this](https://github.com/jlam55555/veikk-linux-driver/issues/40#issuecomment-869042948). The config tool is basically ready, its interface is just a little rough (CLI only) and the installation instructions are not well documented yet, but it is working for me.

yassir-a-p, 2021-08-04:
Just in case this can solve your problem.
Earlier I was about to submit a new issue about my S460 also not getting recognized even as mouse input on my Manjaro, while it's working fine on KDE Neon on the same machine. It confused me because it worked just fine in April, then I hadn't been using it for months, then it just didn't work after I updated my Manjaro several days ago.

I had been trying to figure out why it wasn't working. I manually installed the 2.0 version from here then tried the one from AUR. But it still didn't work. Initially I was accusing that it was because of the naming pattern of the package on Manjaro, as now it's called **linux[version number]-headers** instead of **linux-headers-[version number]**. The current one I use is called **linux510-headers** on Manjaro's official repo.

But fortunately as I was just blindly trying every possible tweaks, I reinstalled the linux headers package while the AUR one is still installed and then it worked.
I don't know if this is/will be useful, but fwiw this is the reinstallation log on my Manjaro:
[linux-headers-reinstallation-2021-08-04.txt](https://github.com/jlam55555/veikk-linux-driver/files/6929017/linux-headers-reinstallation-2021-08-04.txt)

rwikdutta, 2021-08-28:
@jlam55555 
> Just to be sure, can you try `evtest` and see if the VEIKK device is emitting events there? You can also try `xinput` and then `xinput test [device-id]` to see if X is not receiving those events. If [...]

rwikdutta@rwikdutta-Latitude-7490:~$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Sleep Button
/dev/input/event3:	Power Button
/dev/input/event4:	AT Translated Set 2 keyboard
/dev/input/event5:	Dell WMI hotkeys
/dev/input/event6:	DELL081C:00 044E:121F Mouse
/dev/input/event7:	DELL081C:00 044E:121F Touchpad
/dev/input/event8:	Intel HID events
/dev/input/event9:	Intel HID 5 button array
/dev/input/event10:	DELL081C:00 044E:121F UNKNOWN
/dev/input/event11:	Integrated_Webcam_FHD: Integrat
/dev/input/event12:	Video Bus
/dev/input/event13:	HDA Intel PCH Headphone Mic
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event15:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event16:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event17:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event18:	HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event19:	Plantronics Plantronics Blackwire 725
/dev/input/event20:	VEIKK.INC S640 Mouse
/dev/input/event21:	VEIKK.INC S640 Keyboard
/dev/input/event22:	VEIKK.INC S640

I checked event20 and event21 and no events was coming there...all the events were coming to event22...i have added a few lines of output that i was getting with event22 below:
Event: time 1630135369.859147, type 3 (EV_ABS), code 0 (ABS_X), value 29557
Event: time 1630135369.859147, type 3 (EV_ABS), code 1 (ABS_Y), value 20410
Event: time 1630135369.859147, -------------- SYN_REPORT ------------
Event: time 1630135369.863183, type 3 (EV_ABS), code 0 (ABS_X), value 29552
Event: time 1630135369.863183, type 3 (EV_ABS), code 1 (ABS_Y), value 20264
Event: time 1630135369.863183, -------------- SYN_REPORT ------------
Event: time 1630135369.867145, type 3 (EV_ABS), code 0 (ABS_X), value 29530
Event: time 1630135369.867145, type 3 (EV_ABS), code 1 (ABS_Y), value 20103
Event: time 1630135369.867145, -------------- SYN_REPORT ------------
Event: time 1630135369.871172, type 3 (EV_ABS), code 0 (ABS_X), value 29506
Event: time 1630135369.871172, type 3 (EV_ABS), code 1 (ABS_Y), value 19928
Event: time 1630135369.871172, -------------- SYN_REPORT ------------
Event: time 1630135369.875251, type 3 (EV_ABS), code 0 (ABS_X), value 29461
Event: time 1630135369.875251, type 3 (EV_ABS), code 1 (ABS_Y), value 19749
Event: time 1630135369.875251, -------------- SYN_REPORT ------------
Event: time 1630135369.879271, type 3 (EV_ABS), code 0 (ABS_X), value 29413
Event: time 1630135369.879271, type 3 (EV_ABS), code 1 (ABS_Y), value 19563
Event: time 1630135369.879271, -------------- SYN_REPORT ------------
Event: time 1630135369.883181, type 3 (EV_ABS), code 0 (ABS_X), value 29359
Event: time 1630135369.883181, type 3 (EV_ABS), code 1 (ABS_Y), value 19362
Event: time 1630135369.883181, -------------- SYN_REPORT ------------
Event: time 1630135369.887181, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1630135369.887181, -------------- SYN_REPORT ------------