jlam55555/veikk-linux-driver #71: Announcement: official veikk linux drivers now available -- deb and rpm

manojbaishya, 2021-09-19:
To all reading this, we finally have official Linux drivers available for download from the Veikk website, both in deb and rpm format.
Here is the link: https://www.veikk.com/nav/14.html

I would like to thank @jlam55555 for all the hard work he has done over the last year to get these darned tablets working! ❤️ It was very inspiring to read the blog post at https://everything-is-sheep.herokuapp.com/posts/on-developing-a-linux-driver and was planning to submit PRs for added functionality, but that required not only learning C and Linux driver architecture but also reverse engineering the tablet features which is a daunting task. So our work has been cut short by the official releases 😄 

## Process:
1. Download.
2. Install.
3. Reboot.

```/usr/lib/vktablet/vktablet``` must be in Autostart at Login.

## Some screenshots:
```
λ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Linuxmint
Description:	Linux Mint 20.2
Release:	20.2
Codename:	uma
```

### Taskbar icon
[image]

### Tablet configuration
[image]
[image]

### Brush with Pressure Sensitivity in Krita
[image]

apit, 2021-09-20:
wohooo. thx for the great news! my vk640 now works flawlessly :partying_face:

karamelflan, 2021-09-20:
Thank you so much for helping me build my portfolio, jlam!!!!!! I used your
driver in my old Linux machine last July while I was really struggling with
life, and now at present, I can't thank you enough!!!
Once I'm stable, I'll make sure to donate hahaha! Thank you for the year of
support!!!

jlam55555, 2021-09-26:
@manojbaishya This is great news! I am really sorry for letting this project mostly fall to abandonware and never finishing all the mappings but school and work are beating me down nonstop.

Let me know if you want to chat about the implementation. I'm interested in how the button and pen mappings were implemented in the official implementation, purely from a software engineering perspective. I had gotten started (twice) on user-space button mappings, but never found the time to productionize them.

@karamelflan I am very happy to hear that this unofficial stopgap driver was helpful to you! It humbles me a lot to hear that.

I will update the README with this information. Cheers!

asl97, 2021-09-30:
@jlam55555 Seem like @manojbaishya is just a regular user who just happen to saw it.

Hope you wouldn't do anything rash like delete this repo, there are still uses for it.

The official driver is as far as I am aware of, is close source binary and stuffs happens to binary,  ie, it's not statically compiled and questionable support for non-rpm/deb system and other arch.

Not sure how I feel about leaving their program open all the time ether.

jlam55555, 2021-09-30:
@asl97 I was in a rush and didn't check who the "we" referred to in the original post, thanks for giving the heads up.

Don't worry, I will leave this repo open. The new notice I put at the top of the README still holds true -- my development is simply on indefinite hiatus, but now there is an official software (and presumably a channel for official support). I think rpm/deb should cover many of the more casual Linux users and they won't care much about open-source, so it's a good solution -- I'm glad to hear that it's helped some people out already.

manojbaishya, 2021-10-02:
Hey @jlam55555, priorities are important, and you did not have to do this thing anyway!

> Let me know if you want to chat about the implementation.

Uh, I am neither knowledgeable in Linux driver programming, nor C. At work, I use C#/.NET and so it will take me a whole year to ramp up and start contributing. I am not sure what will be the return from those efforts, therefore I do not give my commitment, but thank you for asking, may be some other enthusiast reading this can collaborate with you someday!

> Seem like @manojbaishya is just a regular user who just happen to saw it.

Hi @asl97, thanks for your concerns, yes I am a casual Linux user and am just happy that there is a driver from the very people who designed the electronics. I stumbled upon them as I dual-boot and went to the Veikk website to get updated Windows drivers. Since it has been confirmed that this repo will not be deleted, some other contributor with available bandwidth can pick up the work! I agree that most of the regular users don't care whether the deb driver is open-source or not, it's better to have something than nothing, having purchased the tablet already. Since a lot of people just use ubuntu or its derivatives or fedora, these should cover those use cases. If there are concerns about telemetry or data collection or spyware, then the driver from this repository is a fantastic substitute! Peace and cheers!

dc740, 2021-12-08:
If you check the binary drivers, the license is LGPL. But then they should provide the source code, which I can find.

Did anyone find the source code or a link to it?
Thank you

dc740, 2021-12-09:
UPDATE:
I got in touch with Veikk support center. They told me they won't release the source code. They also refused to provide a license for the binary so I don't know what license applies to it. The package says LGPL, but they refuse to comply.

It's weird that they don't release it. Looks like the binary is just speaking with the raw HID devices, without any specific kernel module. It'd be easier for anyone to just have the source available. The tablet works GREAT under Ubuntu+Krita, so I'd still recommend it.

iamashwin99, 2022-05-30:
The official driver has some problem.
I kept getting a lot of CPU usage from `systemd-journald` (like constant 30-40%) so I ran `journalctl` to find the error:

`usb 1-1: usbfs: process 10508 (vktablet) did not claim interface 2 before use`

Anyone else faced this issue? any way to fox this?
A quick solution I found was to just remove the tablet and plugback in, but that's not a permanet solution.

pheiduck, 2022-08-17:
Link was updated: https://veikk.com/?route=v2/support/down